raspberrypi/linux #5008: Kernel 5.18-rc4 64bit FKMS boot issue

Dark-Sky, 2022-04-25:
### Describe the bug

Ever since kernel 5.18-rc3 I am unable to boot in to a Desktop Environment with FKMS. No problem with 5.18-rc2 .

With my VIZIO tv I have always had to do a manual config in config.txt with the pi4 for KMS/FKMS or all I get is a black screen. Everything works good with KMS configured to boot but all I get is a black screen with FKMS configured to boot. It is like my manual config in config.txt is ignored.

I hooked up to my Gateway monitor and I can see the lines going by at firs boot but when it tries to go in to graphics mode the screen immediately goes black.

With both monitors I am unable to log in with ssh. It is like maybe the system is locked up.

### Steps to reproduce the behaviour

Configure config.txt to boot in to FKMS.

### Device (s)

Raspberry Pi 4 Mod. B

### System

`Linux pi4 5.18.0-rc4-1-MANJARO-ARM-RPI #1 SMP PREEMPT Mon Apr 25 19:08:32 UTC 2022 aarch64 GNU/Linux`

```
[ray@pi4 ~]$ vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
```

### Logs

Unable to retrieve any logs.

popcornmix, 2022-04-27:
Can you test https://github.com/raspberrypi/linux/pull/5009

Dark-Sky, 2022-04-27:
> Can you test #5009

Sure; be glad to.

Dark-Sky, 2022-04-27:
@popcornmix FKMS is now working again with the new commit on branch 5.15.y. Could you please make commits to 5.17.y and 5.18.y.

I appreciate you taking the time to look in to this. We have some DE's that has issues with KMS from time to time that works with FKMS. Their also seems to be some monitors/tv's that works better with FKMS.

pelwell, 2022-04-27:
The newer branches are updated now.

Dark-Sky, 2022-04-27:
Thank you so much!!

Closing issue now...

closed 2022-04-27 by Dark-Sky