raspberrypi/linux #5012: Rpi3 Kernel 5.15.x HiFiBerry DAC+ is not recognised

antonellocaroli, 2022-04-27:
### Describe the bug

I compile the Kernel for ARCH 64
I have no problem with the 5.10.x kernel

But with Kernel 5.15.x the device is not detected

### Steps to reproduce the behaviour

?

### Device (s)

Raspberry Pi 3 Mod. B+

### System

an 20 2022 13:58:39 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start_cd)

5.15.35-v8+ #1 SMP PREEMPT Wed Apr 27 19:35:56 CEST 2022 aarch64 GNU/Linux

### Logs

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.35-v8+ (root@GentooServer-i9) (aarch64-unknown-linux-gnu-gcc (Gentoo 11.2.1_p20220115 p4) 11.2.1 20220115, GNU ld (Gentoo 2.37_p1 p1) 2.37) #1 SMP PREEMPT Wed Apr 27 19:35:56 CEST 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] percpu: Embedded 27 pages/cpu s72856 r8192 d29544 u110592
[    0.000000] pcpu-alloc: s72856 r8192 d29544 u110592 alloc=27*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 251496
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=21d54516-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 914796K/1021952K available (10880K kernel code, 1830K rwdata, 3764K rodata, 3520K init, 904K bss, 41620K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33800 entries in 133 pages
[    0.000000] ftrace: allocated 133 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] random: get_random_bytes called from start_kernel+0x510/0x70c with crng_init=1
[... 215 lines cut ...]
[   17.795236] : Unregister from media controller
[   17.795275] (efault): Unregistering node (null)[0] device node /dev/video0
[   17.795296] (efault): Unregistering node (null)[0] device node /dev/video0
[   17.795314] (efault): Unregistering node (null)[0] device node /dev/video0
[   17.795332] (efault): Unregistering node (null)[0] device node /dev/video0
[   17.797488] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   17.819128] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   17.830649] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   18.024634] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   18.113649] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   18.234178] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   18.236565] usbcore: registered new interface driver brcmfmac
[   18.239115] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.bin failed with error -2
[   18.378513] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   18.378734] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   18.401641] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[   19.612200] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null). Quota mode: none.
[   27.238640] NET: Registered PF_INET6 protocol family
[   27.241667] Segment Routing with IPv6
[   27.241734] In-situ OAM (IOAM) with IPv6
[   27.731082] 8021q: 802.1Q VLAN Support v1.8
[   28.046537] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   28.050301] 8021q: adding VLAN 0 to HW filter on device eth0
[   28.107628] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   28.113063] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

pelwell, 2022-04-27:
See https://github.com/raspberrypi/linux/issues/4898 for a possible explanation and workaround.

antonellocaroli, 2022-04-28:
I have followed the discussion a bit

with kernel 5.10 

```
2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

with kernel 5.15

```
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- 4d -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

in both cases

```
cat /proc/device-tree/chosen/*warnings
dtwarn: overlay 'pi3-disable-bt' has been renamed 'disable-bt'
```

antonellocaroli, 2022-04-28:
for information with rpi4 no problem

pelwell, 2022-04-28:
If you have a spare SD card I suggest you try installing our current 64-bit image, also based on 5.15, just to see if the hardware is detected and the driver loaded - it should appear in the list of sound devices which is accessed via the volume control widget in the menu bar.

pelwell, 2022-04-28:
With the image you have now you can try the following:
```
$ dtc -I fs /proc/device-tree | grep -E "(hifiberry|pcm)"
$ lsmod | grep -E "(hifiberry|pcm)"
$ sudo modprobe snd-soc-hifiberry-dacplus
$ lsmod | grep -E "(hifiberry|pcm)"
```

antonellocaroli, 2022-04-28:
I tried raspbian64 and it works. Could it be that something is missing in the kernel configuration? even though I used bcmrpi3_defconfig.

dtc -I fs /proc/device-tree | grep -E "(hifiberry|pcm)"

```
<stdout>: Warning (unit_address_vs_reg): /soc: node has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): /soc/gpiomem: node has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): /soc/axiperf: node has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): /axi/vc_mem: node has a reg or ranges property, but no unit name
<stdout>: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
<stdout>: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
<stdout>: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
<stdout>: Warning (avoid_default_addr_size): /axi/vc_mem: Relying on default #address-cells value
<stdout>: Warning (avoid_default_addr_size): /axi/vc_mem: Relying on default #size-cells value
<stdout>: Warning (avoid_unnecessary_addr_size): Failed prerequisite 'avoid_default_addr_size'
<stdout>: Warning (unique_unit_address): Failed prerequisite 'avoid_default_addr_size'
<stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 4 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 6 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/i2c@7e804000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/i2c@7e804000/pcm5122@4d:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/aux@7e215000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/csi@7e801000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/csi@7e801000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/mmcnr@7e300000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/mmc@7e202000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/csi@7e800000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/csi@7e800000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/pwm@7e20c000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/thermal@7e212000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/spi@7e215080:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/serial@7e215040:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/smi@7e600000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 1 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 3 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 5 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 7 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 9 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 11 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/i2c@7e205000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/spi@7e2150c0:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/usb@7e980000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 4 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/i2c@7e805000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/i2s@7e203000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/hdmi@7e902000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/hdmi@7e902000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 0 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 2 is not a phandle reference
<stdout>: Warning (clocks_property): /soc/mmc@7e300000:clocks: cell 0 is not a phandle reference
[... cut ...]
snd_soc_hifiberry_dacplus    16384  0
snd_soc_core          245760  3 snd_soc_pcm512x,snd_soc_hifiberry_dacplus,snd_soc_bcm2835_i2s
snd_pcm_dmaengine      20480  1 snd_soc_core
snd_pcm               139264  5 snd_soc_pcm512x,snd_soc_hifiberry_dacplus,snd_soc_bcm2835_i2s,snd_soc_core,snd_pcm_dmaengine
snd_timer              36864  1 snd_pcm
snd                   102400  4 snd_soc_pcm512x,snd_timer,snd_soc_core,snd_pcm
clk_hifiberry_dacpro    16384  0
```
 aplay -l

`aplay: device_list:274: no soundcards found...`

antonellocaroli, 2022-04-28:
Yes, it is the output without modprobe...

closed 2022-04-28 by antonellocaroli

pelwell, 2022-04-28:
I'll have a look for the difference and fix bcmrpi3_defconfig anyway.