raspberrypi/linux #5014: ov7251: camera sensor driver fails to probe enable gpio

tewarid, 2022-04-28:
### Describe the bug

ov7251.c driver expects to be able to successfully probe enable gpio at

https://github.com/raspberrypi/linux/blob/e6f88767eac66b211361c85b5803fe71a4053959/drivers/media/i2c/ov7251.c#L1334

But, ov7251 overlay was changed by 79fc876cf1831311c0845dd0f678818ac27ca20a to move away from the regulator framework and hence `enable-gpios` was removed.

Is anyone working on fixing the driver?

### Steps to reproduce the behaviour

Enable legacy camera support and use the ov7251 overlay with Arducam ov7251.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Using yocto poky distro.

```log
sh-5.1# vcgencmd version
Apr 21 2021 15:48:42 
Copyright (c) 2012 Broadcom
version a48d332c35ee1c1c1ab433228e23317f62dcc5fb (clean) (release) (start_x)
```

Kernel version 5.10.31-v7l.

### Logs

dmesg shows

```log
[    7.729684] ov7251 10-0060: cannot get enable gpio
[    7.737066] ov7251: probe of 10-0060 failed with error -2
[   33.115370] ov7251_dvdd: disabling
[   33.118841] ov7251_dovdd: disabling
[   33.122403] ov7251_avdd: disabling
```

6by9, 2022-04-28:
> Enable legacy camera support and use the ov7251 overlay with Arducam ov7251.

The ov7251 is not supported by the legacy stack.

rpi-5.15.y is now the current branch. 
rpi-5.15.y: https://github.com/raspberrypi/linux/commit/0a2c33ceaefd770ef7194d263a88d3c04645742c
rpi-5.10.y: https://github.com/raspberrypi/linux/commit/42800edf533356c290121a5418e931cb4a0c72d3

5.10.31 is pretty ancient as it was released in April 2021. The overlay change was December 2021. So you appear to have updated your overlays but not your kernel - this is a bad idea as the two should be updated together!

tewarid, 2022-04-28:
> The ov7251 is not supported by the legacy stack.

Interesting - I'll try it with 

> So you appear to have updated your overlays but not your kernel - this is a bad idea as the two should be updated together!

Thanks - that indeed seems to be the case. Overlay change 79fc876cf1831311c0845dd0f678818ac27ca20a happened in Jan 2021 and the kernel source I have does not yet have 42800edf533356c290121a5418e931cb4a0c72d3. I shall update the issue to reflect that.

closed 2022-04-28 by tewarid