raspberrypi/linux #5016: HDMI  sound broken with kernel 5.15.36

pejobo, 2022-04-29:
### Describe the bug

HDMI sound is distorted with 5.15.36 on Pi 3B. See also https://archlinuxarm.org/forum/viewtopic.php?f=57&t=16012

Rolling back to 5.15.34 fixed the issue for me. 

### Steps to reproduce the behaviour

Play sound over HDMI using pulse audio (tested with mp3, aac) and check for distortion, skips, etc. 

### Device (s)

Raspberry Pi 3 Mod. B

### System

Operating System: Arch Linux ARM
          Kernel: Linux 5.15.36-1-rpi-ARCH
    Architecture: arm

### Logs

Nothing shows up in dmsg

### Additional context

Sounds over audio jack is fine.

coffeinflash, 2022-04-30:
Just some logs running kodi-standalone with plain alsa - no pulseaudio, pipewire etc.:

System:
`OS: Arch Linux ARM armv7l
Host: Raspberry Pi 3 Model B Rev 1.`

Kernel:
`Linux Kleinerbaer 5.15.36-1-rpi-ARCH #1 SMP Thu Apr 28 18:45:07 UTC 2022 armv7l GNU/Linux`

kodi.log (playing Livestream):
`    2022-04-30 08:43:42.228 T:595      info <general>: VideoPlayer::OpenFile: pvr://channels/tv/Alle%20Kan%c3%a4le/pvr.waipu_1142741733.pvr
    2022-04-30 08:43:42.233 T:1000     info <general>: Creating InputStream
    2022-04-30 08:43:42.500 T:1000     info <general>: AddOnLog: inputstream.adaptive: SetVideoResolution (1920x1080)
    2022-04-30 08:43:42.710 T:1000     info <general>: AddOnLog: inputstream.adaptive: Successfully parsed manifest file. #Periods: 1, #Streams in first period: 2, Type: live, Download speed: 988745.7895 Bytes/s
    2022-04-30 08:43:43.464 T:1000     info <general>: Creating Demuxer
    2022-04-30 08:43:43.472 T:1000     info <general>: Opening stream: 1001 source: 256
    2022-04-30 08:43:43.673 T:1000     info <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 540, fps: 25.000, 3D: false
    2022-04-30 08:43:43.682 T:1000     info <general>: Display resolution ADJUST : 1920x1080 @ 25.000000 Hz (46) (weight: 0.000)
    2022-04-30 08:43:43.734 T:595      info <general>: VideoPlayer: OnLostDisplay received
    2022-04-30 08:43:43.734 T:595   warning <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
    2022-04-30 08:43:43.734 T:595   warning <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
    2022-04-30 08:43:43.734 T:595      info <general>: GLES: Maximum texture width: 2048
    2022-04-30 08:43:43.736 T:1000     info <general>: Creating video codec with codec id: 27
    2022-04-30 08:43:43.744 T:1000     info <general>: CDVDVideoCodecDRMPRIME::Open - using decoder V4L2 mem2mem H.264 decoder wrapper
    2022-04-30 08:43:43.844 T:595      info <general>: VideoPlayer: OnResetDisplay received
    2022-04-30 08:43:43.949 T:595   warning <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
    2022-04-30 08:43:43.950 T:595   warning <general>: CDVDMessageQueue(video)::Put MSGQ_NOT_INITIALIZED
    2022-04-30 08:43:43.950 T:595   warning <general>: CDVDMessageQueue(audio)::Put MSGQ_NOT_INITIALIZED
    2022-04-30 08:43:43.991 T:1000     info <general>: Creating video thread
    2022-04-30 08:43:43.992 T:1010     info <general>: running thread: video_thread
    2022-04-30 08:43:43.996 T:1000     info <general>: Opening stream: 1002 source: 256
    2022-04-30 08:43:43.997 T:1000     info <general>: Finding audio codec for: 86018
    2022-04-30 08:43:44.029 T:1000     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
    2022-04-30 08:43:44.029 T:1000     info <general>: Creating audio thread
    2022-04-30 08:43:44.030 T:1011     info <general>: running thread: CVideoPlayerAudio::Process()
    2022-04-30 08:43:44.043 T:1011     info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
    2022-04-30 08:43:44.046 T:1011     info <general>: Skipped 1 duplicate messages..
    2022-04-30 08:43:44.046 T:1011     info <general>: Creating audio stream (codec id: 86018, channels: 2, sample rate: 48000, no pass-through)
    2022-04-30 08:43:44.107 T:609      info <general>: CActiveAESink::OpenSink - initialize sink
    2022-04-30 08:43:44.289 T:609      info <general>: CAESinkALSA::Initialize - Attempting to open device "@:CARD=vc4hdmi,DEV=0"
    2022-04-30 08:43:44.312 T:609      info <general>: CAESinkALSA - Unable to open device "sysdefault:CARD=vc4hdmi" for playback
    2022-04-30 08:43:44.318 T:609      info <general>: CAESinkALSA - Unable to open device "default:CARD=vc4hdmi" for playback
    2022-04-30 08:43:44.319 T:609      info <general>: CAESinkALSA::Initialize - Opened device "front:CARD=vc4hdmi,DEV=0"
    2022-04-30 08:43:44.320 T:609      info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
    2022-04-30 08:43:44.320 T:609     error <general>: CAESinkALSA::InitializeHW - Unable to find a suitable output format
    2022-04-30 08:43:44.320 T:609      info <general>: CAESinkALSA::Initialize - Attempting to open device "iec958"
    2022-04-30 08:43:44.327 T:609      info <general>: CAESinkALSA::Initialize - Opened device "iec958:AES0=0x04,AES1=0x82,AES2=0x00,AES3=0x02"
    2022-04-30 08:43:44.329 T:609      info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
    2022-04-30 08:43:44.329 T:609      info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
    2022-04-30 08:43:44.644 T:595      info <general>: Loading skin file: VideoFullScreen.xml, load type: KEEP_IN_MEMORY
    2022-04-30 08:43:44.653 T:595      info <general>: [WHITELIST] Searching the whitelist for: width: 720, height: 540, fps: 25.000, 3D: false
    2022-04-30 08:43:44.664 T:595      info <general>: Display resolution ADJUST : 1920x1080 @ 25.000000 Hz (46) (weight: 0.000)
    2022-04-30 08:43:44.782 T:1000     info <general>: AddOnLog: inputstream.adaptive: PosTime (100744)
    2022-04-30 08:43:44.954 T:1000     info <general>: AddOnLog: inputstream.adaptive: Seek time (100.7) for stream: 196609 continues at 100.8 (PTS: 29249864533)
    2022-04-30 08:43:44.954 T:1000    error <general>: AddOnLog: inputstream.adaptive: SeekTime: Cannot get the stream sample reader
[... cut ...]
    2022-04-30 08:43:52.692 T:608   warning <general>: ActiveAE - large audio sync error: 1022.663990
    2022-04-30 08:43:52.729 T:608   warning <general>: ActiveAE - large audio sync error: 1031.017323
    2022-04-30 08:43:52.766 T:608   warning <general>: ActiveAE - large audio sync error: 1039.359510
    2022-04-30 08:43:52.804 T:608   warning <general>: ActiveAE - large audio sync error: 1046.710603
    2022-04-30 08:43:52.839 T:608   warning <general>: ActiveAE - large audio sync error: 1057.026646`

[dmesg](https://paste.opensuse.org/65040553)

neo1973, 2022-05-02:
I can confirm the issue. CEC also stopped working after 5.15.34.

popcornmix, 2022-05-03:
It looks like it started with https://github.com/raspberrypi/rpi-firmware/commit/1edb1cc63bfe2987d170abf1ff62f9e7f2068c95
(i.e. the clock framework improvements).
I see cec failure on pi3 but not pi4.

neo1973, 2022-05-03:
> @neo1973 are you also on arch and rpi3?

Yes, I have the same setup as the other two commenters.

popcornmix, 2022-05-06:
@pejobo @neo1973 @coffeinflash can you test [this firmware](https://drive.google.com/uc?id=1SpiqT1W1N96rnQRgFsx6WRU5olvGR22s&export=download) and confirm if it fixes issue?

coffeinflash, 2022-05-06:
@popcornmix 
Thank you! Works for me :-)

neo1973, 2022-05-07:
Today I did full update of my Raspberry Pi and the issue is gone. I downgraded the kernel and firmware packages but can't reproduce the error anymore. I have no idea what change fixed it :(

popcornmix, 2022-05-07:
The stable kernel (5.15.32) doesn't have the issue.
The issue only for appeared for users running the testing rpi-update kernel.
And it has now been fixed with latest rpi-update kernel.

neo1973, 2022-05-07:
I'm on 5.15.36. But my mistake was to downgrade the raspberrypi-firmware package and not raspberrypi-bootloader which contains the relevant files. After downgrading raspberrypi-bootloader the issue was back. The current raspberrypi-bootloader on Arch contains the "firmware: arm_loader_dvfs: Support CLOCK_HDMI as boostable clock" commit and with that everything is back to normal :+1:

popcornmix, 2022-05-10:
@pejobo is this resolved for you with latest firmware?

pejobo, 2022-05-10:
I could try this weekend.

pejobo, 2022-05-10:
I was able to test earlier as expected. Works for me, too.
Thank you for the quick fix 👍

closed 2022-05-11 by popcornmix