raspberrypi/linux #5017: HDMI display doesn't come up if not booted with HDMI connected

lategoodbye, 2022-05-01:
### Describe the bug

Hi, not sure this ever worked before. If i boot my Raspberry Pi 3 B Plus without HDMI attached and then connect the HDMI display afterwards the displays doesn't comes up and stays black. Looking at the interrupts shows that vc4 hvs (hotplug interrupt?) doesn't increase on plugin and out.

Expected behavior:
The HDMI display comes up like if the board has booted with HDMI connected.

### Steps to reproduce the behaviour

- do not connect the HDMI cable to the Raspberry Pi 3 B Plus
- boot Raspberry Pi 3 B Plus
- connect the HDMI cable of the display

There are not modifications to the config.txt (same as shipped by April 2022 release)

### Device (s)

Raspberry Pi 3 Mod. B+

### System

System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage4

Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
Revision	: a020d3
Serial		: 00000000377bb917
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 24 2022 13:20:54 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

### Logs

cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
 50:        411          0          0          0  ARMCTRL-level   1 Edge      3f00b880.mailbox
 51:       1148          0          0          0  ARMCTRL-level   2 Edge      VCHIQ doorbell
 67:          0          0          0          0  ARMCTRL-level  42 Edge      vc4
 68:          0          0          0          0  ARMCTRL-level  43 Edge      3f004000.txp
 75:        346          0          0          0  ARMCTRL-level  50 Edge      DMA IRQ
 77:       8085          0          0          0  ARMCTRL-level  52 Edge      DMA IRQ
 78:          0          0          0          0  ARMCTRL-level  53 Edge      DMA IRQ
 89:      48392          0          0          0  ARMCTRL-level  64 Edge      dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 90:          0          0          0          0  ARMCTRL-level  65 Edge      vc4 hvs
 97:          0          0          0          0  ARMCTRL-level  72 Edge      vc4 hdmi cec
 99:          0          0          0          0  ARMCTRL-level  74 Edge      vc4 crtc
102:          0          0          0          0  ARMCTRL-level  77 Edge      vc4 crtc
103:          0          0          0          0  ARMCTRL-level  78 Edge      vc4 crtc
110:        104          0          0          0  ARMCTRL-level  85 Edge      3f805000.i2c
113:        770          0          0          0  ARMCTRL-level  88 Edge      mmc0
[... 95 lines cut ...]
[   10.561546] Registered IR keymap rc-cec
[   10.561806] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   10.562095] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input3
[   10.571061] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.571994] usbcore: registered new interface driver brcmfmac
[   10.581760] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.bin failed with error -2
[   10.667205] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   10.671498] Registered IR keymap rc-cec
[   10.671755] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   10.672051] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input4
[   11.127366] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.127544] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.179635] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[   11.298755] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   11.304596] Registered IR keymap rc-cec
[   11.304915] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   11.305412] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input5
[   11.314795] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   11.315369] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[   11.315990] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   11.317393] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   11.317971] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   11.318492] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   11.337884] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   11.338205] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes

popcornmix, 2022-05-11:
@lategoodbye do you agree with my analysis?
Does hotplug work for you when mutter is used and fail to work when mutter is not used?

lategoodbye, 2022-05-12:
@popcornmix Sorry for the late reply, but i didn't receive a notification. Can you please tell which kernel version you used? I noticed that Maxime recently fixed a related issue 82f448e685fed020305d85169523a8633601258a, but my version didn't include this change. I will try to retest at the weekend.

popcornmix, 2022-05-12:
I was using latest rpi-update kernel (would have been rpi-5.15.36, but I don't think newer ones behave any differently).

There was a small window when 5.15.34 would hang on boot when booted without hdmi on Pi0-3, but for me that was fatal (I couldn't ssh in or log in through uart).

The mutter issue I've described doesn't interfere with uart/ssh access - just whether hdmi gets enabled on a hotplug if it wasn't enabled previously.

EdiDD, 2022-05-13:
This vc4 kernel driver is broken. It hangs (and there is no video signal) after some time if hdmi is switched to another source producing errors:

```

[drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:89:crtc-3] commit wait timed out
[drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
[drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:79:plane-3] commit wait timed out
[drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
------------[ cut here ]------------
WARNING: CPU: 0 PID: 47497 at drivers/gpu/drm/vc4/vc4_hdmi.c:1552 vc4_hdmi_encoder_post_crtc_enable+0x109c/0x11c8 [vc4]
Modules linked in: xt_nat xt_tcpudp veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink xt_addrtyp>
CPU: 0 PID: 47497 Comm: kworker/0:0 Tainted: G        WC        5.15.32-v8+ #1538
Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Workqueue: events output_poll_execute [drm_kms_helper]
pstate: 400000c5 (nZcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
pc : vc4_hdmi_encoder_post_crtc_enable+0x109c/0x11c8 [vc4]
lr : vc4_hdmi_encoder_post_crtc_enable+0xc08/0x11c8 [vc4]
Call trace:
 vc4_hdmi_encoder_post_crtc_enable+0x109c/0x11c8 [vc4]
 vc4_crtc_atomic_enable+0x3d8/0x4b0 [vc4]
 drm_atomic_helper_commit_modeset_enables+0x210/0x258 [drm_kms_helper]
 vc4_atomic_commit_tail+0x3c4/0x730 [vc4]
 commit_tail+0xac/0x1a8 [drm_kms_helper]
 drm_atomic_helper_commit+0x168/0x378 [drm_kms_helper]
 drm_atomic_commit+0x54/0x68 [drm]
 drm_client_modeset_commit_atomic.isra.16+0x1ec/0x290 [drm]
 drm_client_modeset_commit_locked+0x68/0x1a8 [drm]
 drm_client_modeset_commit+0x38/0x60 [drm]
 __drm_fb_helper_restore_fbdev_mode_unlocked+0xd4/0x118 [drm_kms_helper]
 drm_fb_helper_set_par+0x44/0x80 [drm_kms_helper]
 drm_fb_helper_hotplug_event.part.31+0xb8/0xe8 [drm_kms_helper]
 drm_fbdev_client_hotplug+0x44/0x1c0 [drm_kms_helper]
 drm_client_dev_hotplug+0x80/0xd0 [drm]
 drm_kms_helper_hotplug_event+0x3c/0x50 [drm_kms_helper]
 output_poll_execute+0x1e0/0x230 [drm_kms_helper]
 process_one_work+0x1f8/0x4a0
 worker_thread+0x50/0x480
 kthread+0x140/0x158
 ret_from_fork+0x10/0x20
```

kernel: **5.15.32-v8**
/boot/config.txt :
dtoverlay=**vc4-kms-v3d**
max_framebuffers=2
hdmi_force_hotplug=1

**vc4-fkms-v3d** is not working well with kodi 19. Is there any other stable video driver (as temporary replacement) working with kodi 19 ?

popcornmix, 2022-05-13:
@EdiDD your issue doesn't sound at all related to this one.
It seems much closer to https://github.com/raspberrypi/linux/issues/4962

lategoodbye, 2022-05-14:
@popcornmix First i upgrade to 5.15.38 and the issue still occured, then i changed `/usr/bin/startlxde-pi` as you suggested and it worked. Thanks, would be great to get this fixed in Raspbian OS.

popcornmix, 2023-01-24:
Okay to close this as it's not fixable in firmware?
I have been told that this won't be an issue in RPiOS Bookworm.

lategoodbye, 2023-01-24:
Sorry, i missed this.

closed 2023-01-24 by lategoodbye