raspberrypi/linux #5018: Please enable module at25

Broezel, 2022-05-02:
### Describe the bug

Please enable module at25 (CONFIG_EEPROM_AT25).
I'm using this module successfully together with the Adafruit SPI Non-Volatile FRAM Breakout - 64Kbit / 8KByte
(https://www.adafruit.com/product/1897). Thanks in advance.

### Steps to reproduce the behaviour

Nothing

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage4
Apr 13 2022 14:51:05 
Copyright (c) 2012 Broadcom
version c073992a577d160d8c30bfcf746d2208951c0537 (clean) (release) (start)
Linux rpi21 5.15.35-v7+ #1547 SMP Tue Apr 26 11:47:07 BST 2022 armv7l GNU/Linux

pelwell, 2022-05-03:
That's now enabled in rpi-5.15.y and newer branches. There should be a build available from rpi-update in the next few days.

popcornmix, 2022-05-10:
@Broezel okay to close?

Broezel, 2022-05-10:
Hi. YES: Thank you very much.

Mit freundlichen Grüßen / with best regards

Olaf Klein
IT System Administrator

[cid:Logo-Vosla-negativ_fbg_78f2d7b3-6795-4ba3-b84c-b8dfe450c316.bmp]<www.vosla.com>

vosla GmbH
L.-F.-Schönherr-Straße 15, 08523 Plauen, Germany

Tel: +49 3741 396-265 Mobile: +49 175 7238848 Fax: +49 3741 396-508
***@***.***   www.vosla.com

Managing Director: Carsten Riedel
Corporate Registration: Chemnitz HRB 27180
Our privacy policy can be found at: www.vosla.com/datenschutz<http://www.vosla.com/datenschutz>

Warum dürfen wir Ihnen per eMail schreiben?
Hiermit informieren wir Sie gemäß Art. 13 (1) bzw. Art. 14 (3) b DSGVO darüber, dass wir die Daten, die der Kommunikation mit Ihnen zugrunde liegen, ausschließlich zu Kommunikationszwecken auf der Grundlage von Art. 6 (1) a oder f DSGVO verarbeiten. Die erforderliche Datenschutzerklärung finden sie über diesen Link <https://www.vosla.com/datenschutz/> auf unserer Webseite im Kapitel Datenschutz. Da in der Vergangenheit zwischen Ihnen oder Ihrem Unternehmen und vosla bzw. ihren Repräsentanten bereits eine geschäftliche Verbindung bestanden hat oder der Kommunikation Ihre Einwilligung zugrunde liegt, stellt diese eMail keinen Verstoß gegen §7 Gesetz gegen unlauteren Wettbewerb dar.

Why are we allowed to email you?
In accordance with Article 13 (1) or Article 14 (3) b GDPR, we hereby inform you that we process the data underlying communication with you solely for communication purposes on the basis of Article 6 (1) (a) or f GDPR. The required privacy policy can be found via this link <https://www.vosla.com/en/privacypolicy/> on our website in the Privacy chapter. Since in the past there has already been a business connection between you or your company and vosla or its representatives, or if the communication is based on your consent, this e-mail does not con-stitute a violation of § 7 Gesetz gegen den unlauteren Wettbewerb (UWG).

closed 2022-05-10 by popcornmix