raspberrypi/linux #5024: Pi4 cannot stay connected to Miktrotik hap2 AC router - lP address expires and Pi cannot rejoin until reboot

darrylgibbs, 2022-05-07:
### Describe the bug

Pi4 refuses to stay connected to Mikrotik router beyond first connection.  The IP address does not autorenew.  It just counts down and at 0s, it falls off the network and cannot reconnect until a full system reboot.

### Steps to reproduce the behaviour

I've tried 2x Mikrotik hap2 ACs running the latest RouterOS 7.2.3 Routerboard firmware and RouterOS 7.3 packages, on all interfaces.  Connect the Pi to the network, OS installed on an SSD connected via USB3.0 interface.  Happens with a fresh install of RPi OS Bullseye.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

Mar 24 2022 13:19:26
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

Linux homelab 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

### Additional context

This started immediately after changing to the latest 64bit Bullseye OS.  It also happens on the new Ubuntu 22.04 LTS for ARM.  That is the only appreciable change.

Bot123123, 2022-05-09:
What about the other APs? is it reproducible only with Microtik or with any other WiFi AP, e.g. based on the OpenWRT?

seamusdemora, 2022-05-11:
@darrylgibbs :
> I've now set a static IP on the Pi itself, and after 2 days, I'm seeing no issues so far. Until now, the static IP assignment was done on the router, as I always have done previously.

Are you using `dhcpcd`?
If so, how did you set the static configuration in `/etc/dhcpcd.conf`?

darrylgibbs, 2022-05-11:
So originally had the Mikrotik router do the static IP reservation for me, and I left the Pi as defaults.  This is how I've assigned static IPs on my network since getting the router without issues.

I have now set the static IP on the Pi itself using `/etc/dhcpcd.conf`.  See my additions below:

```interface eth0
static ip_address=192.168.88.250/24
static routers=192.168.88.1
static domain_name_servers=8.8.8.8 fd51:42f8:caae:d92e::1```

seamusdemora, 2022-05-12:
@darrylgibbs : It seems you have things configured in a workable state now. FWIW, here's my take on what happened:

In your original post:
> The IP address does not autorenew. It just counts down and at 0s, it falls off the network and cannot reconnect until a full system reboot.

Right - that is as it should be AFAIK. A static IP doesn't use _autorenewal_ because it's **static**. 

If I followed the sequence correctly, you 
   1. Declared a static IP in your router (which I assume is also your DHCP server) for your RPi, but did not add the static IP config in `dhcpcd.conf`.  

   The problem with that is `dhcpcd.conf` - by default - is configured to request a (dynamic) IP address; see `man dhcpcd.conf`. And so, absent a `static` configuration, `dhcpcd` followed its default configuration, and requested a dynamic IP address from your DHCP server. I do not know how your DHCP server handled this request for IP ADDR assignment since it already had a reservation for the RPi, but the fact that your RPi networked successfully for a period before receiving a timeout message suggests that your DHCP server initially provided a response to `dhcpcd`, but ignored its subsequent renewal requests. `dhcpcd` continued to send these requests because it was still under its default config as a dynamic client. 

   2. When you added the `static` config to `dhcpcd.conf`, `dhcpcd` stopped sending renewal requests, and so for now at least - everyone is happy? 

And now comes the part of the answer known as `"unsolicited advice"`: You should carefully consider your need for a static IP address. In my experience, they are very rarely needed - especially on a private IP LAN. I initiated a [recent discussion (#2496)](https://github.com/raspberrypi/documentation/issues/2496#event-6342337937) on how to communicate this via the "official documentation", but the maintainers opted for [*less specific* form of the documentation.](https://www.raspberrypi.com/documentation/computers/configuration.html#static-ip-addresses) This of course is only my opinion. 

If you read `man dhcpcd.conf` carefully, under the heading `static value`, you will note that the author of `dhcpcd` advises specifically against the use of a `static` configuration:  

>For IPv4, you should use the inform ipaddress option instead of setting a static address.  

That is excellent advice from a purely networking perspective. Unfortunately, a "bug" crept into the `dhcpcd` code that created a **_log event tsunami_** when the `inform` option was used. That bug has been found and fixed, but has not yet been released into distribution. If you're interested in using the `inform` option _sans log tsunami_, I have posted a [recipe on this GitHub page](https://github.com/seamusdemora/PiFormulae/blob/master/FixDhcpcd.md) that will walk you through the patching process, and allow you to build a *de-bugged* `dhcpcd`. It's a bit of a hack as I'm not a package maintainer, but it does work. 

So - that's it. AFAIK, your current configuration should work indefinitely - as long as nothing in your network configuration changes. That's another downside to the `static` config - it's brittle. If your DNS server, or your Gateway router change address, you'll have to manually repair that.

darrylgibbs, 2022-05-12:
[image]

I agree with you, static IPs shouldn't renew, but if you look at the image from my Router, you will see what I mean.  Those IP's with a **D** at the beginning are dynamic, and those without are static.  But ALL have (at the far right) a **bound** (meaning it's verified as on the network and reachable) tag, and has a lease countdown from 10:00 which auto resets at 05:00 for all IP types.  The Pi just counted down to 0:00 an then disappeared from the list.

Answering your questions:

1.  I did assign the static IP on the router which is also my DHCP server.  The Pi itself was set to receive a dynamic address.  Basically default network functionality.  This is how I've always done things on all my devices.  The device is just set to receive a dynamic address, but if I wanted a static IP, I just told the router to give **X** MAC a static IP.  Further, the **adblock** device (in the image) is a RPi Zero v1, using an ethernet adapter.  I gave that a static IP at the router level, and it works flawlessly. 
2. Yeah so far it's all rainbows and butterflies.  Working fine.

I've used static IPs for this Pi4 since it is my home server running a bunch of different services, (samba shares, a variety of docker containers etc) so I want the address as static so that things wouldn't just break if one day a new IP address was given to the device.  But, even dynamic addresses were not be renewed, so I really don't know what is going on.

I think I will wait for the patch to come from official sources. I know enough to see there is an issue and most likely find what it is, but I' daft enough that I will likely break something instead of fixing it.

Thanks for your help!

closed 2022-05-12 by darrylgibbs