raspberrypi/linux #5029: Bullseye (Lite) for Pi Zero W 1.1 appears to use graphical.target as the default target in systemd, which will never start

Makeshift, 2022-05-11:
### Describe the bug

A fresh install of Bullseye (Lite) for Pi Zero W 1.1 results in a systemd default target of `graphical.target`, which is also a symlink for `multi-user.target`. This means that many services never start, as `graphical.target` will never be active.

```
pi@raspberrypi:~ $ systemctl get-default
graphical.target
pi@raspberrypi:~ $ systemctl is-active graphical.target 
inactive
pi@raspberrypi:~ $ systemctl is-active multi-user.target
inactive
pi@raspberrypi:~ $ systemctl is-active default.target 
inactive
```

### Steps to reproduce the behaviour

1) Use the [Raspberry Pi Imager](https://www.raspberrypi.com/software/) to install Bullseye (Lite) on a Pi Zero W 1.1.
2) SSH to the machine, `sudo apt update` and `sudo apt upgrade` (I restarted at this point just for giggles, but that probably isn't necessary)
3) Run `systemctl get-default`, it will output `graphical.target`.
4) Run `systemctl is-active graphical.target`, it will output `inactive`
5) Run `systemctl is-active multi-user.target`, it will output `inactive`

This means that anything depending on `multi-user.target` (which is functionally the default for most simple services, eg. https://github.com/pimoroni/grow-python/issues/18) will never be able to start.

### Device (s)

Raspberry Pi Zero W / WH

### System

```
System Information
------------------

Raspberry Pi Zero W Rev 1.1 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage2

Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux
Revision	: 9000c1
Serial		: 00000000e1f61231
Model		: Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 24 2022 13:20:54 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root      122613804 1549228 116047180   2% /
devtmpfs           88616       0     88616   0% /dev
tmpfs             219904       0    219904   0% /dev/shm
tmpfs              87964     668     87296   1% /run
tmpfs               5120       4      5116   1% /run/lock
[... 588 lines cut ...]
[   37.237642] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops [vc4])
[   37.249125] Registered IR keymap rc-cec
[   37.249457] rc rc0: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0
[   37.249774] input: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0/input4
[   37.888700] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops [vc4])
[   37.929622] Registered IR keymap rc-cec
[   37.929935] rc rc0: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0
[   37.930261] input: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0/input5
[   37.966379] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops [vc4])
[   37.967292] vc4-drm soc:gpu: bound 20004000.txp (ops vc4_txp_ops [vc4])
[   37.968010] vc4-drm soc:gpu: bound 20206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   37.968682] vc4-drm soc:gpu: bound 20207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   37.969354] vc4-drm soc:gpu: bound 20807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   37.969931] vc4-drm soc:gpu: bound 20c00000.v3d (ops vc4_v3d_ops [vc4])
[   38.042990] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   38.043556] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   41.105430] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   41.105474] Bluetooth: BNEP filters: protocol multicast
[   41.105509] Bluetooth: BNEP socket layer initialized
[   41.315191] NET: Registered PF_ALG protocol family
[   44.132924] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   45.756731] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   47.299563] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   48.491238] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
```

XECDesign, 2022-05-13:
> results in a systemd default target of `graphical.target`, which is also a symlink for `multi-user.target`

Not sure what you mean by that bit. `graphical.target` is not a symlink for `multi-user.target`

> This means that many services never start, as `graphical.target` will never be active.

That's not quite how that works. Everything that should start will start either way. In addition to services which would start with `multi-user.target`, it will also start services that are wanted by `graphical.target`. Since this is a lite image, there are no such services.

> There shouldn't be a desktop, this is lite!

And there's isn't one. Having `graphical.target` as the default even when no desktop is installed is consistent with Debian.

The issue you actually have is that `userconfig.service` is running. If you've just flashed using rpi-imager, [you wouldn't be able to ssh in](https://rptl.io/newuser).

If you've set the username and password and enabled ssh in rpi-imager,  userconfig shouldn't be running.

I suspect there were additional steps taken to get the system in this state which were not mentioned in the "steps to reproduce this issue" section.

Makeshift, 2022-05-13:
> Not sure what you mean by that bit. graphical.target is not a symlink for multi-user.target

Apologies, this was a mistake on my part. This isn't the case (and I've edited it out of my original issue)

> The issue you actually have is that userconfig.service is running. If you've just flashed using rpi-imager, [you wouldn't be able to ssh in](https://rptl.io/newuser).

> If you've set the username and password and enabled ssh in rpi-imager, userconfig shouldn't be running.

You are completely correct, thank you for this. Proven by:

```
● userconfig.service - User configuration dialog
     Loaded: loaded (/lib/systemd/system/userconfig.service; enabled; vendor preset: enabled)
     Active: activating (start) since Wed 2022-05-11 21:21:45 BST; 1 day 18h ago
   Main PID: 369 (userconf-servic)
      Tasks: 2 (limit: 415)
        CPU: 21.327s
     CGroup: /system.slice/userconfig.service
             ├─369 /bin/sh -e /usr/lib/userconf-pi/userconf-service
             └─927 whiptail --inputbox Please enter new username: 20 60 <----------------------------------

May 11 21:21:45 raspberrypi systemd[1]: Starting User configuration dialog...
```

This issue was caused by me incorrectly assuming that because I provided an SSH key, I wouldn't need to enter a username and password (as the key would just be applied to the default `pi` account). This machine is headless, so I didn't see any prompt, and I couldn't see that it was waiting specifically on `userconfig` to finish.

The only difference to my creation steps in the original post would be:

1) Use the [Raspberry Pi Imager](https://www.raspberrypi.com/software/) to install Bullseye (Lite) on a Pi Zero W 1.1, with advanced options set as follows:
    ✔️ Enable SSH - Allow public-key-authentication only (with a key provided)
    ❌ Set username and password (This is why userconfig is running and waiting for input)
    ✔️ Configure wireless LAN

Apologies for my misdiagnosis, and thank you for the help. 

I would perhaps request a little note in the advanced options saying that bootup will not fully complete on a headless setup without a user/pass set, but this is the wrong place for that.

closed 2022-05-13 by Makeshift

XECDesign, 2022-05-13:
I think you might have an older version of rpi-imager. v1.7.2 shouldn't let you do that.

The next OS release will inform you of what you need to know if you enable ssh, but don't configure the user.

Makeshift, 2022-05-13:
> I think you might have an older version of rpi-imager. v1.7.2 shouldn't let you do that.

I am running 1.7.2 on two separate computers. On one of them it acts as you describe and if I tick "Enable SSH" it automatically ticks "Set username and password" and won't let me untick it.

On the other machine, (the one I happened to create the image for this pi from 🤦 ) as you can see in this screenshot, "Enable SSH" is ticked, and "Set username and password" is unticked, which shouldn't be possible:

[image]

I can freely tick and untick it on the above screenshot, and yet can't replicate it on my other PC. Is this perhaps some weird config leftover from upgrading from a previous version? This is really weird!

Makeshift, 2022-05-13:
I'm totally satisfied to call this a completely unique issue that nobody else will ever have (and therefore solved), and just burn that entire computer. At this point I'm just curious how that can even _happen_

pelwell, 2022-05-13:
>  Is this perhaps some weird config leftover from upgrading from a previous version?

That sounds plausible. You could try an uninstall (perhaps deleting the directory as well) before torching your PC, but I don't want to spoil your fun.

XECDesign, 2022-05-13:
Yeah, if you had settings saved to persist across rpi-imager sessions, then upgraded rpi-imager, it might just read the old conf file and not complain.