raspberrypi/linux #5031: Reading encrypted DVDs in drives attached via USB hangs indefinitely or is slow

meerkatmcr, 2022-05-12:
### Describe the bug

When trying to read encrypted DVDs (i.e. ones with CSS applied to them) in drives attached over USB, either the read hangs indefinitely, or takes a surprisingly long time to return (mounting the filesystem takes ~30-40 seconds, for example). Only one read seems to hang like this, and after mounting, the filesystem can be read normally. When playing such a disc (using mplayer), it works fine after an initial delay.

This behaviour does not appear when reading unencrypted DVDs, or encrypted Blu-Ray discs. I don't think I've got any CD-ROMs or DVD-ROMs to test with (all DVDs I have tried have been DVD Video discs), though I can try an audio CD if that will help. I also don't have any unencrypted Blu-Ray discs.

I have two optical drives in external enclosures:
- a Hitachi/LG GH24 DVD burner inside a LINDY model 43138 enclosure
- a Hitachi/LG BH16 Blu-Ray burner inside an OWC "Mercury Pro" model OWCMR3UKIT enclosure

both connected via USB3 to my Pi 4.

### With the Blu-Ray drive
 - reading Blu-Ray discs (I have only one, and it's encrypted) works fine
 - reading unencrypted DVDs also works fine
 - reading encrypted DVDs hangs seemingly forever (doesn't seem to matter what you are doing; only that you're trying to access the disc)
 - the hang seems to happen inside a `read()` system call
 - when the read is hanging, the drive's eject button refuses to work
 - when the read is hanging, I *think* running `eject` just causes that to hang as well
 - the only way to terminate the hanging process is to remove power from the drive, at which point the hanging system call returns with an error

### With the DVD drive
 - obviously, Blu-Ray discs cannot be read
 - reading unencrypted DVDs works fine
 - reading encrypted DVDs works, but some reads take a very long time: mounting the filesystem takes over 40 seconds, most of which is spent inside a `read()` system call

### Steps to reproduce the behaviour

Insert an encrypted DVD (i.e. basically any commercial DVD Video disc) into an optical drive attached over USB to the Pi. Attempt to mount the filesystem on the disc, or otherwise read from it.

### Device (s)

Raspberry Pi 4 Mod. B

### System

- Kernel: 5.15.38-v8+ built from commit 3126b2f03a3b64b583837f7fe26486f236b8358f of this repo.
- firmware: e5a963efa66a1974127860b42e913d2374139ff5
- OS: Gentoo Linux (rolling release, but up to date as of this morning)

### Logs

dmesg output when mounting an encrypted DVD in the DVD drive:
```
[ 1145.983720] usb 2-1.4: new SuperSpeed USB device number 10 using xhci_hcd
[ 1146.012877] usb 2-1.4: New USB device found, idVendor=1234, idProduct=5678, bcdDevice=19.05
[ 1146.012898] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1146.012905] usb 2-1.4: Product: VLI Product String
[ 1146.012911] usb 2-1.4: Manufacturer: VLI manufacture String
[ 1146.012916] usb 2-1.4: SerialNumber: 0000000000007478
[ 1146.016310] usb-storage 2-1.4:1.0: USB Mass Storage device detected
[ 1146.016787] scsi host3: usb-storage 2-1.4:1.0
[ 1147.029168] scsi 3:0:0:0: CD-ROM            HL-DT-ST DVDRAM GH24NSD5  LV00 PQ: 0 ANSI: 0
[ 1147.036396] sr 3:0:0:0: Power-on or device reset occurred
[ 1147.050215] sr 3:0:0:0: [sr0] scsi3-mmc drive: 48x/48x writer dvd-ram cd/rw xa/form2 cdda tray
[ 1147.056808] sr 3:0:0:0: Attached scsi CD-ROM sr0
[ 1155.684444] sr 3:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=0s
[ 1155.684464] sr 3:0:0:0: [sr0] tag#0 Sense Key : 0x5 [current] 
[ 1155.684471] sr 3:0:0:0: [sr0] tag#0 ASC=0x6f ASCQ=0x3 
[ 1155.684479] sr 3:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 07 ff fc 00 00 02 00
[ 1155.684486] print_req_error: 46 callbacks suppressed
[ 1155.684489] blk_update_request: I/O error, dev sr0, sector 2097136 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[ 1155.713719] sr 3:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=0s
[ 1155.713738] sr 3:0:0:0: [sr0] tag#0 Sense Key : 0x5 [current] 
[ 1155.713745] sr 3:0:0:0: [sr0] tag#0 ASC=0x6f ASCQ=0x3 
[ 1155.713753] sr 3:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 07 ff fc 00 00 02 00
[ 1155.713759] blk_update_request: I/O error, dev sr0, sector 2097136 op 0x0:(READ) flags 0x0 phys_seg 8 prio class 0
[ 1155.713771] buffer_io_error: 94 callbacks suppressed
[ 1155.713775] Buffer I/O error on dev sr0, logical block 2097136, async page read
[ 1155.713784] Buffer I/O error on dev sr0, logical block 2097137, async page read
[... 275 lines cut ...]

for comparison, a successful mount of a Blu-Ray disc in the Blu-Ray drive:
```
[  813.435000] usb 2-1.4: new SuperSpeed USB device number 7 using xhci_hcd
[  813.460242] usb 2-1.4: New USB device found, idVendor=1e91, idProduct=de2c, bcdDevice= 1.00
[  813.460262] usb 2-1.4: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  813.460268] usb 2-1.4: Product: Mercury Pro Optical
[  813.460274] usb 2-1.4: Manufacturer: Other World Computing
[  813.460279] usb 2-1.4: SerialNumber: 21052488101D
[  813.464519] usb-storage 2-1.4:1.0: USB Mass Storage device detected
[  813.465295] scsi host3: usb-storage 2-1.4:1.0
[  814.472061] scsi 3:0:0:0: CD-ROM            HL-DT-ST BD-RE  BH16NS40  1.05 PQ: 0 ANSI: 0
[  814.572192] sr 3:0:0:0: Power-on or device reset occurred
[  814.632984] sr 3:0:0:0: [sr0] scsi3-mmc drive: 0x/48x writer dvd-ram cd/rw xa/form2 cdda tray
[  814.673083] sr 3:0:0:0: Attached scsi CD-ROM sr0
[  844.678289] UDF-fs: INFO Mounting volume 'BRIDGET_JONES_G51', timestamp 2009/08/10 19:37 (103c)
```

### Additional context

I have not yet tried swapping the drives between enclosures.

I have also not tried either drive in a machine with native SATA, rather than in an enclosure over USB.

Some time ago [a similar issue](https://bugzilla.redhat.com/show_bug.cgi?id=1822948) was reported against Fedora 31, and doesn't seem to have been solved before that release went end-of-life. Notably, the enclosure the user reported difficulties with sounds like the same to one of mine, and the symptoms are identical.

JamesH65, 2022-05-12:
Have you the equipment to test this on a x86 linux device? I get similar symptoms on Ubuntu on x86 so it could be a Linux wide issue rather than Pi related.

meerkatmcr, 2022-05-12:
Behaviour is *not* entirely the same on the x86 machine.

Note that Linux Mint automounts discs when they are inserted, so I can't strace the mount command as I was before. Does seem to be mounting normally and not through FUSE or something, though.

This testing was done through a USB2 port, since there wasn't a readily available USB3 port. The machine reports its kernel version as `5.14.0-1036-oem #40-Ubuntu`.

### In the Blu-Ray drive
 - all three of the test discs I used when testing with the Pi were readable, and with no noticeable delays such as found on the Pi
 - there are many fewer I/O errors reported in dmesg (sometimes there will be no errors at all; sometimes two, no matter whether the mount ultimately succeeds or not)
 - the `blk_update_request` I/O errors that do appear all have lines with them saying `Sense Key: "Illegal Request"` and `Add. Sense: Read of scrambled sector without authentication`
 - the encrypted DVD I used for testing is part of a six-disc release, so I tried all the discs in the set:
   - disc 1 was readable with a delay
   - disc 2 was readable immediately
   - disc 3 was not readable
   - disc 4 was readable with a delay 
   - disc 5 was readable immediately
   - disc 6 was not readable

### In the DVD drive
 - the unencrypted disc was readable immediately
 - all of the encrypted discs were readable, but with delays of varying length (all roughly similar to those found on the Pi though)
 - dmesg output contained similar I/O errors to those mentioned above for the Blu-Ray drive, though more of them

I can rearrange peripherals to make a USB3 port available on that machine and test through that, if it's necessary.

Is it worth swapping the drives between the two enclosures and seeing if that changes anything?

meerkatmcr, 2022-06-01:
Given that this is probably not specific to the RPi kernel, I will close this and raise it upstream. Thank you for your time.

closed 2022-06-01 by meerkatmcr