raspberrypi/linux #5032: 5.18 Unicam seems to have issues

6by9, 2022-05-12:
### Describe the bug

IMX290 native driver should support 1920x1080@60fps, and that works fine on 5.15.38.
It doesn't work on 5.18.0-rc6, but does work if we've incorrectly configured the clock frequency to be twice the actual oscillator (ie halving the sensor speed, and 1080p60 config becomes 1080p30).

We need to test all sensors on 5.18 to see if this is just some odd configuration issue on this module, or something more significant.

### Steps to reproduce the behaviour

5.18.0-rc6 with Arducam IMX462 (updated IMX290/327).

```dtoverlay=imx290,media-controller=0``` for 37.125MHz clock config.
```v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null``` doesn't produce frames.

```dtoverlay=imx290,clock-frequency=74250000,media-controller=0``` for (incorrect) 74.25MHz clock config.
```v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null``` DOES produce frames, but only at half speed.

Doesn't matter whether 2 or 4 data lanes are used.

### Device (s)

Raspberry Pi CM4

### System

Raspberry Pi OS.
Self-built debug firmware from 29th March 2022.
5.18.0-rc6

### Additional context

@davidplowman @naushir FYI

lategoodbye, 2022-05-12:
Not sure this is related by there is a pending clock fix for 5.18:
https://lore.kernel.org/lkml/ee39ddd1-bfce-012d-5e04-448d779ed995@i2se.com/T/

ArduCAM, 2022-05-13:
I'm not sure if this is the same issue, but we had a camera related issue on 5.18,
We tested IMX219, IMX477 and some Arducam Pivariety series and they all have the same problem

**System Info:**
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.18.0-rc6-v7l+ #11 SMP Fri May 13 08:42:04 CST 2022 armv7l GNU/Linux
```
```
pi@raspberrypi:~ $ sudo vcgencmd version
May 10 2022 17:23:09 
Copyright (c) 2012 Broadcom
version a256c26efd3e41d6ce5c9513bf2a3909a118e1cc (clean) (release) (start)
```
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
Hardware	: BCM2711
Revision	: d03114
Serial		: 10000000bdf7ba8a
Model		: Raspberry Pi 4 Model B Rev 1.4
```

**list-formats for imx219:**
```
pi@raspberrypi:~ $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
		Size: Discrete 3280x2464
		Size: Discrete 1920x1080
		Size: Discrete 1640x1232
		Size: Discrete 640x480
	[1]: 'RG10' (10-bit Bayer RGRG/GBGB)
		Size: Discrete 3280x2464
		Size: Discrete 1920x1080
		Size: Discrete 1640x1232
		Size: Discrete 640x480
	[2]: 'RGGB' (8-bit Bayer RGRG/GBGB)
		Size: Discrete 3280x2464
		Size: Discrete 1920x1080
		Size: Discrete 1640x1232
		Size: Discrete 640x480
```

**v4l2-ctl test**
```
pi@raspberrypi:~ $ v4l2-ctl --set-fmt-video=width=1920,height=1080,pixelformat='RG10' --stream-mmap --stream-count=-1
<<<<<<<<<<<< 70.37 fps, dropped buffers: 71
<<<<<<<< 67.20 fps, dropped buffers: 51
<<<<<<<< 67.47 fps, dropped buffers: 69
<<<<<<< 65.24 fps, dropped buffers: 38
<<<<<<<<<<< 59.24 fps, dropped buffers: 30
<<<<<<<< 59.52 fps, dropped buffers: 49
<<<<<<<<<<<<< 60.43 fps, dropped buffers: 51
<<<<<< 58.65 fps, dropped buffers: 91
<< 57.81 fps, dropped buffers: 8
<<<<<<<<<< 56.96 fps, dropped buffers: 37
<<<<<<<< 55.74 fps, dropped buffers: 43
<<<<<< 55.41 fps, dropped buffers: 40
<<<<<<<< 55.33 fps, dropped buffers: 38
<<<<<<<<< 55.03 fps, dropped buffers: 40
<<<< 54.36 fps, dropped buffers: 44
<<<^C
```

The frame rate is very unstable, and using libcamera to access the camera cannot get normal images (it seems that there is a problem with data reception).
[image]

Am I missing some important setting?

6by9, 2022-05-13:
@lategoodbye Thanks for the pointer. I'll check that out.

@ArduCAM The main thing you've missed is that 5.18 is still at release candidate stage. Some issues are to be expected, and I mainly raised this issue so that the observed problems didn't get forgotten.
5.18 is very unlikely to be an LTS release (typically the last release of the year), so unlikely to be adopted by Raspberry Pi OS, and generally gets less testing than LTS releases.

popcornmix, 2022-05-13:
@lategoodbye the link to that patch was useful to me who noticed upstream hdmi output wasn't working on 5.18 (but was on 5.17 and downstream 5.18). Downstream 5.18 uses more firmware clocks so avoids that issue.

6by9, 2022-05-13:
Yup, that clock patch means v4l2-ctl streams, and libcamera gives me sensible pictures with IMX462, and the correct framerate/clock-frequency.

This is with a downstream kernel, but Unicam uses ```clocks BCM2835_CLOCK_CAM0``` or ```clocks BCM2835_CLOCK_CAM1``` for the lp clock, so was being hit by it.

Problem understood, and hopefully that clock patch will make it into the final 5.18 kernel. I'm assuming we won't worry about backporting it in the interim.

pelwell, 2022-05-13:
>  I'm assuming we won't worry about backporting it in the interim.

Is there any reason _not_ to backport it, if it means we can eliminate a source of issues even for a few days?

pelwell, 2022-05-13:
Now applied (currently as 44da93fa3497fd6fa00d7e00b24a5733a6ae200f).

ArduCAM, 2022-05-16:
> @ArduCAM The main thing you've missed is that 5.18 is still at release candidate stage. Some issues are to be expected, and I mainly raised this issue so that the observed problems didn't get [...]

Got it just to confirm if there is a problem with the driver we submitted.

nawaid-ahmed98, 2022-06-04:
Hi Everyone My name is nawaid ahmed and i am new in open source projects. currently i am doing compititve programing But Now i want to get started open source .If anyone guide me i will be be mean a lot to me . you can also contact me on twitter: https://twitter.com/nawaidahmed98

6by9, 2022-06-07:
Clock fix is in 5.18.0, so closing

closed 2022-06-07 by 6by9