raspberrypi/linux #5034: KMS: 4096x2160p60 causes hang/unresponsive pi unless overclocked

alanbork, 2022-05-16:
### Describe the bug

as of 1c56b3a58974f725e1ac63214eeaf914c5908302, I still get hangs/blank screens if I try to use 4096x2160p60 unless I overclock the cpu via config.txt (core_freq=600 core_freq_min=600)

dmesg reports:

vc4-drm gpu: [drm] ERROR Timed out waiting for commit
[ 273.107442] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] ERROR [CRTC:94:crtc-3] flip_done timed out

If I understand correctly this should be fixed by https://github.com/raspberrypi/linux/pull/4940. 

### Steps to reproduce the behaviour

reproducible using kms and modetest at the console (no x). Have not tried X but I see no reason to think it will do better. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

System updated using rpi-update to 1c56b3a58974f725e1ac63214eeaf914c5908302

timg236, 2022-05-16:
I think it will require core_freq=600 because overclocking is required for this to work. Although, maybe you'll be able to remove core_freq_min

alanbork, 2022-05-16:
/root #cat /boot/.firmware_revision
1c56b3a58974f725e1ac63214eeaf914c5908302

/root #base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
AP///////wBN2QP5AQEBAQEbAQOAXzZ4Cg3JoFdHmCcSSEwhCACBgKnAcU+zAAEBAQEBAQEBCOgA
MPJwWoCwWIoAuBcyAAAeAjqAGHE4LUBYLEUAuBcyAAAeAAAA/ABTT05ZIFRWICAqMDAKAAAA/QAX
Pg6IPAAKICAgICAgAfECA1nwW2FgXV5fYh8QFAUTBCAiPD4SFgMHERUCBgFlZiwNfwcVB1A9B7xX
BACDDwAAbgMMADAAuDwvAIABAgMEZ9hdxAF4gAHiAMvjBf8B5Q8DAAAG4wYNAQEdAHJR0B4gbihV
ALgXMgAAHgAAAAAAAAAAAAAAAAAAAAAAAAAAhQ==

and though you didn't ask for it:

/root #modetest
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...done
Encoders:
id      crtc    type    possible crtcs  possible clones
31      94      TMDS    0x00000008      0x00000001
40      0       TMDS    0x00000010      0x00000002
49      0       Virtual 0x00000001      0x00000004

Connectors:
id      encoder status          name            size (mm)       modes   encoders
32      31      connected       HDMI-A-1        950x540         44      31
  modes:
        index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  #0 3840x2160 60.00 3840 4016 4104 4400 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: preferred, driver
  #1 4096x2160 60.00 4096 4184 4272 4400 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #2 4096x2160 59.94 4096 4184 4272 4400 2160 2168 2178 2250 593407 flags: phsync, pvsync; type: driver
  #3 4096x2160 50.00 4096 5064 5152 5280 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #4 4096x2160 24.00 4096 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #5 4096x2160 23.98 4096 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #6 3840x2160 59.94 3840 4016 4104 4400 2160 2168 2178 2250 593407 flags: phsync, pvsync; type: driver
  #7 3840x2160 50.00 3840 4896 4984 5280 2160 2168 2178 2250 594000 flags: phsync, pvsync; type: driver
  #8 3840x2160 30.00 3840 4016 4104 4400 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #9 3840x2160 29.97 3840 4016 4104 4400 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #10 3840x2160 25.00 3840 4896 4984 5280 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #11 3840x2160 24.00 3840 5116 5204 5500 2160 2168 2178 2250 297000 flags: phsync, pvsync; type: driver
  #12 3840x2160 23.98 3840 5116 5204 5500 2160 2168 2178 2250 296703 flags: phsync, pvsync; type: driver
  #13 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: userdef, driver
  #14 1920x1080 59.94 1920 2008 2052 2200 1080 1084 1089 1125 148352 flags: phsync, pvsync; type: driver
  #15 1920x1080i 30.00 1920 2008 2052 2200 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #16 1920x1080i 29.97 1920 2008 2052 2200 1080 1084 1094 1125 74176 flags: phsync, pvsync, interlace; type: driver
  #17 1920x1080 50.00 1920 2448 2492 2640 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: driver
  #18 1920x1080i 25.00 1920 2448 2492 2640 1080 1084 1094 1125 74250 flags: phsync, pvsync, interlace; type: driver
  #19 1920x1080 30.00 1920 2008 2052 2200 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #20 1920x1080 29.97 1920 2008 2052 2200 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #21 1920x1080 24.00 1920 2558 2602 2750 1080 1084 1089 1125 74250 flags: phsync, pvsync; type: driver
  #22 1920x1080 23.98 1920 2558 2602 2750 1080 1084 1089 1125 74176 flags: phsync, pvsync; type: driver
  #23 1680x1050 59.88 1680 1728 1760 1840 1050 1053 1059 1080 119000 flags: phsync, nvsync; type: driver
  #24 1600x900 60.00 1600 1624 1704 1800 900 901 904 1000 108000 flags: phsync, pvsync; type: driver
  #25 1280x1024 60.02 1280 1328 1440 1688 1024 1025 1028 1066 108000 flags: phsync, pvsync; type: driver
  #26 1152x864 75.00 1152 1216 1344 1600 864 865 868 900 108000 flags: phsync, pvsync; type: driver
  #27 1280x720 60.00 1280 1390 1430 1650 720 725 730 750 74250 flags: phsync, pvsync; type: driver
  #28 1280x720 59.94 1280 1390 1430 1650 720 725 730 750 74176 flags: phsync, pvsync; type: driver
  #29 1280x720 50.00 1280 1720 1760 1980 720 725 730 750 74250 flags: phsync, pvsync; type: driver
[... 2495 lines cut ...]
        332 COLOR_ENCODING:
                flags: enum
                enums: ITU-R BT.601 YCbCr=0 ITU-R BT.709 YCbCr=1 ITU-R BT.2020 YCbCr=2
                value: 1
        333 COLOR_RANGE:
                flags: enum
                enums: YCbCr limited range=0 YCbCr full range=1
                value: 0
        334 CHROMA_SITING_H:
                flags: range
                values: 0 65536
                value: 0
        335 CHROMA_SITING_V:
                flags: range
                values: 0 65536
                value: 0
        336 zpos:
                flags: range
                values: 1 17
                value: 17

Frame buffers:
id      size    pitch

/root #

alanbork, 2022-05-16:
> this mode ***@***.***)  always works with overclocking, since firmwares
> released at least a year ago. It's never worked without it - it's always an
> instant fail under kms. I vaguely think it used to work under fkms with a
> much older firmware without overclocking but honestly it's been a while
> since I ran fkms at 4k and when I just tested it recently fkms also
> required overclocking. BTW FKMS has some serious regressions these days,
> even low demanding modes sometimes don't work (480i) for instance.

force_turbo=1 results in a fail for kms at  ***@***.***

alanbork, 2022-05-16:
To be more clear:

full config.txt that fails:

#following lines set up kms and required overclocking
dtoverlay=vc4-kms-v3d
#core_freq=600
#core_freq_min=600
force_turbo=1

if I swap which overclocking lines are commented out it works fine.

popcornmix, 2022-05-16:
> force_turbo=1 results in a fail for kms at  ***@***.***

Just to confirm it results in a `drm_atomic_helper_wait_for_flip_done` timeout in dmesg?

I'm running with your edid. It doesn't produce a picture on my dell monitor for 4096x2160@60, but does for other modes.
I've been running a script:
```
#! /bin/sh
#
SLEEP=5
while true;
do
   for i in 1024x768 3840x2160-30 1920x1080 3840x2160-60 4096x2160-60;
   do
     echo "Testing $i"
     (sleep $SLEEP; echo "whatever") | modetest -M vc4 -s 32:$i
   done
done
```
which produces the flip timeout when run with:
```
dtoverlay=vc4-kms-v3d
hdmi_enable_4kp60=1
```
but for me, adding either `force_turbo=1` or `core_freq_min=550` does seem to avoid the flip timeout (it's run for tens of minutes).

> full config.txt that fails:

You didn't list hdmi_enable_4kp60=1. I assume you do have that?

alanbork, 2022-05-16:
Yes, this message:
drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] ERROR [CRTC:94:crtc-3] flip_done timed out

yes,

hdmi_enable_4kp60=1

popcornmix, 2022-05-16:
I'll leave my script running overnight. But I don't seem to be getting the same behaviour as you (which I feel I should when using your edid).

I'm on RPiOS bullseye, with boot to console enabled. I have your edid `Manufacturer: SNY`, `Model: 63747`, `Serial Number: 16843009`. I'm on latest rpi-update kernel/firmware:
pi@pi4:~ $ uname -a
```
Linux pi4 5.15.40-v7l+ #1555 SMP Mon May 16 16:18:30 BST 2022 armv7l GNU/Linux
pi@pi4:~ $ vcgencmd version 
May  6 2022 13:29:01 
Copyright (c) 2012 Broadcom
version 84953e1f6aea5fd52ce22bf95213b5cb907a74ac (tainted) (release) (start)
```
`core_freq_min=550`  avoids the flip timeout for me.
I've tried with/without `disable_overscan`/`disable_fw_kms_setup` but that doesn't affect behaviour.

I note you posted modetest with:
```
94      338     (0,0)   (1920x1080)
  #0 1920x1080 60.00 1920 2008 2052 2200 1080 1084 1089 1125 148500 flags: phsync, pvsync; type: userdef, driver
```
Was this straight from boot. I boot up with:
94      338     (0,0)   (3840x2160)
  #0 3840x2160 60.00 3840 4016 4104 4400 2160 2168 2178 2250 594000 flags: phsync, pvsync; 3d:0; type: preferred, userdef, driver
Or more readably:
```
pi@pi4:~ $ kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (94) 3840x2160@60.00 594.000 3840/176/88/296/+ 2160/8/10/72/+ 60 (60.00) 0x5 0x68
      Plane 3 (84) fb-id: 338 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30)
        FB 338 3840x2160
```
After `modetest -M vc4 -s 32:4096x2160-60` what do you get from:
```
pi@pi4:~ $ vcgencmd measure_clock core
frequency(1)=549993152
```

popcornmix, 2022-05-17:
Does `core_freq=600` (without `core_freq_min=600` or `force_turbo=`) work for you?
Does 4096x2160@50 work for you (with just `hdmi_enable_4kp60=1` and no clock boosts)?

alanbork, 2022-05-17:
> Does `core_freq=600` (without `core_freq_min=600` or `force_turbo=1`) work for you? Does 4096x2160@50 work for you (with just `hdmi_enable_4kp60=1` and no clock boosts)?

Would this info still be useful in light of the draft commit that came after? I'm happy to do the testing if it's still useful.

popcornmix, 2022-05-17:
Yes, both these tests are still useful. It's not clear whether 4096x2160@50  suffers from the same issue as there are fewer pixels per second to handle. Although it does use same pixel clocks rate, with larger horizontal blanking, so it may not be easier.

I suspect you need `core_freq_min` rather than `core_freq_max`, although I don't know if that's been confirmed.

andrum99, 2022-05-17:
I have a suspicion that board-specific clock and voltage timings might be making this work for popcornmix and not for alanbork, assuming popcornmix's setup is actually producing the requested video timings verbatim from the EDID. If core_freq=550 is all that is needed on one Pi, and it definitely produces the same video timings, then the correct solution would be to figure out what the difference between boards is, since it may be that this is fixable by adjusting the board-specific voltage and clock tuning. I may, of course, have misunderstood.

alanbork, 2022-05-17:
Just in case I grabbed the latest rpi-update:

/root #cat /boot/.firmware_revision
01ebd10b0e654679d88d3de50062cc5121a8d1ab

holding these lines of config.txt constant for all tests:

max_framebuffers=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
hdmi_enable_4kp60=1
disable_overscan=1

then setting these:

dtoverlay=vc4-kms-v3d
#overclcking only for kms
core_freq=600
core_freq_min=600

these modes work fine:
4096x2160p50
 4096x2160p60

then turning all overclocking off:
dtoverlay=vc4-kms-v3d
#overclcking only for kms
#core_freq=600
#core_freq_min=600

4096x2160p50  works fine
4096x2160p60.0 fails as before
4096x2160p59_94 fails (first time testing)

by fail I mean TV detects no signal + dmesg says 

[drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:94:crtc-3] flip_done timed out

then with these lines:
dtoverlay=vc4-kms-v3d
core_freq=600

4096x2160p50 fine
4096x2160p59_94 fails
4096x2160p60 fails

You also asked about vcgencmd measure_clock core

not sure which configuration you wanted that run under, but here's the output with just core_freq=600

frequency(1)=533329088

a bit different from yours...

testing with :
dtoverlay=vc4-kms-v3d
core_freq_min=600

4096x2160p50  is fine
4096x2160p59_94 is fine
4096x2160p60.0 is fine

 Did I miss any combo's?

popcornmix, 2022-05-30:
@alanbork does latest rpi-update kernel behave okay?
i.e. only allow 4096x2160@60 when it will work?

alanbork, 2022-05-31:
> i.e. only allow ***@***.*** when it will work?

Still not back to be able to test but will be able to by Friday.

timg236, 2022-06-09:
core_freq_min means that the core clock will never be less than that value whereas core_freq increases the max core-freq (to overclock). In a perfect world the KMS driver will always be able to perfectly calculate the lowest possible frequency (to save power) for the display to be stable. However, the complexity of the HVS frequency requirements + memory bandwidth (core is also AXI freq) means there's some margin.
If the 1% failure is reproducible e.g. mode-switch sequence + set of layers then that would be good to know.

timg236, 2022-06-10:
"Thanks for explaining. You are saying core_freq really should be called core_freq_max for consistency?"

Without doing the code archaeology I would guess that most clocks were fixed frequency i.e. clock_freq and then gained _min when the drivers were updated to dynamically lower the frequency to save power.

alanbork, 2022-06-10:
So that's a yes?

pelwell, 2022-06-10:
Obvs.

alanbork, 2022-06-10:
I think this is safely closeable.  I thought somebody had volunteered in this thread to add a note to the documentation of the new requirements for 4k support but I guess I imagined it; it should definitely make it there at some point.

closed 2022-06-10 by alanbork