raspberrypi/linux #5045: Long delay to enumerate some usb devices

LonerDan, 2022-05-23:
### Describe the bug

I am making a custom image using Buildroot. The stock kernel defconfig has a large amount of bloat modules for my use case, so I made a stripped-down kernel config. The system boots, but there is a large delay of ~15 seconds before the USB enumerates the last two devices (lan78xx USB-to-ethernet and MCP2221 USB to UART). I suspect there may be an issue with dwc_otg driver having some dependency that is not declared in the Kconfig portion, but I am not sure where to start looking exactly, since that driver seems pretty complex and spread across multiple files and directories. Perhaps someone who has deeper insight into that driver could give me a hint. 

### Steps to reproduce the behaviour

- Strip the 'bcmrpi3' defconfig of optional drivers.

[stock kernel config](https://github.com/raspberrypi/linux/files/8753082/rpi3_stock_kernel.txt)
[my stripped kernel config](https://github.com/raspberrypi/linux/files/8753086/rpi3_64_minified.txt)

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Buildroot 2022.02.1 'raspberrypi3_64_defconfig' with minimal changes (mainly kernel version bump to 1.20220331)
uname -a: `Linux (none) 5.15.32-v8 #2 SMP PREEMPT Fri May 20 17:10:28 CEST 2022 aarch64 GNU/Linux`

### Logs

[Stock kernel config bootlog](https://github.com/raspberrypi/linux/files/8753016/bootlog_stock_config.txt)
[My stripped kernel config bootlog](https://github.com/raspberrypi/linux/files/8753019/bootlog_modified_config.txt)

pelwell, 2022-05-23:
The issue creation page you clicked through says:

> Please do not use GitHub for asking questions. If you simply have a question, then the Raspberry Pi forums are the best place to ask it. Thanks in advance for helping us keep the issue tracker clean!

pelwell, 2022-05-23:
But try adding `initcall_debug` to cmdline.txt and looking for gaps.

closed 2022-05-23 by pelwell

LonerDan, 2022-05-23:
I am sorry if you see it differently, but this was meant as a report of an issue with the dwc_otg driver.