raspberrypi/linux #5046: USB device requires unplugging and re-plugging to make it work after reboot.

ReneElizondo, 2022-05-24:
### Describe the bug

I am trying to get a device called [ThingMagic USB Pro UHF Reader](https://www.jadaktech.com/products/thingmagic-rfid/thingmagic-usb-pro-uhf-reader/) to work properly with a Raspberry Pi 4. For the most part, everything works perfectly fine, until I reboot or turn on / off the Pi, at that point, the device in no longer recognized and I must unplug and re-plug it in order to get it to work again.

Please note that someone else doing the same thing with a Raspberry Pi 3 said that the Pi 3 works perfectly fine, so this appears to be an issue with the Pi 4. I am new to Linux / Raspberry Pi so I have no idea what I am doing, however below is some information that I hope will help troubleshoot the issue.

Also interesting to note is that if I connect some USB hubs (not all but some) to the Raspberry Pi and then the USB device to the hub things work just fine (the unit is recognized by the Raspberry Pi).

Finally, after the OS has booted up, the mouse and keyboard (connected to other USB ports) will freeze and stay frozen until some time elapse and then they become available.

Using standard (recommended) Raspberry PI OS (32-bit). 

Below is a snapshot of what I think are the important bits from dmesg:

[image]

### Steps to reproduce the behaviour

Unfortunately, to reproduce this issue you will need a device called [ThingMagic USB Pro UHF Reader](https://www.jadaktech.com/products/thingmagic-rfid/thingmagic-usb-pro-uhf-reader/). But if you have one :), then all you need to do is plug it in, boot the Raspberry Pi and you will experience the pain.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage4

$ vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

$ uname -a
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

### Logs

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.32-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP Thu Mar 31 19:39:41 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:79:62:B7 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=e6fc1d7b-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000014890000-0x0000000018890000] (64MB)
[    0.000000] Memory: 3601320K/4050944K available (10240K kernel code, 1390K rwdata, 3292K rodata, 2048K init, 590K bss, 121944K reserved, 327680K cma-reserved, 3264512K highmem)
[... 432 lines cut ...]
[   15.693332] Bluetooth: HCI UART protocol Broadcom registered
[   16.031483] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.031495] Bluetooth: BNEP filters: protocol multicast
[   16.031507] Bluetooth: BNEP socket layer initialized
[   16.061973] NET: Registered PF_ALG protocol family
[   16.125259] cryptd: max_cpu_qlen set to 1000
[   16.285021] Bluetooth: RFCOMM TTY layer initialized
[   16.285041] Bluetooth: RFCOMM socket layer initialized
[   16.285115] Bluetooth: RFCOMM ver 1.11
[   18.014135] usb 1-1.3: device descriptor read/64, error -110
[   31.837036] cam-dummy-reg: disabling
[   33.378065] usb 1-1.3: device descriptor read/64, error -110
[   33.597902] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[   48.822642] usb 1-1.3: device descriptor read/64, error -110
[   64.184345] usb 1-1.3: device descriptor read/64, error -110
[   64.304891] usb 1-1-port3: attempt power cycle
[   64.964096] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[   65.587170] xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x24.
[   66.427163] xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x24.
[   66.644210] usb 1-1.3: device not accepting address 7, error -22
[   66.744233] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[   67.367316] xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x24.
[   68.207179] xhci_hcd 0000:01:00.0: ERROR: unexpected setup address command completion code 0x24.
[   68.424293] usb 1-1.3: device not accepting address 8, error -22
[   68.425031] usb 1-1-port3: unable to enumerate USB device

solmoller, 2022-05-30:
looks familiar to #5005 , as adding an USB hub fixes the problem sometimes

popcornmix, 2022-06-08:
@JamesH65 did you mean @P33M ?

JamesH65, 2022-06-09:
I did indeed, mixing up my usernames....this has also come up on the forums.

P33M, 2022-06-09:
It goes without saying that we don't have one of those devices.

This is a full-speed device so is accessed behind a transaction translator. The very first transaction to the device times out which suggests it's not in a state where it responds to the a Setup packet - which should not happen. My guess is whatever reset strategy is implemented inside the device doesn't account for whatever intermediate state the bus is in while the Pi is rebooting - note that we power cycle ports over reset.

One workaround to try would be to configure USB mass-storage boot and set the bootconf.txt parameter USB_MSD_PWR_OFF_TIME to some large number, e.g. 2000. The Pi will power cycle the USB ports when attempting mass-storage boot, but fail and boot from SD.

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-4-boot-flow
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_PWR_OFF_TIME

ReneElizondo, 2022-06-09:
@P33M 

Thank you very much for your help, unfortunately, the suggested workaround did not work, I changed the value to 2000, 5000 and some other values in between but the same error kept occurring.

[image]

There are some things that I wanted to point out... but just keep in mind that I have no idea what I am talking about here so most of my comments are probably going to be gibberish.

- The ThingMagic device fails to be recognized even if its plugged in to a powered USB-3 hub that is in turn connected to the Pi. In this case, the powered USB-3 hub keeps the ThingMagic device on all the time even as the Pi is being rebooted. So If the ThingMagic device was responding and working fine while the Pi was on, then I assume that it should be on a ready state as the Pi is being rebooted given that it never lost power.
- I am guessing here that ThingMagic device takes milliseconds to get ready, I assume that this is true because when I plug the ThingMagic device to the Pi while the Pi is running, the ThingMagic device gest recognize immediately (literally). That being the case, the ThingMagic device appears to transitions to a ready state very, very fast.
- Looking at the ThingMagic device power light, I see that the unit turns on long before the boot process even begins (before the multicolor splash screen shows up). I guessing here that there is plenty of time for the ThingMagic device to get to its ready state before its called upon.

Could it be that the Pi OS is sending some bad messages to the ThingMagic device during startup that messes up the device? Or could it be a bug on the USB drivers that have a hard time talking to the new USB hardware? This may explain why things work on the Pi 3 but not on the Pi 4?

Do you have any other suggestions to try? Is there anything you want me to do to help troubleshoot this? Do you need SSH access to the Pi with the ThingMagic device connected to it? Let me know what I can do, we are really eager to see this being fixed assuming this issue is fixable.

Thanks.

ReneElizondo, 2022-06-09:
Continuing with my potential gibberish comments... just wanted to reiterate that when the ThingMagic USB device is connected, the problem is not just with the ThingMagic device, the problem spawns to the other USB devices. In my case I have a mouse and a keyboard connected to the other USB ports and both the mouse and keyboard become locked (unusable) while the OS is trying to figure out how to sort the USB issue.

I assume that if this issue was strictly related to the ThingMagic USB device, the keyboard and mouse would continue to work without a problem, the fact that they do not, tells me that something bigger is getting messed up.

P33M, 2022-06-10:
The disconnected mouse/keyboard are a side-effect of Linux trying to recover the device as evidenced by "attempt power cycle" messages in dmesg.

What is the boot order of the system? For the power cycle workaround to have an effect, USB mass-storage must come before SD. 

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER

ReneElizondo, 2022-06-10:
Originally the boot order was the default order (there was no BOOT_ORDER entry outputted by `rpi-eeprom-config` command). So I went in and change things to the following:

[image]

Unfortunately, that still did not work. I can tell right away that is not going to work because I can see that the ThingMagic device is powered on before the boot process start, when that happens the ThingMagic device is never recognized.

One way that I have have been able to make the ThingMagic device work during the bootup process is to wait a couple of seconds after the boot process starts and then plug in the ThingMagic USB device. Usually, plug it in some time before the `Welcome to Raspbian GNU/Linus 11 (bullseye)!` message shows up. If I do that, the boot process will continue and the ThingMagic device is successfully recognized. So something is happening during that time that messes up the ThinkgMagic serail communications.

In case its relevant, this is what I get during the initial boot screen:

[image]

timg236, 2022-06-10:
The USB power off happens all the time on R1.4 during sdram init or before the first boot mode on r1.1. There’s no way to make it zero since the hardware will cause a very brief power off by itself.
During boot mode 4 you can tell the bootloader by excluding by VID/PID (see eeprom config docs). It’s worth a shot but probably won’t make a difference!

ReneElizondo, 2022-06-10:
mmm... this may be a little over my skill set (I am very new to the Raspberry Pi), I could try this but I may need some more specific instructions to be able to test that if its ok.

I am curious though about the significance of the `xHC-CMD err: 36 type: 11` error,  this is happening during the "core" boot stage, I don't even have the sd card inserted in the Pi and I still see the error. The error only shows up if the ThingMagic devic is connected, if the ThingMagic  device is not connected there are no error whatsoever. Also, under previous conditions, I would just unplug and replug the ThingMagic device and things would work, this time, no matter how many time I unplug and replug the ThingMagic device the `xHC-CMD` error keeps showing up.

As usually, have no clue what I am talking about but it almost seems to me as if the USB issue was at the hart of the Raspberry Pi functionality given that the OS is not even playing a part at the time of the error.

P33M, 2022-06-13:
The command response code is the same as is seen in Linux - split transaction error. The difference in behaviour between a replug on Linux and a replug with the bootloader is likely due to timing - the bootloader has nothing better to do except pay attention to USB whereas Linux can be a little slower on the uptake.

Given that device enumeration fails right out of the gate, what happens if you change the probing scheme that Linux uses by adding ` usbcore.old_scheme_first=1` to /boot/config.txt ?

popcornmix, 2022-06-13:
>  by adding usbcore.old_scheme_first=1 to /boot/config.txt ?

I think this should be `/boot/cmdline.txt`

ReneElizondo, 2022-06-15:
Unfortunately, adding `usbcore.old_scheme_first=1` to the cmdline.txt did not work.

dmesg output:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.32-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP Thu Mar 31 19:39:41 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 503104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:A3:0B:24 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=e2d5ff95-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usbcore.old_scheme_first=1
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000015a00000-0x0000000019a00000] (64MB)
[    0.000000] Memory: 1587560K/2019328K available (10240K kernel code, 1390K rwdata, 3292K rodata, 2048K init, 590K bss, 104088K reserved, 327680K cma-reserved, 1232896K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35355 entries in 104 pages
[    0.000000] ftrace: allocated 104 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x574/0x758 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000002] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000021] Switching to timer-based delay loop, resolution 18ns
[    0.000330] Console: colour dummy device 80x30
[    0.000398] printk: console [tty1] enabled
[    0.000466] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000496] pid_max: default: 32768 minimum: 301
[    0.000640] LSM: Security Framework initializing
[    0.000802] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000833] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002438] cgroup: Disabling memory control group subsystem
[    0.002690] CPU: Testing write buffer coherency: ok
[    0.002749] CPU0: Spectre BHB: using loop workaround
[    0.003212] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004463] Setting up static identity map for 0x200000 - 0x20003c
[    0.004687] rcu: Hierarchical SRCU implementation.
[    0.005639] smp: Bringing up secondary CPUs ...
[    0.006873] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006891] CPU1: Spectre BHB: using loop workaround
[    0.008245] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.008263] CPU2: Spectre BHB: using loop workaround
[    0.009589] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[... 406 lines cut ...]
[   14.892181] Bluetooth: HCI device and connection manager initialized
[   14.892211] Bluetooth: HCI socket layer initialized
[   14.892232] Bluetooth: L2CAP socket layer initialized
[   14.892279] Bluetooth: SCO socket layer initialized
[   14.910074] Bluetooth: HCI UART driver ver 2.3
[   14.910091] Bluetooth: HCI UART protocol H4 registered
[... cut ...]
[   21.851570] usb 1-1.3: device descriptor read/64, error -110
[   31.831942] cam-dummy-reg: disabling
[   37.212603] usb 1-1.3: device descriptor read/64, error -110
[   37.432325] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[   52.653227] usb 1-1.3: device descriptor read/64, error -110
[   68.013445] usb 1-1.3: device descriptor read/64, error -110
[   68.133934] usb 1-1-port3: unable to enumerate USB device
[  103.775993] v3d fec00000.v3d: MMU error from client CLE (4) at 0x14a1000, pte invalid
```

popcornmix, 2022-06-15:
> Unfortunately, adding usbcore.old_scheme_first=1 to the cmdline.txt did not work.

You can confirm adding the option has worked with `cat /proc/cmdline` and checking for `usbcore.old_scheme_first=1`

6by9, 2022-06-15:
```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=E4:5F:01:A3:0B:24 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=e2d5ff95-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usbcore.old_scheme_first=1
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
```

```usbcore.old_scheme_first=1``` is there on the end.

P33M, 2022-06-15:
In that case that pretty much exhausts all the tunable options for altering device probe behaviour. Does the device work when plugged into a Linux x86 PC with xHCI controllers?

ReneElizondo, 2022-06-15:
Thanks everyone for your help.

I am still struggling with the fact that if I plug the ThingMagic device 2 or 3 seconds after the Pi starts the bootup process everything works fine. Also, I can unplug and re-plug the ThingMagic device while the Pi is on as many times as I want  (we have done that literally hundreds of times) and everything works just fine. To me, this feels more like right at startup (very early in the bootup process), something in the hardware or software is happening that puts the ThingMagic device on a bad and unrecoverable state rendering the device unusable. Is this not a possible scenario?

Other potential relevant info may be that we have been using these devices for years (5 or 6 years) on Windows (laptop / desktops) and we have never experienced any problems (I am sure those computers use xHCI controllers).

I will try to put together a Linux computer where I can try this out, however, the company that sells the ThingMagic device said that they have tested this in Linux and they experienced no issues. They did test the ThingMagic device on a Raspberry Pi 4 after I reported the issue to them and they were able to reproduce the problem but they have no intention of fixing the problem on their side because they claim it only fails on Raspberry Pi 4 computers.

Would it help if I ship you guys a ThingMagic device for you to test with? Those things are very expensive but its worth it for us given that we have developed a product that uses the Raspberry Pi that worked in a test environment but is failing big time in a production environment (after it has been deployed for a couple for days).... this is not the only issue we are experiencing by the way, there is another big issue but we are hoping that if we can fix this issue the other issue may get fixed too... 

Also, we are ok with paying for this issue to get fixed (assuming the cost is reasonable), we are petty desperate to get this fixed.

Please let me know.

Thanks.

ReneElizondo, 2022-06-19:
In case it matters, the other issue we are experiencing has to do with the ThingMagic USB device constantly disconnecting and reconnecting. Example log below:

```
Jun 14 19:26:56 raspberrypi kernel: usb 1-1.2: USB disconnect, device number 5
Jun 14 19:26:56 raspberrypi kernel: usb 1-1.2.1: USB disconnect, device number 6
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2: new high-speed USB device number 7 using xhci_hcd
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2: Product: USB2.0 HUB
Jun 14 19:26:57 raspberrypi kernel: hub 1-1.2:1.0: USB hub found
Jun 14 19:26:57 raspberrypi kernel: hub 1-1.2:1.0: 4 ports detected
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2.1: new full-speed USB device number 8 using xhci_hcd
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2.1: New USB device found, idVendor=2008, idProduct=1004, bcdDevice= 0.01
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2.1: Product: USBPro5121348010572KH+
Jun 14 19:26:57 raspberrypi kernel: usb 1-1.2.1: Manufacturer: ThingMagic, Inc.
Jun 14 19:26:57 raspberrypi kernel: cdc_acm 1-1.2.1:1.0: ttyACM0: USB ACM device
```

This happens many times a day, eventually unfortunately, luck runs out and at some point, when the ThingMagic USB device tries to reconnect you start seeming errors such as:

```
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2: clear tt 1 (8000) error -71
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2: clear tt 1 (8000) error -71
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2: clear tt 1 (8000) error -71
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot disable (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot reset (err = -71)
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
Jun 15 23:58:09 raspberrypi kernel: usb 1-1.2-port1: cannot disable (err = -71)
```

```
Jun 16 06:17:03 raspberrypi kernel: hub 1-1.2:1.0: hub_ext_port_status failed (err = -71)
Jun 16 06:17:03 raspberrypi kernel: usb 1-1.2: Failed to suspend device, error -71
```

```
Jun 16 08:36:14 raspberrypi kernel: usb 1-1.2: device descriptor read/64, error -71
Jun 16 08:36:14 raspberrypi kernel: usb 1-1.2: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
Jun 16 08:36:14 raspberrypi kernel: usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Jun 16 08:36:14 raspberrypi kernel: usb 1-1.2: Product: USB2.0 HUB
Jun 16 08:36:14 raspberrypi kernel: hub 1-1.2:1.0: USB hub found
Jun 16 08:36:14 raspberrypi kernel: hub 1-1.2:1.0: 4 ports detected
Jun 16 08:36:15 raspberrypi kernel: usb 1-1.2.1: new full-speed USB device number 26 using xhci_hcd
Jun 16 08:36:16 raspberrypi kernel: xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 3.
Jun 16 08:36:16 raspberrypi kernel: usb 1-1.2: USB disconnect, device number 25
Jun 16 08:36:16 raspberrypi kernel: xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 3.
Jun 16 08:36:16 raspberrypi kernel: usb 1-1.2.1: device not accepting address 26, error -22
```

`usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?`

Thanks.

antonypenn, 2022-06-21:
You could make a small piece of hardware to fix your issue. Make a small USB male-female lead, cut the +5v lead and connect that to a MOSFET transistor, controlled by a GPIO pin. That way, you can power it on and off via one of the GPIO pins. I do this to control other pieces of hardware.

ReneElizondo, 2022-06-21:
@antonypenn Thanks, we actually thought of doing something similar to your suggestion, the problem is that we also discovered that the ThingMagic USB device disconnects and reconnects all the time while the Pi is on (https://github.com/raspberrypi/linux/issues/5046#issuecomment-1159818309), eventually the reconnect starts failing and thing go downhill from there. Its gets bad enough that eventually, the device will fail to reconnect and even un-plugiin and re-plugin the ThingMagic USB Device will still fail, only a reboot can get things to work again.

I am starting to think the issue may be related to USB hardware issues, even the smallest voltage change appears to trigger a disconnect / reconnect on the ThingMagic USB device. Some USB devices like mouse and keyboards appear to be less sensitive to voltage changes. It make look like this is unique to the ThingMagic device but I see others have similar issues.

Just keep hoping a solution will be found at sometime, or at least find the source cause even with no solution, jus an answer of what may be going on would be great.

antonypenn, 2022-06-21:
You might want to try a different USB cable. I think you mentioned that your cable is captive, which is a shame. I have just solved a problem that uses the same chip as your product (thats how I came across your post) and solved it by replacing the OEM's supplied cable, with one I had lying around here. Especially for high speed (USB2) and over, the cable quality (screening etc) is very important. One option (if you are brave enough) is to cut your cable off the device, just a few inches long, and put a USB plug on it, then use a good quality USB extension lead.

ReneElizondo, 2022-07-06:
@P33M 

> In that case that pretty much exhausts all the tunable options for altering device probe behavior. Does the device work when plugged into a Linux x86 PC with xHCI controllers?

We ended up putting together a computer and installed Debian OS (the latest versions), the install was "pure" (no dual boot, no windows, no nothing), just pure Linux on that machine. We rebooted the machine several times as well as turn the computer off and on several times and every single time the ThingMagic USB device was recognized without a problem. I have included the dmesg below in case you may find it useful.

That said, this (to me at least) appears to indicate that this is a bug with the Raspberry Pi. If that is the case, what is the next step? Is this something that you guys would look into and attempt to fix? We would really like to get this issue resolved even if it means having to pay to fix the issue (assuming the cost is reasonable).

Thanks.

```
[    0.000000] Linux version 5.10.0-15-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Debian 5.10.120-1 (2022-06-09)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.0-15-amd64 root=UUID=81d46985-fa68-408f-85c0-ebc2a40740e0 ro quiet
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000b66aafff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b66ab000-0x00000000b798efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b798f000-0x00000000b7b8efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b7b8f000-0x00000000b7c0efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000b7c0f000-0x00000000b7c0ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b7c10000-0x00000000be7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000023d7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.60 by HP
[    0.000000] efi: ACPI=0xb7c0e000 ACPI 2.0=0xb7c0e014 ESRT=0xb6e75b98 TPMFinalLog=0xb7b37000 SMBIOS=0xb6e74000 MEMATTR=0xb2cdd018 MOKvar=0xb2849000 RNG=0xb6e72d98 TPMEventLog=0xa4f72018 
[    0.000000] efi: seeding entropy pool
[    0.000000] Kernel is locked down from EFI Secure Boot; see man kernel_lockdown.7
[    0.000000] secureboot: Secure boot enabled
[    0.000000] SMBIOS 3.1 present.
[    0.000000] DMI: HP HP EliteBook 840 G6/8549, BIOS R70 Ver. 01.06.00 08/03/2020
[    0.000000] tsc: Detected 1800.000 MHz processor
[    0.000013] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000018] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000026] last_pfn = 0x23d800 max_arch_pfn = 0x400000000
[    0.000032] MTRR default type: write-back
[    0.000033] MTRR fixed ranges enabled:
[    0.000035]   00000-9FFFF write-back
[    0.000037]   A0000-BFFFF uncachable
[    0.000039]   C0000-FFFFF write-protect
[    0.000040] MTRR variable ranges enabled:
[    0.000042]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.000044]   1 base 00BC000000 mask 7FFC000000 uncachable
[    0.000046]   2 base 2000000000 mask 6000000000 uncachable
[    0.000048]   3 base 1000000000 mask 7000000000 uncachable
[    0.000049]   4 base 0800000000 mask 7800000000 uncachable
[    0.000051]   5 base 0400000000 mask 7C00000000 uncachable
[    0.000052]   6 base 4000000000 mask 4000000000 uncachable
[    0.000053]   7 disabled
[    0.000054]   8 disabled
[    0.000055]   9 disabled
[    0.000835] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001388] last_pfn = 0xb7c10 max_arch_pfn = 0x400000000
[    0.015384] esrt: Reserving ESRT space from 0x00000000b6e75b98 to 0x00000000b6e75bf8.
[    0.015395] e820: update [mem 0xb2849000-0xb284bfff] usable ==> reserved
[    0.015420] Using GB pages for direct mapping
[    0.016146] RAMDISK: [mem 0x3b468000-0x3dc47fff]
[    0.016149] ACPI: Early table checksum verification disabled
[    0.016153] ACPI: RSDP 0x00000000B7C0E014 000024 (v02 HPQOEM)
[... 1179 lines cut ...]
[    5.593189] bluetooth hci0: Direct firmware load for intel/ibt-20-1-3.sfi failed with error -2
[    5.593932] Bluetooth: hci0: Failed to load Intel firmware file (-2)
[    5.610571] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    5.660075] r8152 4-1.3.3:1.0 enx3024a94383c9: renamed from eth0
[    5.688611] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[... cut ...]
[    9.572435] hid-generic 0003:03F0:0667.0004: hiddev1,hidraw2: USB HID v1.11 Device [Cypress Semiconductor WinUSB] on usb-0000:06:00.0-1.3.7/input1
[   10.152189] usbcore: registered new interface driver snd-usb-audio
[   13.954540] rfkill: input handler enabled
[   14.901935] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[   15.125115] rfkill: input handler disabled
[  133.365636] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7

```

P33M, 2022-07-06:
Without representative hardware there is little chance of a diagnosis. If you have a unit spare that you can loan, then please email info << at >> raspberrypi.com and we can take a look.

P33M, 2022-07-14:
I've now got one in my possession. The device can't handle being suspended, and never responds to SETUP packets after a resume.

You can reproduce this without a Pi 4 - using a powered USB hub that does not switch its downstream port power, plug the RFID reader into the downstream port. Then connect the upstream port to a PC, and the device fails to enumerate.

The glitchy mouse behaviour on a Pi 4 is because Linux is trying to power-cycle the hub port. The upstream root ports need to be power cycled.

P33M, 2022-07-14:
I can recover the port state by forcibly power cycling the Pi 4's downstream ports with uhubctl.

`sudo uhubctl -l 1-1 -a 2` brings it back to life.

It's not included in Raspberry Pi OS by default - install it with `sudo apt install uhubctl`.

ReneElizondo, 2022-07-14:
Thanks @P33M 

Its great to know that we now have a diagnosis. Quick questions if you don't mind. What do you mean by "**_does not switch its downstream port power_**"? You mean the port does not switch its power on and off? Just trying to make sure I understand what is going on and may need more of a  layman's terms explanation!

Thanks.

P33M, 2022-07-14:
As in, the hub either does not implement any Vbus switching at all (no hardware support) or it doesn't use an upstream port disconnect to imply "turn off downstream Vbus".

ReneElizondo, 2022-07-14:
So what changed between the Pi 3 and Pi 4 so that in the Pi 3 things work just fine but in the Pi 4 things fail? Why can't the Pi 4 do what the Pi 3 does?

JamesH65, 2022-07-14:
The Pi4 uses a completely different USB hub chip that is directly connected to the PCIe port on the Pi4 SoC. The Pi3 has a USB hub attached to the USB2 port.

P33M, 2022-07-14:
USB port suspend is a state inferred by the /absence/ of any activity following a port reset, start-of-frame token or host data token for more than 3ms. It seems that the reboot sequence of a Pi 4 has a transient state where suspend is being inferred, and this breaks the RFID reader. It's almost certainly a firmware flaw with the device.

ReneElizondo, 2022-07-15:
I finally got the chance to do some more research on this. As you correctly stated, I was able to reproduce the hub problem even in Windows. I think this information is going to be incredibly helpful (THANK YOU), I will send this new discovery to the people that created the ThingMagic device to see if they can use this info to fix their stuff.... although I seriously doubt it.

That said, there is something that just does not feel right about all this. I understand that the ThingMagic device has some issues, nevertheless, even with those issues the device works on most computers under the typical use case (which is plugging the device directly to the computer USB port).

Now, I understand that this is not the kind of device you can buy at the corner store but its a very popular device among RFID users and they sell millions of these things worldwide. These millions of ThingMagic devices are used by millions of different computers and they all work just fine (under the mentioned typical use case).... except on the Raspberry Pi 4.

So the question that begs to be asked is why does the Raspberry Pi 4 does things differently? I am not saying that the Raspberry Pi 4 is doing the wrong thing, I am simply saying that it is doing things differently than most if not all other computers out there (including the Pi 3). Why not simply take the tried and true protocol / exact sequence of events that happens when plugging a USB device into a normal computer and make those sequence of events to be the same in the Pi 4 so that the behavior on the Pi 4 is the same as the rest of the computers? Which means, there would be no issues just plugin the ThingMagic device directly in the Pi 4 (typical use case). What harm or negative effect can come from that?

Thanks.

JamesH65, 2022-07-15:
The specific behaviour here is that provided by the PCIe USB hub chip. This is provided by a third party, I think a different manufacturer to the completely different chip used on the Pi3. They obviously do something slightly different, as is common between manufacturers. We buy these devices in, and rely on the manufacturers to adhere to the specifications, which on the whole they do. In this case it looks like there is some very specific behaviour that is different between the chips, despite both adhering to the USB specification. The issue is that the ThingMagic device does NOT appear to adhere to the USB specification (although P33M will confirm). This is apparently fine in most circumstances, but not in this specific one. Whether the manufacturer of the PCIe hub chip would be willing to update their firmware to compensate for the Thingmagic's fault is unknown. It might also not be "get roundable".

JamesH65, 2022-07-15:
Worth adding that the huge majority of USB devices work absolutely fine on the Pi4.

ReneElizondo, 2022-07-15:
Thank you guys for all the help with trying to figure out what the issue with the ThingMagic device was even though as it turns out this wasn't really a Raspberry Pi 4 issue (sorry about that). I have reported the new findings to the company that makes the ThingMagic device, lets see how things go from here.

Before moving on from this issue, I was wondering, given that you guys have a ThingMagic device there to play with, any chance you guys would have some time to investigate why the device constantly disconnects and reconnects after the Pi 4 registers the ThingMagic device? This appears to be a similar issue as to what other are experiencing with SSDs.

The only way to stop that problem from happening is by adding a USB hub between the Pi and the ThingMagic device (in our case, this needs to be a powered USB hub 2.0 since USB 2.0 hubs seem to play well with the Pi 4). But what is puzzling is that the device seems to disconnect / reconnect even when the ThingMagic device is idling (using almost zero power). So I don't think this is an issue with the Pi 4 not being able to deliver enough power, I think there is more to the story.

To reproduce the issue simply plug the ThingMagic device into the Pi 4, make sure its property registered by the Pi 4 and then just wait. For a more realistic scenario I can provide a script that calls the ThingMagic reader every minute but I was able to replicate the issue even without making calls to the ThingMagic reader.

The problem with testing this is that the issue does not surface immediately, sometimes you see it happening in the same day, sometimes things work great for 10 days and then everything starts crashing. But eventually, it does happen.

Here is the link where I previously reported the issue in case you guys are interested in pursuing this: https://github.com/raspberrypi/linux/issues/5046#issuecomment-1159818309

Thanks.

ReneElizondo, 2022-07-27:
> I've now got one in my possession. The device can't handle being suspended, and never responds to SETUP packets after a resume.
>
> You can reproduce this without a Pi 4 - using a powered USB hub that [...]

@P33M So I ended up reporting this issue to the guys that manufacture the ThingMagic device, not surprisingly they were not able to replicate the issue. By the way, we had no problem replicating the issue here with multiple USB hubs from multiple manufacturers but somehow for some reason, they could not.

The thing about this is that there are powered USB hubs that do not work (I think that is the majority) but there are indeed some powered USB hubs that actually do work. Do you happen to know of a way to detect if a powered USB hub would work or not based on something such as product description or perhaps based on the chip the USB hub uses? In other words, how can I know if the `powered USB hub does not switch its downstream port power`?

Perhaps if you give me the exact brand and model of the USB hub you tested this on I can send those guys the link so they can buy that one, I am asking you because I am pretty sure those guys are based in India and I wonder if USB hub brands like the one you have are more easily available on the other side of the world (I am based in the USA).

Thanks.

P33M, 2022-07-27:
Have they replicated the issue with a Raspberry Pi 4 Model B? The mode of failure is the same in either case, the hub test was specifically designed to eliminate the Pi as a source of misbehaviour.

ReneElizondo, 2022-07-27:
They did replicate the issue with the Pi 4, however, they claimed that they had no interest in fixing the issue because I was the only one having that issue and because the problem was specific to the Pi 4 (which that part is true).

However, now that we are able to reproduce the issue in Windows things are different, if the issue happens in Windows they may address the problem. To be honest, I don't care about Windows, what I care about is that if the problem is addressed in Windows it will automatically be addressed in the Pi. But before we get there, they need to be able to replicate the problem using the powered USB hub and so far no success.

Hey, do you happen to have the brand and model of the USB hub you used to replicate the issue?

Thanks.

P33M, 2022-07-28:
The hub was a Startech.com ST7300U3M.

ReneElizondo, 2022-08-11:
So in case anyone is curious about this, the company that makes the ThingMagic device presumably tested this issue using no less than five USB hubs:

- USB 3 Hub: Anker 4-Port Ultra Slim USB3.0 Data Hub. Model Number: A7516.
- USB 2.0 Hub Belkin F4U020
- USB 3.0 Hub [StarTech.com](https://urldefense.com/v3/__http:/StarTech.com__;!!FIRN_V0KSA!dyPoIIedOr51XDnuvyzJ4-LFmFxFCclMO8o4DWmw17ijfdxD85qeScyFS7tDFcFpw6pzO1-9zsIDnRkIR7VPoySpLMTJAA$) ST4300MINI
- USB 3.0 Hub: Dhruvga DHV-HUB-0107 USB 4 Port 
- USB 3.0 Hub: Anker 10-port USB hub Model Number: A7515

According to them, they tested this on two computers running Windows 10 and none of the hubs experience the issue. I happen to have the StarTech hub and tested this on three computers and the issue showed up on the tree computers....

The experiment was conducted as follows:

1.	Get the powered hub.
2.	Plug the USB hub to its power supply.
3.	Plug the ThingMagic device to the USB hub.
4.	Plug the USB hub to a computer.
5.	Unplug and re-plug the **_hub_** from the computer (**_not_** the ThingMagic device from the hub) a couple of times to see the error show up every time the hub is disconnected.

So tired of dealing with this.... about to throw the towel and call it a day.

symbios24, 2022-09-11:
> So in case anyone is curious about this, the company that makes the ThingMagic device presumably tested this issue using no less than five USB hubs:
>
> * USB 3 Hub: Anker 4-Port Ultra Slim USB3.0 Data [...]

Hi, you said in win 10 it works ok why you dont put windows 10 arm on the Raspberry Pi 4?

ReneElizondo, 2022-09-12:
Windows is not really a contender when it comes to embedded operating systems (everything is Linux based now). I am not even sure their are still seriously supporting their Windows IoT platform. If at some point in the future, I see the option to install Windows  from the "Raspberry pi imager" application, then maybe.... just maybe there will be a glimpse of  hope that "Windows is back". In the meantime, Windows on the Pi sound as realistic as Windows on a smart phone.

symbios24, 2022-09-12:
You can always try them, from what i read from reviews they don't perform bad if installed in ssd drive

P33M, 2022-12-19:
This device doesn't adhere to the USB spec, but there is a workaround - power cycling it after Linux comes back up. There won't be a software change to special-case this device.

closed 2022-12-19 by P33M

ReneElizondo, 2023-06-06:
No software fix, I essentially continued talking to the company that makes the device and after they were able to reproduce the issue on Windows using the USB hub suggested by P33M (Startech.com ST7300U3M - but most USB 3 based hubs will have the same issue), the company went back and did a firmware update on the device that fixed the issue.

Have you try connecting your Cyberpower UPS to a USB 3 based hub on Windows? If you do, I bet you will have the same issue even in Windows and if that is the case, the company that makes that device may consider fixing the product (if you are very lucky).

Thanks.

jedi-ml, 2023-08-20:
@ReneElizondo Did you find any way to fix this problem we have same problem with that case :(

jedi-ml, 2023-08-20:
@ReneElizondo 
Maybe I found solution :

1. using by kernel command line option , i.e. cmdline.txt add this option for disable authorization of usbcore 
usbcore.authorized_default=0
2. after booting , turn off usb port or hub using by uhubctl
sudo uhubctl -a on -p 1 -l 1-1
3. enable authorization  to hub
echo 1 | sudo tee -a /sys/bus/usb/devices/1-1/authorized
4. it will be detected, and give authorization to USB device 
echo 1 | sudo tee -a /sys/bus/usb/devices/1-1/1-1.1/authorized

:)

rogeriocassares, 2023-12-28:
Hello @jedi-ml ,

On Rpi 4B, these commands worked for me on port 3 of USB:
```bash
uhubctl -l 1-1 -a 0
uhubctl -l 1-1 -a 1
echo 1 | tee -a /sys/bus/usb/devices/1-1/authorized
echo 1 | tee -a /sys/bus/usb/devices/1-1.3/authorized
```

Then I start a *service* with this and started the RFID Tag Reader

buganini, 2024-06-14:
https://forums.raspberrypi.com/viewtopic.php?p=2212759&sid=80da380daffed86a3ff189cce108af17#p2212759
may be related

buganini, 2024-06-14:
uhubctl & authorized doesn't 100% work for me

my boot script is now

```shell
board=$(tr -d '\0' < /proc/device-tree/model)

if [[ "${board}" =~ "Pi 5" ]]; then
    # https://github.com/raspberrypi/linux/issues/5046
    inited=$(tr -d '\0' < /sys/bus/usb/devices/3-2/authorized)
    if [[ "${inited}" = "0" ]]; then
        sleep 7
        sudo uhubctl -l 1 -a 0
        sudo uhubctl -l 3 -a 0
        sudo uhubctl -l 1 -a 1
        sudo uhubctl -l 3 -a 1
        echo 1 | sudo tee -a /sys/bus/usb/devices/3-2/authorized
        echo 1 | sudo tee -a /sys/bus/usb/devices/3-2:1.0/authorized
        echo 1 | sudo tee -a /sys/bus/usb/devices/3-2:1.1/authorized

        sleep 5
        while true ; do
            sleep 1
            if dmesg | grep hci0 | grep -q timeout; then
                sudo reboot
                exit
            fi
            if hciconfig hci0 | grep -q Manufacturer; then
                break
            fi
        done

    fi
fi
```

not that it almost always fails without `sleep 5`