raspberrypi/linux #5054: Kernel OOPS undefined instruction: events_power_efficient fb_flashcursor

gorky, 2022-06-02:
### Describe the bug

Kernel Panic with crash.  Usually crash freezes both screen and log output, but this time a kernel panic was dumped to the syslog.  Attached.

When system is rebooted, will stay up for a couple of days before this crash occurs again.

### Steps to reproduce the behaviour

PI has been sitting for a day or two, attached to a KVM Switch, w/o the monitor attached.

### Device (s)

Raspberry Pi 2 Mod. B v1.2

### System

steve@rothman:~ $ raspinfo
-bash: raspinfo: command not found
steve@rothman:~ $ sudo raspinfo
[sudo] password for steve: 
sudo: raspinfo: command not found
steve@rothman:~ $ 

steve@rothman:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

steve@rothman:~ $ vcgencmd version
-bash: vcgencmd: command not found
steve@rothman:~ $ sudo vcgencmd version
sudo: vcgencmd: command not found

steve@rothman:~ $ uname -a
Linux rothman 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

### Logs

Jun  1 14:04:34 rothman kernel: [254304.183246] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
Jun  1 14:04:34 rothman kernel: [254304.183290] Modules linked in: xt_tcpudp nf_nat_ftp xt_REDIRECT xt_conntrack nf_conntrack_ftp xt_LOG nf_log_syslog xt_limit xt_multiport iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw ipt_REJECT nf_reject_ipv4 iptable_filter rpcsec_gss_krb5 sha256_generic cfg80211 rfkill 8021q garp stp llc sg vc4 joydev cec bcm2835_isp(C) drm_kms_helper uas bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 videobuf2_common snd_compress snd_pcm_dmaengine videodev syscopyarea sysfillrect snd_bcm2835(C) sysimgblt mc fb_sys_fops snd_pcm snd_timer snd vc_sm_cma(C) raspberrypi_hwmon uio_pdrv_genirq uio fixed drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
Jun  1 14:04:34 rothman kernel: [254304.183594] CPU: 1 PID: 1424 Comm: kworker/1:2 Tainted: G         C        5.15.32-v7+ #1538
Jun  1 14:04:34 rothman kernel: [254304.183609] Hardware name: BCM2835
Jun  1 14:04:34 rothman kernel: [254304.183619] Workqueue: events_power_efficient fb_flashcursor
Jun  1 14:04:34 rothman kernel: [254304.183650] PC is at drm_fbdev_fb_imageblit+0x0/0x40 [drm_kms_helper]
Jun  1 14:04:34 rothman kernel: [254304.184134] LR is at soft_cursor+0x1a0/0x20c
Jun  1 14:04:34 rothman kernel: [254304.184150] pc : [<7f2a138c>]    lr : [<806c555c>]    psr: 60000013
Jun  1 14:04:34 rothman kernel: [254304.184159] sp : 87377df0  ip : 81c7992f  fp : 87377e2c
Jun  1 14:04:34 rothman kernel: [254304.184165] r10: 81c79920  r9 : 826ce038  r8 : 00000001
Jun  1 14:04:34 rothman kernel: [254304.184172] r7 : 00000001  r6 : 81ff3000  r5 : 87377e4c  r4 : 826ce000
Jun  1 14:04:34 rothman kernel: [254304.184182] r3 : 7f2a138c  r2 : 826ce048  r1 : 826ce000  r0 : 81ff3000
Jun  1 14:04:34 rothman kernel: [254304.184190] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun  1 14:04:34 rothman kernel: [254304.184204] Control: 10c5387d  Table: 0b78806a  DAC: 00000055
Jun  1 14:04:34 rothman kernel: [254304.184209] Register r0 information: slab kmalloc-1k start 81ff3000 pointer offset 0 size 1024
Jun  1 14:04:34 rothman kernel: [254304.184251] Register r1 information: slab kmalloc-128 start 826ce000 pointer offset 0 size 128
Jun  1 14:04:34 rothman kernel: [254304.184276] Register r2 information: slab kmalloc-128 start 826ce000 pointer offset 72 size 128
Jun  1 14:04:34 rothman kernel: [254304.184298] Register r3 information: 67-page vmalloc region starting at 0x7f287000 allocated at load_module+0xb94/0x2840
Jun  1 14:04:34 rothman kernel: [254304.184333] Register r4 information: slab kmalloc-128 start 826ce000 pointer offset 0 size 128
Jun  1 14:04:34 rothman kernel: [254304.184359] Register r5 information: non-slab/vmalloc memory
Jun  1 14:04:34 rothman kernel: [254304.184371] Register r6 information: slab kmalloc-1k start 81ff3000 pointer offset 0 size 1024
Jun  1 14:04:34 rothman kernel: [254304.184396] Register r7 information: non-paged memory
Jun  1 14:04:34 rothman kernel: [254304.184405] Register r8 information: non-paged memory
Jun  1 14:04:34 rothman kernel: [254304.184414] Register r9 information: slab kmalloc-128 start 826ce000 pointer offset 56 size 128
Jun  1 14:04:34 rothman kernel: [254304.184447] Register r10 information: slab kmalloc-8k start 81c78000 pointer offset 6432 size 8192
Jun  1 14:04:34 rothman kernel: [254304.184472] Register r11 information: non-slab/vmalloc memory
Jun  1 14:04:34 rothman kernel: [254304.184481] Register r12 information: slab kmalloc-8k start 81c78000 pointer offset 6447 size 8192
Jun  1 14:04:34 rothman kernel: [254304.184506] Process kworker/1:2 (pid: 1424, stack limit = 0x994da2d0)
Jun  1 14:04:34 rothman kernel: [254304.184522] Stack: (0x87377df0 to 0x87378000)
Jun  1 14:04:34 rothman kernel: [254304.184535] 7de0:                                     00000010 80a45ee8 00000000 00000010
Jun  1 14:04:34 rothman kernel: [254304.184547] 7e00: 87377e34 81edda00 87377e4c 00000001 00000042 81ff3000 80f05008 00000020
Jun  1 14:04:34 rothman kernel: [254304.184561] 7e20: 87377ec4 87377e30 806c512c 806c53c8 8010ce38 8010bc64 00000002 00000002
Jun  1 14:04:34 rothman kernel: [254304.184573] 7e40: 00000720 00000007 80e8fc50 00000000 baae0001 85098fc0 00000000 00000078
[... 18 lines cut ...]
Jun  1 14:04:34 rothman kernel: [254304.184822]  r10:baae9305 r9:00000007 r8:00000002 r7:00000720 r6:814e8400 r5:806c4e38
Jun  1 14:04:34 rothman kernel: [254304.184829]  r4:81ff3000
Jun  1 14:04:34 rothman kernel: [254304.184834] [<806c00ec>] (fb_flashcursor) from [<8013cec4>] (process_one_work+0x250/0x57c)
Jun  1 14:04:34 rothman kernel: [254304.184863]  r9:00000000 r8:baae9300 r7:00000040 r6:baae5b40 r5:82b76500 r4:81ff31a8
Jun  1 14:04:34 rothman kernel: [254304.184868] [<8013cc74>] (process_one_work) from [<8013d250>] (worker_thread+0x60/0x5c4)
Jun  1 14:04:34 rothman kernel: [254304.184895]  r10:baae5b40 r9:80f03d00 r8:baae5b58 r7:00000008 r6:baae5b40 r5:82b76518
Jun  1 14:04:34 rothman kernel: [254304.184901]  r4:82b76500
Jun  1 14:04:34 rothman kernel: [254304.184905] [<8013d1f0>] (worker_thread) from [<80145580>] (kthread+0x178/0x194)
Jun  1 14:04:34 rothman kernel: [254304.184935]  r10:87376000 r9:852d1e74 r8:00000000 r7:82b76500 r6:8013d1f0 r5:84db1780
Jun  1 14:04:34 rothman kernel: [254304.184942]  r4:8ee74540
Jun  1 14:04:34 rothman kernel: [254304.184947] [<80145408>] (kthread) from [<801000d4>] (ret_from_fork+0x14/0x20)
Jun  1 14:04:34 rothman kernel: [254304.184964] Exception stack(0x87377fb0 to 0x87377ff8)
Jun  1 14:04:34 rothman kernel: [254304.184976] 7fa0:                                     00000000 00000000 00000000 00000000
Jun  1 14:04:34 rothman kernel: [254304.184987] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun  1 14:04:34 rothman kernel: [254304.184997] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jun  1 14:04:34 rothman kernel: [254304.185009]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80145408
Jun  1 14:04:34 rothman kernel: [254304.185016]  r4:84db1780
Jun  1 14:04:34 rothman kernel: [254304.185028] Code: e58d3099 ebffaa0d e24b8051 e89d39d8 (e1a03099)
Jun  1 14:04:34 rothman kernel: [254304.185041] ---[ end trace 768974e7daf6d487 ]---

### Additional context

System is running booted from an SD Card, but OS, etc. are on a NFS Drive.

5 more PIs with the same setup.  The PI3's are not having this lockup problem.

popcornmix, 2022-06-03:
What OS are you running?
Not having raspinfo/vcgencmd and the old date in "Raspberry Pi reference 2019-07-10" suggests it's not a recent RPiOS image.

gorky, 2022-06-03:
Raspian, don't remember what version was originally loaded. 
Debian Bullseye is what is currently in the /etc/apt/sources.list, so current with Bullseye.

popcornmix, 2022-06-06:
This isn't an issue I've seen reported elsewhere. "Oops - undefined instruction" suggests memory corruption of the kernel code. That can be caused by overclock or power supply issues.

Can you confirm you are seeing this on more than on Pi board?
Can you confirm if the crashes always have this backtrace ("Oops - undefined instruction" and drm_fbdev_fb_imageblit)?

Your image seems very old (possibly buster or even stretch) but with a recent kernel - do you use rpi-update?
Even with buster or stretch, vcgencmd should be on the path so it's possible things are broken there.
It might be worth setting up a clean bullseye system and seeing if you still see the issue?