raspberrypi/linux #5059: lost frames/incomplete draining in bcm2835-v4l2-codec.c

fduncanh, 2022-06-09:
### Describe the bug

There are two (related?) bugs in driver bcm2835-v4l2-codec.c working with the GStreamer-1.21 (current development) Video4Linux2 plugin where VIDIOC_DQBUF  does not dequeue a few frames that were previously successfully queued to the driver by VIDIOC_QBUF.    The bugs are associated with Caps renegotiations.

* GStreamer Video4Linux2 maintainer  **Nicolas Dufresne (ndufresne)** believes the evidence shows these are driver bugs, not GStreamer v4l2 bugs.

* Both produce the GStreamer-1.21 g_warning   "**v4l2h264dec0: Too old frames, bug in decoder -- please file a bug**" which is triggered by Garbage Collection in gst_v4l2_video_dec_loop of queued-but-undequeued frames, which should not occur.

*  The bugs are seen when using the Video4Linux2 plugin from the GStreamer-1.21 development branch (compiled from source) which is now finally working extremely well on Raspberry Pi OS (32/64 bit, Bullseye)  (apart from these annoying warnings from **each** lost frame)

1.    If the initial frame of the h264 bitstream starts with a SPS NAL,  a Caps renegotiation is triggered.  Frames 0,1,2... are successfully queued by VIDIOC_QBUF in gst_v4l2_allocator_qbuf in the OUTPUT thread (gst_v4l2_video_dec_handle_frame) but only frames x, x+1, x+2 ,... (where x is variable between frame 2 and frame 5, maybe a race condition) are dequeued  by VIDIOC_DQBUF in gst_v4l2_allocator_dqbuf in the CAPTURE thread (gst_v4l2_video_dec_loop).

2.   If a midstream frame with a SPS NAL is detected, and triggers CAPS renegotiation,  CMD_STOP is sent to the decoder by gst_v4l2_video_dec_finish, and all pending frames should be drained by VIDIOC_DQBUF in gst_v4l2_allocator_dqbuf, but sometimes (not always) the decoder sends V4L2_BUF_FLAG_LAST prematurely, and a few frames are left un-dequeued.    With the test video clip provided here, where a number of midstream Caps renegotiations occur, the number of "lost frames"  is different in each event (but does not seem to vary each time the video is played, unlike 1.)

This has been discussed (a bit confusedly, on my part) at

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/2518

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/2517

(These suggested  GStreamer  Merge Requests are cosmetic attempts to reduce the level of warning messages, not to fix the underlying bug which appears to be a bcm2835-v4l2-codec.c driver bug.)

### Steps to reproduce the behaviour

Use the latest https://gitlab.freedesktop.org/gstreamer/gstreamer  GStreamer code, built with latest meson 0.62.2 and ninja.

Use the pipeline 

```
#!/bin/sh
gst-launch-1.0 filesrc location=$1 ! video/x-h264 ! h264parse ! capssetter caps="video/x-h264, colorimetry=bt709" ! v4l2h264dec ! v4l2convert ! kmssink sync=false
```
where $1 is the test h264 video clip
 https://gitlab.freedesktop.org/gstreamer/gstreamer/uploads/f98eab8fe15d2f03df3001febdc63cf6/videodump.h264

* This is a 20sec capture from  Big Buck Bunny streamed from YouTube to an Apple iPad, then re-streamed at 30fps with Apple AirPlay to an Airplay server running on Raspberry Pi model 4B 64bit , which renders video with GPU decoding using Video4Linux2 and bcm2835. The bitstream  was captured as it was sent into the Gstreamer video pipeline. 

* The capssetter in the pipeline is needed because Apple's colorimetry 1:3:5:1 is not recognized by GStreamer during Caps negotiation, and bt709 = 2:3:5:1 works.   

*  The h264 clip  has frames 0-712.   Frames 0,  114,  270, 454, 560  contain a type 7 (SPS) + a type 8 (PPS) + a type 5 (VCL) NAL, other frames just contain a single type-1 (VCL) NAL.    (frames are type MPLANE).   The midstream SPS NALs occur when the iPad is rotated to change the picture between Landscape and Portrait.  (changes image width, height is maintained at 1080)

By adding code to monitor the timestamp in V4l2 buffers as they are queued (VIDIOC_QBUF) and dequeued (VIDIOC_DQBUF)  by the gstv4l2allocator, and checking pending frames before and after draining, I get the following frame history:

```
Frame history   (sample video videodump.h264 has 712 frames):

1.    0-113  Landscape     lost first few frames : 0-x,  x can be 2,3 or 4   last frames OK without loss.
2.  114-269  Portrait      lost last 6 frames  : 264-269
3.  270-453  Landscape     lost last 3 frames  : 451-453
4.  454-559  Portrait      lost last 6 frames  : 554-559
5.  560-712  Landscape     lost last 2 frames  : 711-712

Symptom:
segment  1:  Landscape  first frame dequeued by VIDIOC_DQBUF = x+1 (should be 0), last frame dequeued 113.
segment  2:  Portrait   first frame dequeued by VIDIOC_DQBUF = 114, last frame dequeued 263 (should be 269)
segment  3:  Landscape  first frame dequeued by VIDIOC_DQBUF = 270, last frame dequeued 450 (should be 453)
segment  4:  Portrait   first frame dequeued by VIDIOC_DQBUF = 454, last frame dequeued 553 (should be 559)
segment  5:  Landscape  first frame dequeued by VIDIOC_DQBUF = 560, last frame dequeued 710 (should be 712)
```

The GStreamer code uses "Garbage Collection" to drop any  pending frame that is more than 100 frames older than the current frame being rendered,  with an un-hideable  warning message for each of these "lost frames".

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$cat /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

$vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

$uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

$lsmod
Module                  Size  Used by
cmac                   16384  3
algif_hash             20480  1
aes_arm64              16384  3
algif_skcipher         20480  1
af_alg                 32768  6 algif_hash,algif_skcipher
bnep                   28672  2
hci_uart               49152  1
btbcm                  28672  1 hci_uart
bluetooth             450560  26 hci_uart,btbcm,bnep
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
8021q                  32768  0
garp                   16384  1 8021q
stp                    20480  1 garp
llc                    20480  2 stp,garp
snd_soc_hdmi_codec     20480  2
joydev                 24576  0
brcmfmac              331776  0
brcmutil               24576  1 brcmfmac
bcm2835_codec          49152  0
bcm2835_v4l2           45056  0
vc4                   303104  6
bcm2835_isp            32768  0
cfg80211              864256  1 brcmfmac
cec                    53248  1 vc4
v3d                    86016  0
rfkill                 36864  6 bluetooth,cfg80211
bcm2835_mmal_vchiq     36864  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
gpu_sched              40960  1 v3d
drm_kms_helper        282624  3 vc4
v4l2_mem2mem           40960  1 bcm2835_codec
raspberrypi_hwmon      16384  0
videobuf2_dma_contig    24576  2 bcm2835_codec,bcm2835_isp
videobuf2_vmalloc      20480  1 bcm2835_v4l2
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       65536  8 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
snd_soc_core          253952  2 vc4,snd_soc_hdmi_codec
i2c_brcmstb            16384  0
snd_compress           20480  1 snd_soc_core
snd_bcm2835            24576  1
snd_pcm_dmaengine      20480  1 snd_soc_core
videodev              262144  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
snd_pcm               135168  5 snd_bcm2835,snd_soc_hdmi_codec,snd_compress,snd_soc_core,snd_pcm_dmaengine
mc                     57344  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
snd_timer              36864  1 snd_pcm
vc_sm_cma              36864  2 bcm2835_mmal_vchiq,bcm2835_isp
rpivid_mem             16384  0
snd                   102400  12 snd_bcm2835,snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
nvmem_rmem             16384  0
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
drm                   573440  5 gpu_sched,drm_kms_helper,v3d,vc4
fuse                  135168  1
drm_panel_orientation_quirks    24576  1 drm
backlight              20480  2 drm_kms_helper,drm
ip_tables              32768  0
x_tables               45056  1 ip_tables
ipv6                  552960  42

```

6by9, 2022-06-16:
(Having some woes building GStreamer as FFmpeg wants atomic operations that aren't supported by the normal Raspberry Pi OS gcc. https://trac.ffmpeg.org/ticket/9275 and https://git.videolan.org/?p=ffmpeg.git;a=commitdiff;h=2f0a214a6202516b4dda2bb22b6b3ac20e465d6d)

AFAIK the codec should be flushing the buffers back. CMD_STOP passes an EOS flag into the codec, and that should be returned when the codec has emptied the internal FIFOs. I'll have to look at logging once I have a build.

6by9, 2022-06-16:
Enabling the driver logging via module parameter ```debug``` set to 5, I get the logging:
```
GST_Codec$ grep -e op_buffer_cb -e "Submitted ip" gst_log.txt | grep pts
[17041.593876] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 20498, pts 0, flags 0004
[17041.594897] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 980, pts 1000000, flags 0004
[17041.596588] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 8592, pts 2000000, flags 0004
[17041.596801] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 4672, pts 3000000, flags 0004
[17041.597194] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 5374, pts 4000000, flags 0004
[17041.597383] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 773, pts 5000000, flags 0004
[17041.597579] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 2027, pts 6000000, flags 0004
[17041.598159] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 18775, pts 7000000, flags 0004
[17041.598678] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 19807, pts 8000000, flags 0004
[17041.599773] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 22906, pts 9000000, flags 0004
[17041.600248] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 11876, pts 10000000, flags 0004
[17041.602150] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 21747, pts 11000000, flags 0004
[17041.603063] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 45931, pts 12000000, flags 0004
[17041.603374] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 23400, pts 13000000, flags 0004
[17041.605035] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 25903, pts 14000000, flags 0004
[17041.605519] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 24510, pts 15000000, flags 0004
[17041.607394] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 24000, pts 16000000, flags 0004
[17041.610997] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 24867, pts 17000000, flags 0004
[17041.613986] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:fc9f5d98, length:96, flags 0000, pts -9223372036854775808
```
This is the format change event
```
[17041.624836] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:e091a542, length:2350080, flags 0004, pts 0
[17041.630610] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:2fcc7b2c, length:2350080, flags 0004, pts 1000000
[17041.636910] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:17c9ff30, length:2350080, flags 0004, pts 2000000
[17041.642241] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:317a2701, length:2350080, flags 0004, pts 3000000
[17041.647626] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:52ba8b1c, length:2350080, flags 0004, pts 4000000
[17041.672197] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 78170, pts 18000000, flags 0004
[17041.676909] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:80393b2c, length:2350080, flags 0004, pts 5000000
[17041.683134] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:b53a40a3, length:2350080, flags 0004, pts 6000000
[17041.685595] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 100821, pts 19000000, flags 0004
[17041.688676] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:8b4ac413, length:2350080, flags 0004, pts 7000000
[17041.693962] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:fd3609f2, length:2350080, flags 0004, pts 8000000
[17041.699693] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:c9c04209, length:2350080, flags 0004, pts 9000000
[17041.704039] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 105847, pts 20000000, flags 0004
[17041.711825] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 108889, pts 21000000, flags 0004
[17041.717334] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 113313, pts 22000000, flags 0004
[17041.719292] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 101767, pts 23000000, flags 0004
[17041.721872] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 92820, pts 24000000, flags 0004
[17041.723837] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 121786, pts 25000000, flags 0004
[17041.726785] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 116313, pts 26000000, flags 0004
[17041.728354] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 95225, pts 27000000, flags 0004
[17041.729463] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 83321, pts 28000000, flags 0004
[17041.730470] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 78553, pts 29000000, flags 0004
[17041.731811] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 63947, pts 30000000, flags 0004
[17041.732792] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 58970, pts 31000000, flags 0004
[17041.733686] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 57364, pts 32000000, flags 0004
[17041.733985] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 36699, pts 33000000, flags 0004
[17041.734835] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 33192, pts 34000000, flags 0004
[... cut ...]
[17045.580564] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 80206, pts 168000000, flags 0004
[17045.582177] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 78119, pts 169000000, flags 0004
[17045.583170] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
```
Another large buffer and weird pts.
Again FFprobe -show_packets doesn't list it
```
size=80200 (pts 168000000)
size=78113 (pts 169000000)
size=73097 (pts 170000000)
size=80790 (pts 171000000)
size=78110 (pts 172000000)
```

6by9, 2022-06-16:
```
[17045.583382] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 73103, pts 170000000, flags 0004
[17045.585082] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 80796, pts 171000000, flags 0004
[17045.586080] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 78116, pts 172000000, flags 0004
[17045.586289] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:2d8c0ce1, length:1536000, flags 0004, pts 3294966
[17045.587495] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 73168, pts 173000000, flags 0004
[17045.590786] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 80491, pts 174000000, flags 0004
[17045.591164] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:023ead59, length:1357824, flags 0004, pts 120000000
[17045.595300] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 77832, pts 175000000, flags 0004
[17045.598032] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 73003, pts 176000000, flags 0004
[17045.600246] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 81389, pts 177000000, flags 0004
[17045.603872] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 78655, pts 178000000, flags 0004
[17045.608750] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 73825, pts 179000000, flags 0004
[17045.612039] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 82356, pts 180000000, flags 0004
[17045.612177] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.615004] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:63c66d76, length:1536000, flags 0004, pts 3294966
```
We now get a load of op_buffer_cb back for pts 3294966, implying that the input buffer had multiple frames inside it.
(Whilst we want one NAL per frame, we don't enforce it and will decode all data given to us)
```
[17045.615710] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 77376, pts 181000000, flags 0004
[17045.619199] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:22175fc1, length:1357824, flags 0004, pts 121000000
[17045.620677] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 72939, pts 182000000, flags 0004
[17045.623939] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 81206, pts 183000000, flags 0004
[17045.628350] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 79052, pts 184000000, flags 0004
[17045.644710] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.647518] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:e0d4f601, length:1536000, flags 0004, pts 3294966
[17045.651526] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:374af6f3, length:1357824, flags 0004, pts 122000000
[17045.678776] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.681569] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:5338e836, length:1536000, flags 0004, pts 3294966
[17045.687101] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:fe87b074, length:1357824, flags 0004, pts 123000000
[17045.697962] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 73762, pts 185000000, flags 0004
[17045.711568] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.714365] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:2d8c0ce1, length:1536000, flags 0004, pts 3294966
[17045.718404] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:edd0c65b, length:1357824, flags 0004, pts 124000000
[17045.745043] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.747843] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:63c66d76, length:1536000, flags 0004, pts 3294966
[17045.751911] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:023ead59, length:1357824, flags 0004, pts 125000000
[17045.779080] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.781877] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:e0d4f601, length:1536000, flags 0004, pts 3294966
[17045.786590] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:22175fc1, length:1357824, flags 0004, pts 126000000
[17045.811849] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.814642] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:5338e836, length:1536000, flags 0004, pts 3294966
[17045.818617] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:374af6f3, length:1357824, flags 0004, pts 127000000
[17045.845128] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.847932] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:2d8c0ce1, length:1536000, flags 0004, pts 3294966
[17045.851930] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:fe87b074, length:1357824, flags 0004, pts 128000000
[17045.879109] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 1347840, pts 3294966, flags 0004
[17045.881902] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:63c66d76, length:1536000, flags 0004, pts 3294966
[... cut ...]
Looking at the QBUF/DQBUF logging by setting dev_debug to 0x0b, it doesn't show a QBUF with these weird timestamps, so I'm now puzzled as to how our call to ```v4l2_m2m_buf_remove``` in [```device_run```](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L1323) is giving us such a weird buffer. Those queues are all handled by the v4l2_m2m framework.
It's as if we have two independent decodes happening at once, as the odd PTS values are also of size 1536000 bytes instead of 1357824 bytes. I also have two independent counts of the number of output buffers

6by9, 2022-06-16:
(Sorry for splitting the above in 2 - Github did a funny, and it won't allow me to attach the logs either at the moment)

6by9, 2022-06-16:
Doh, ```v4l2convert``` is using the same bcm2835-codec code as ```v4l2h264dec```. That explains the multiple instances that don't show up in /dev/video10's V4L2 logging, as v4l2convert uses /dev/video12.

Dropping v4l2convert works initially as both v4l2h264dec and kmssink can handle NV12, but it then throws a wobbler when the resolution change comes along.
```
pi@raspberrypi:~ $ gst-launch-1.0  -vvv filesrc location=videodump.h264 ! video/x-h264 ! h264parse ! capssetter caps="video/x-h264, colorimetry=bt709" ! v4l2h264dec ! kmssink sync=false
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstKMSSink:kmssink0: display-width = 1280
/GstPipeline:pipeline0/GstKMSSink:kmssink0: display-height = 800
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1440, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)1:3:5:1, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.2
/GstPipeline:pipeline0/GstCapsSetter:capssetter0.GstPad:src: caps = video/x-h264, width=(int)1440, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.2
/GstPipeline:pipeline0/v4l2h264dec:v4l2h264dec0.GstPad:sink: caps = video/x-h264, width=(int)1440, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.2
/GstPipeline:pipeline0/GstCapsSetter:capssetter0.GstPad:sink: caps = video/x-h264, width=(int)1440, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)1:3:5:1, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.2
/GstPipeline:pipeline0/v4l2h264dec:v4l2h264dec0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1440, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt601, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstKMSSink:kmssink0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1440, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt601, framerate=(fraction)0/1
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)810, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)1:3:5:1, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.1
/GstPipeline:pipeline0/GstCapsSetter:capssetter0.GstPad:src: caps = video/x-h264, width=(int)810, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.1
0:00:00.0 / 99:99:99.
** (gst-launch-1.0:11233): WARNING **: 16:33:46.543: v4l2h264dec0: Too old frames, bug in decoder -- please file a bug
0:00:00.0 / 99:99:99.
** (gst-launch-1.0:11233): WARNING **: 16:33:46.576: v4l2h264dec0: Too old frames, bug in decoder -- please file a bug
** (gst-launch-1.0:11233): WARNING **: 16:33:46.610: v4l2h264dec0: Too old frames, bug in decoder -- please file a bug
** (gst-launch-1.0:11233): WARNING **: 16:33:46.643: v4l2h264dec0: Too old frames, bug in decoder -- please file a bug
0:00:00.0 / 99:99:99.
** (gst-launch-1.0:11233): WARNING **: 16:33:46.676: v4l2h264dec0: Too old frames, bug in decoder -- please file a bug
/GstPipeline:pipeline0/v4l2h264dec:v4l2h264dec0.GstPad:sink: caps = video/x-h264, width=(int)810, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)bt709, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)high, level=(string)4.1
/GstPipeline:pipeline0/GstCapsSetter:capssetter0.GstPad:sink: caps = video/x-h264, width=(int)810, height=(int)1080, framerate=(fraction)0/1, coded-picture-structure=(string)frame, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)1:3 [... cut ...]
0:00:04.932840755 11233  0x2138040 ERROR           v4l2videodec gstv4l2videodec.c:898:gst_v4l2_video_dec_handle_frame:<v4l2h264dec0> not negotiated
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: Internal data stream error.
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbaseparse.c(3681): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH264Parse:h264parse0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:03.672763335
Setting pipeline to NULL ...
Freeing pipeline ...
```

Using ```videoconvert``` works, so there's something odd in requiring a conversion.

fduncanh, 2022-06-17:
Looking at your buffer llen output I see that all frames are 6 bytes bigger than the sum of their NALs  in the test video bitstream, but I see in the GST_DEBUG=7 output from GStreamer that h264parse inserts  an  AUD  into the bitstream at the end of  each frame, and gstreamer agrees with the sizes reported by bcm2835-codec for the VIDIOC_QBUF.
```
GST_Codec$ grep -e op_buffer_cb -e "Submitted ip" gst_log.txt | grep pts
[17041.593876] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 20498, pts 0, flags 0004
[17041.594897] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 980, pts 1000000, flags 0004
[17041.596588] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 8592, pts 2000000, flags 0004
[17041.596801] bcm2835-codec bcm2835-codec: device_run: Submitted ip buffer len 4672, pts 3000000, flags 0004

fduncanh, 2022-06-17:
As noted in the bug report, there are two distinct issues: the  (1) why the the first few frames after the initial frame with SPS never get dequeued by VIDIOC_DQBUF, and (2) why (except maybe the first time CMD_STOP is sent) a few frame are left pending aftre the draining finishes (VIDIOC_DQBUF sends what seems to be a premature eos signal.) 

I have a second test video with no midstream SPS changes, which shows only error (1), and drains all frames correctly the first time CMD_STOP is send, this time at the end of the video.     This is only 2 examples, so I don't know if it is always the case that the first draining works and subsequent ones are incomplete. 

Possibly it might be easier to find why the draining procedure is (sometimes) incomplete. - error 2.

6by9, 2022-06-17:
Hmm
```
[ 2329.864376] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:71c9e5ab, length:96, flags 0000, pts -9223372036854775808
[ 2329.864391] bcm2835-codec bcm2835-codec: handle_fmt_changed: Format changed: buff size min 2350080, rec 2350080, buff num min 1, rec 1
[ 2329.864399] bcm2835-codec bcm2835-codec: handle_fmt_changed: Format changed to 1440x1088, crop 1440x1080, colourspace 49464A59
[ 2329.864408] bcm2835-codec bcm2835-codec: handle_fmt_changed: Format was 1440x1088, crop 1440x1080
[ 2329.868208] bcm2835-codec bcm2835-codec: handle_fmt_changed: interlace mode 0, v4l2 field 1
[ 2329.869272] video10: VIDIOC_DQEVENT: type=0x5, pending=0, sequence=0, id=0, timestamp=2329.812584024
[ 2329.870342] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: type: 9 - return buffers
[ 2329.872925] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:c5dcfecd, length:2350080, flags 0004, pts 0
[ 2329.872952] bcm2835-codec bcm2835-codec: op_buffer_cb: length 2350080, flags 4, idx 0
[ 2329.872971] bcm2835-codec bcm2835-codec: op_buffer_cb: done 1 output buffers
[ 2329.879341] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:7cf24288, length:2350080, flags 0004, pts 1000000
[ 2329.879353] bcm2835-codec bcm2835-codec: op_buffer_cb: length 2350080, flags 4, idx 1
[ 2329.879363] bcm2835-codec bcm2835-codec: op_buffer_cb: done 2 output buffers
[ 2329.884457] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:67d70acc, length:2350080, flags 0004, pts 2000000
[ 2329.884467] bcm2835-codec bcm2835-codec: op_buffer_cb: length 2350080, flags 4, idx 2
[ 2329.884479] bcm2835-codec bcm2835-codec: op_buffer_cb: done 3 output buffers
[ 2329.889868] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:2c1be03d, length:2350080, flags 0004, pts 3000000
[ 2329.889894] bcm2835-codec bcm2835-codec: op_buffer_cb: length 2350080, flags 4, idx 3
[ 2329.889906] bcm2835-codec bcm2835-codec: op_buffer_cb: done 4 output buffers
[ 2329.895564] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:e5e42683, length:2350080, flags 0004, pts 4000000
[ 2329.895574] bcm2835-codec bcm2835-codec: op_buffer_cb: length 2350080, flags 4, idx 4
[ 2329.895583] bcm2835-codec bcm2835-codec: op_buffer_cb: done 5 output buffers
[ 2329.895946] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: done
[ 2329.895961] video10: VIDIOC_STREAMOFF: type=vid-cap-mplane
[ 2329.896123] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6d83b1f8, vb e07e2688
[ 2329.896137] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6d83b1f8, vb aef92a4f
[ 2329.896147] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6d83b1f8, vb 5b653572
[ 2329.896156] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6d83b1f8, vb 615df263
[ 2329.896165] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:6d83b1f8, vb 7f0296d7
[ 2329.896180] video10: VIDIOC_REQBUFS: count=0, type=vid-cap-mplane, memory=mmap
```
So they have been decoded already (presumably the output buffer was of an appropriate size), but because it's doing a flush in stop_streaming, they're being discarded. I thought I'd been through this loop before. Presumably the CAPTURE queue format has been set to the format, so the buffers are of an appropriate size.
That's a quick one to fix in the firmware.

 I wonder if error 2 is similar. I'm investigating.

fduncanh, 2022-06-17:
@6by9 

I tested with 3 more examples.
For error 2.

In each case the **first** CMD_STOP draining works perfectly.   If the last pending frame is e.g. 209, the last VIDIOC_DQBUF
returns an empty  buffer with ts = 209 033 333 000  and the V4L2_BUF_FLAG_LAST flag set, terminating the drain.

All later CMD_STOP events leave some undrained frames.   The number lost seems to be correlated with the frame size in bytes which is varying between smaller and bigger  as  the image changes between portrait and  landscape  at fixed pixel height.

This seems to suggest something in the firmware is left unreset after a drain finishes, and affects  subsequent drainings?

fduncanh, 2022-06-18:
@6by9

```
[87401.242673] bcm2835-codec bcm2835-codec: op_buffer_cb: done 229 output buffers
[87401.257050] bcm2835-codec bcm2835-codec: ip_buffer_cb: port 000000004babdc68 buf 000000007a67d6d4 length 0, flags 0
[87401.257072] bcm2835-codec bcm2835-codec: ip_buffer_cb: no error. Return buffer 0000000080ba05ac
[87401.257088] bcm2835-codec bcm2835-codec: ip_buffer_cb: done 270 input buffers
[87401.269965] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 00000000f0464d8d
[87401.269988] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 00000000f0464d8d vbuf->flags 0, seq 0, bytesused 1357824
[87401.270018] bcm2835-codec bcm2835-codec: device_run: off we go
[87401.270034] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[87401.270068] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 00000000f0464d8d
[87401.274378] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:0000000092ff66a3, length:1357824, flags 0004, pts 228000000
[87401.274409] bcm2835-codec bcm2835-codec: op_buffer_cb: length 1357824, flags 4, idx 4
[87401.274430] bcm2835-codec bcm2835-codec: op_buffer_cb: done 230 output buffers
[87401.303295] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 00000000ff462e88
[87401.303322] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 00000000ff462e88 vbuf->flags 0, seq 0, bytesused 1357824
[87401.303351] bcm2835-codec bcm2835-codec: device_run: off we go
[87401.303367] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[87401.303400] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 00000000ff462e88
```
* below is second CMD_STOP, when frame 270  input buffer is found to have a SPS

```
[87401.305084] bcm2835-codec bcm2835-codec: vidioc_decoder_cmd, cmd 1
[87401.307694] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:00000000d7cdb61b, length:1357824, flags 0004, pts 229000000
[87401.307720] bcm2835-codec bcm2835-codec: op_buffer_cb: length 1357824, flags 4, idx 0
[87401.307740] bcm2835-codec bcm2835-codec: op_buffer_cb: done 231 output buffers
[87401.307755] bcm2835-codec bcm2835-codec: ip_buffer_cb: port 000000004babdc68 buf 0000000054f0c84d length 0, flags 0
[87401.307773] bcm2835-codec bcm2835-codec: ip_buffer_cb: eos buffer returned.
[87401.336617] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 0000000052358f57
[87401.336644] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 0000000052358f57 vbuf->flags 0, seq 0, bytesused 1357824

<snip>
```

* here are the frame losses after the premature  draining from SECOND CMD_STOP event when frame 270  starts a new  resolution  1440 x1088 old length (portrait )1357824, new length (landscape) 2350080:
```
[87402.436656] bcm2835-codec bcm2835-codec: device_run: off we go
[87402.436672] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[87402.436701] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 00000000f0464d8d
[87402.440962] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:0000000092ff66a3, length:1357824, flags 0004, pts 263000000
[87402.440989] bcm2835-codec bcm2835-codec: op_buffer_cb: length 1357824, flags 4, idx 4
[87402.441009] bcm2835-codec bcm2835-codec: op_buffer_cb: done 265 output buffers
[87402.469941] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 00000000ff462e88
[87402.469968] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 00000000ff462e88 vbuf->flags 0, seq 0, bytesused 1357824
[87402.469994] bcm2835-codec bcm2835-codec: device_run: off we go
[87402.470010] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[87402.470039] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 00000000ff462e88
[87402.470172] bcm2835-codec bcm2835-codec: op_buffer_cb: status:0, buf:00000000d7cdb61b, length:0, flags 0001, pts 263033333
[87402.470198] bcm2835-codec bcm2835-codec: op_buffer_cb: length 0, flags 1, idx 0
[87402.470212] bcm2835-codec bcm2835-codec: op_buffer_cb: Empty buffer - flags 0001
[87402.470224] bcm2835-codec bcm2835-codec: Sending EOS event
```

* Above is the EOS event after DQBUF of frame 263 (the last frame to properly Drain  after the SPS in frame 270 is detected)

*below the empty EOS buffer plus buffers 264, 265, 266, 267, 268, 269 are dropped  (7 cleanups)
```
[87402.470244] bcm2835-codec bcm2835-codec: op_buffer_cb: done 266 output buffers
[87402.503275] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 0000000052358f57
[87402.503302] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 0000000052358f57 vbuf->flags 0, seq 0, bytesused 1357824

[87402.503331] bcm2835-codec bcm2835-codec: device_run: off we go
[87402.503346] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[87402.503376] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 0000000052358f57
[87402.536604] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 0000000016dd5261
[87402.536629] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 0000000016dd5261 vbuf->flags 0, seq 0, bytesused 1357824

[87402.536656] bcm2835-codec bcm2835-codec: device_run: off we go
[... 40 lines cut ...]
```
[87402.574393] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000049f93a0, vb 00000000ff462e88
[... cut ...]
[87402.654127] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_init: ctx:00000000049f93a0, vb 00000000f0464d8d
[87402.656771] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_init: ctx:00000000049f93a0, vb 00000000c311515c
[87402.660686] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_init: ctx:00000000049f93a0, vb 0000000016dd5261
[87402.665224] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_init: ctx:00000000049f93a0, vb 0000000052358f57
[87402.667827] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_init: ctx:00000000049f93a0, vb 00000000ff462e88

```

fduncanh, 2022-06-22:
The strange thing is that the first CMD_STOP always seems to work perfectly, and all the subsequent ones send EOS early. 
This is in 6/6 examples.   I could try more  but it seems significant.

fduncanh, 2022-07-03:
@6by9 

To reduce the error mesage spamming, I got updates to gstreamer development  branch(1.21)  merged to cleanup all frames in  groups of dropped frames dropped  with a single g_warning each time one of these errors occurs.    The two different errors now generate different error messages which now report how many frames were lost, and what range of system frame numbers were lost.

Here are the new error messages generated by the test video clip I provided

```
** (gst-launch-1.0:147904): WARNING **: 00:52:36.960: v4l2h264dec0: 5 initial frames before frame 5 were not dequeued: bug in decoder -- please file a bug

** (gst-launch-1.0:147904): WARNING **: 00:52:45.715: v4l2h264dec0: 6 frames 264-269 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug

** (gst-launch-1.0:147904): WARNING **: 00:52:51.881: v4l2h264dec0: 3 frames 451-453 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug

** (gst-launch-1.0:147904): WARNING **: 00:52:55.380: v4l2h264dec0: 6 frames 554-559 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug

** (gst-launch-1.0:147904): WARNING **: 00:53:00.497: v4l2h264dec0: 2 frames 711-712 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug

```

popcornmix, 2022-07-05:
The latest rpi-update firmware contains a potential fix for this issue.

6by9, 2022-07-05:
> The latest rpi-update firmware contains a potential fix for this issue.

It's only a fix for the first error where a colourspace only change resulted in the initial buffers being lost.
The second error is still being investigated.

fduncanh, 2022-07-05:
@6by9 thanks for fixing the  colorspace-only change issue!

It seems to me that the key clue in the unresolved incomplete-draining issue is that the first drain event always works perfectly, but all later ones somehow send eos too soon.    Maybe something is left set after the first draining event that should have been unset?   clearly some memory of earlier drainings seems to remain..

engnfrc, 2022-08-23:
Are there any updates to this issue? It's been a while since the last update. I have a post on the RPi forums (https://forums.raspberrypi.com/viewtopic.php?p=2031719#p2031719) where I've been talking with @6by9 and an email out to the Gstreamer Devel Listserv <gstreamer-devel@lists.freedesktop.org> where I'm talking with **Nicolas Dufresne (ndufresne)** and would really like to find a resolution.  It sounds like this issue may be a contributing factor?

engnfrc, 2022-08-23:
Example pipeline that works with one make/model of camera, but not this camera (ELP-USBFHD01M):
`GST_DEBUG=*:4 GST_DEBUG_DUMP_DOT_DIR=$PWD/pipelines gst-launch-1.0 -e -vvv v4l2src device=/dev/videoHEL1 num-buffers=1 ! image/jpeg,width=1280,height=720,framerate=60/1 ! jpegparse ! identity name=identity_pre silent=false ! v4l2jpegdec ! identity name=identity_post silent=false ! filesink location=ELP-USBFHD01M_1frame.raw >ELP-USBFHD01M_1frame_raw.log 2>&1`
[image]
[Pipeline - Raw.zip](https://github.com/raspberrypi/linux/files/9406955/Pipeline.-.Raw.zip)
[Pipeline - JPEG.zip](https://github.com/raspberrypi/linux/files/9406957/Pipeline.-.JPEG.zip)

6by9, 2022-08-23:
Fix for issue 1 has been available for a while.
Sorry, other priorities have stopped me looking at issue 2. It's still on the list, but as I've already said, the issue appears to be buried fairly well down in the codec code.

6by9, 2022-08-23:
I've just noticed that wasn't fduncanh asking for a status update.

This issue is ONLY related to video decode on resolution change with H264 (or MPEG4, H263, MPEG2, or VC-1 on Pi0-3) losing buffers.
JPEG (or more accurately MJPEG in V4L2) goes through a different codec block and is going to be unrelated.

engnfrc, 2022-08-23:
Thanks! Based on that feedback I will generate a new issue.

fduncanh, 2023-01-24:
Hi @6by9, 
 any chance that this  "frames_left_undrained" issue (issue 2)  is in the kernel driver, or will it be in the firmware?

Now that GStreamer 1.22 is  out (which I wont be supplying v4l2 patches for,  that also silence the warning, since v1.22 is fully fixed, at least to work with uxplay) this bug will produce a GStreamer  error message each time the draining is incomplete.

(like this:)

```
WARNING ** 18:25:17.769: v4l2h264dec1: 2 frames 5135-5136 left undrained after CMD_STOP,  eos sent too early:  bug in decoder -- please file a bug
```

6by9, 2023-01-25:
> any chance that this "frames_left_undrained" issue (issue 2) is in the bcm2835_codec kernel driver, or will it be in the firmware?

From previous digging I believe it is in the firmware, and fairly well buried down in the codec parsing code which I don't really know my way around.