raspberrypi/linux #5061: volume levels set by amixer does not correspond to what alsamixer shows

jose1711, 2022-06-11:
### Describe the bug

If I set a volume of `PCM` control to 50% using `amixer` I expect `alsamixer` to mirror the same value. Unless `pulseaudio` is running though this is not the case.

### Steps to reproduce the behaviour

1. Make sure `pulseaudio` is *not* running
2. Run `amixer set PCM 50%`
3. Check in `alsamixer`
4. `alsamixer` shows 13% (instead of 50)

### Device (s)

Raspberry Pi 3 Mod. B

### System

```
Raspberry Pi reference 2017-09-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 496e41575eeb9fa13f394ffb407b7bc1d00b21c2, stage5
Aug 15 2019 12:08:48 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start_x)
Linux malina1 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

### Additional context

One may use the following exponential function to map values set in `amixer` to values shown in `alsamixer`:
```py
x=50; exp(0.0403*x)*1.764817  # 13.23741119471247
```
The problem vanishes when pulseaudio takes care of the audio.
The same issue has been reported also at:
- https://quabr.com/57373138/raspberry-pi-3-1-jack-sound-too-low-amixer-percentage-working-worng
- https://forums.raspberrypi.com/viewtopic.php?t=198553 (french)

pelwell, 2022-06-12:
Does this apply to all sound devices or just one in particular?

jose1711, 2022-06-12:
> Does this apply to all sound devices or just one in particular?

Just tried another device - a USB sound card. It's not the same but similar, e. g. `amixer -c2 sset PCM 50%` sets 37 % in `alsamixer` for instance.

jose1711, 2022-06-12:
Sorry, nothing to do with kernel/ALSA (`amixer -M` fixed the problem).

closed 2022-06-12 by jose1711