raspberrypi/linux #5062: HiFiBerry DAC2 HD defective dtoverlay in its eeprom

seanmiddleton99, 2022-06-14:
### Describe the bug

It seems that this DAC  contains a defective dtoverlay in its eeprom. Its dtoverlay doent match with the kernel details. I made HiFiberry aware of this and they agree that this is the case.

### Steps to reproduce the behaviour

In my case Raspberry Pi i need to inculde the following comands in my config.txt file to prefetn the eeprom from loading:

dtoverlay=hifiberry-dacplushd
force_eeprom_read=0

### Device (s)

Raspberry Pi 3 Mod. B

### System

TinyCore Linux

pelwell, 2022-06-14:
You have come across a known problem - the built-in overlay is no longer compatible with the latest DTB files. It is possible to encode the _name_ of an overlay into the EEPROM (this is now the recommended method to get a HAT's overlay loaded), but that doesn't help you.

Short of reflashing the EEPROM there isn't really any way to fix the problem, but you can improve the workaround a bit. The dtoverlay command is the correct way to load the updated overlay, but disabling the EEPROM reading is a bit of a blunt instrument. All you really need is to disable the automatic loading of the built-in overlay, which you can do by putting `dtoverlay=` (i.e. with no overlay name) at the top of config.txt (or at least before any other `dt*` directive. In other words, replace the lines above with:
```
dtoverlay=
dtoverlay=hifiberry-dacplushd
```

pelwell, 2022-06-14:
Do you have anything to add, @j-schambacher?

j-schambacher, 2022-06-14:
Nothing to add, I fully agree with @pelwell's recommendation.

closed 2022-07-20 by pelwell