raspberrypi/linux #5064: 5.15.32 crashing sometimes

teletypo, 2022-06-20:
### Describe the bug

Hi!

I have built a yocto-image for my CM4, and I'm using linux 5.15.32 from this repository. The CM4's USB2 lines are connected to a FE1.1S 4port hub.

I am normally seeing happy operation, and can see 2 devices connected to the hub's downstream ports. But very rarely, the system boots up and there's a crash/warning/calltrace in dmesg, and then USB initialization seem to fail (`[   13.984092] usb 3-1: device descriptor read/64, error -110`), and neither of the 2 devices show up. I am puzzled how this can happen only sometimes, since the system boots from readonly-mounted storage, so every bootup should be the same.

The dmesg when it happens is attached. My questions are:

1. What is happening?
2. What does the call trace "say"?
3. Is this a known problem? Should I try to update to https://github.com/raspberrypi/linux/tree/rpi-5.19.y ?
4. What can I do to debug this better?

Thank you for your help!

### Steps to reproduce the behaviour

Boot the system many times until it happens.

### Device (s)

Raspberry Pi CM4 Lite

### System

Linux version 5.15.32-v8

### Logs

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP PREEMPT Tue Apr 5 18:36:06 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 29 pages/cpu s78232 r8192 d32360 u118784
[    0.000000] pcpu-alloc: s78232 r8192 d32360 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=... vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait root=/dev/mmcblk0p2 logo.nologo console=ttyAMA0,115200
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3814172K/4050944K available (12800K kernel code, 2002K rwdata, 4616K rodata, 3968K init, 975K bss, 171236K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40528 entries in 159 pages
[    0.000000] ftrace: allocated 159 pages with 6 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[... 514 lines cut ...]
[    9.797039] EXT4-fs (mmcblk0p7): resizing filesystem from 96196608 to 96196608 blocks
[   13.984092] usb 3-1: device descriptor read/64, error -110
[   29.600048] usb 3-1: device descriptor read/64, error -110
[   29.900038] usb 3-1: new high-speed USB device number 3 using dwc2
[   33.764069] cam-dummy-reg: disabling
[   35.232044] usb 3-1: device descriptor read/64, error -110
[   50.848045] usb 3-1: device descriptor read/64, error -110
[   50.960109] usb usb3-port1: attempt power cycle
[   52.031901] audit: type=1334 audit(1655481583.379:4): prog-id=7 op=LOAD
[   52.038701] audit: type=1334 audit(1655481583.383:5): prog-id=8 op=LOAD
[   52.365963] audit: type=1325 audit(1655481583.703:6): table=filter family=2 entries=0 op=xt_register pid=290 comm="iptables-restor"
[   52.381727] audit: type=1300 audit(1655481583.703:6): arch=c00000b7 syscall=209 success=yes exit=0 a0=5 a1=0 a2=40 a3=7fccc3fd80 items=0 ppid=1 pid=290 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables-restor" exe="/usr/sbin/xtables-legacy-multi" key=(null)
[   52.412319] audit: type=1327 audit(1655481583.703:6): proctitle=2F7573722F7362696E2F69707461626C65732D726573746F7265002D77002D2D002F6574632F69707461626C65732F69707461626C65732E72756C6573
[   52.429814] audit: type=1325 audit(1655481583.715:7): table=filter family=2 entries=4 op=xt_replace pid=290 comm="iptables-restor"
[   52.441779] audit: type=1300 audit(1655481583.715:7): arch=c00000b7 syscall=208 success=yes exit=0 a0=5 a1=0 a2=40 a3=55bd48eb10 items=0 ppid=1 pid=290 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables-restor" exe="/usr/sbin/xtables-legacy-multi" key=(null)
[   52.496192] audit: type=1327 audit(1655481583.715:7): proctitle=2F7573722F7362696E2F69707461626C65732D726573746F7265002D77002D2D002F6574632F69707461626C65732F69707461626C65732E72756C6573
[   52.524539] audit: type=1325 audit(1655481583.715:8): table=nat family=2 entries=0 op=xt_register pid=290 comm="iptables-restor"
[   52.536234] audit: type=1300 audit(1655481583.715:8): arch=c00000b7 syscall=209 success=yes exit=0 a0=5 a1=0 a2=40 a3=7fccc3fd80 items=0 ppid=1 pid=290 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables-restor" exe="/usr/sbin/xtables-legacy-multi" key=(null)
[   52.751877] exFAT-fs (mmcblk3p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   53.257115] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   53.336854] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   53.345492] bcmgenet fd580000.ethernet ethsensor1: Link is Down
[   54.321896] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   54.325076] lo ate my IP address
```

pelwell, 2022-06-20:
The problem reported by the kernel is that there is an interrupt that "nobody care[s]" about:
```
[    8.379418] irq 20: nobody cared (try booting with the "irqpoll" option)
...
[    8.417169]  __report_bad_irq+0x54/0xe4
[    8.421004]  note_interrupt+0x32c/0x3ec
[    8.424843]  handle_irq_event+0xdc/0x15c
[    8.428766]  handle_fasteoi_irq+0xcc/0x230
[    8.432863]  handle_domain_irq+0xa4/0xf0
[    8.436785]  gic_handle_irq+0x58/0xe0
[    8.440446]  do_interrupt_handler+0x3c/0x70
[    8.444631]  el1_interrupt+0x30/0x80
[    8.448211]  el1h_64_irq_handler+0x18/0x24
[    8.452306]  el1h_64_irq+0x78/0x7c
[    8.455706]  __do_softirq+0x118/0x504
[    8.459366]  __irq_exit_rcu+0xd0/0x114
[    8.463118]  irq_exit+0x1c/0x54
[    8.466261]  handle_domain_irq+0xa8/0xf0
[    8.470183]  gic_handle_irq+0x58/0xe0
[    8.473843]  call_on_irq_stack+0x2c/0x5c
[    8.477766]  do_interrupt_handler+0x5c/0x70
[    8.481951]  el1_interrupt+0x30/0x80
[    8.485530]  el1h_64_irq_handler+0x18/0x24
[    8.489625]  el1h_64_irq+0x78/0x7c
...
[    8.556773] handlers:
[    8.559040] [<000000007df74e85>] dwc2_handle_common_intr [dwc2]
[    8.565006] [<00000000282e2e1c>] usb_hcd_irq
[    8.569282] Disabling IRQ #20
```
I think this means that either there was no handler for the interrupt or that any handlers returned IRQ_NONE (not mine). Earlier in the log we can see that IRQ 20 is the dwc2 USB interrupt:
```
[    7.765195] dwc2 fe980000.usb: irq 20, io mem 0xfe980000
```
and the list of handlers includes "dwc2_handle_common_intr". Without reading the driver I would guess that is in an unexpected kind of interrupt, either because it is a feature that isn't being used or because the driver just isn't ready to receive one at that time (both of which sound likes bugs or major limitations).

pelwell, 2022-06-20:
i.e. `dtoverlay=dwc2,dr_mode=host`

teletypo, 2022-06-20:
Thank you for the quick help, everybody! Here's the config.txt:

```
# cat /uboot/config.txt 
[cm4]
dtoverlay=pcie-32bit-dma
dtoverlay=dwc2,dr_mode=host
dtoverlay=gpio-shutdown,gpio_pin=3
hdmi_force_hotplug=1
hdmi_mode=16
gpio=17=op,dh
dtparam=i2c_vc=on
dtoverlay=i2c-rtc,pcf85063a,i2c_csi_dsi
dtparam=ant2
dtoverlay=uart0
dtoverlay=uart1
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
dtoverlay=uart5
enable_uart=1
```
Bootloader version:
```
# vcgencmd bootloader_version
Feb 16 2021 13:23:36
version d6d82cf99bcb3e9a166a34cfde53130957a36bd3 (release)
timestamp 1613481816
update-time 1613481816
capabilities 0x0000001f
```

Here's the dmesg from a happy boot just 10 minutes after the problem showed, no changes were done to the system:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP PREEMPT Tue Apr 5 18:36:06 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 29 pages/cpu s78232 r8192 d32360 u118784
[    0.000000] pcpu-alloc: s78232 r8192 d32360 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=... vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait root=/dev/mmcblk0p2 logo.nologo console=ttyAMA0,115200
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3814172K/4050944K available (12800K kernel code, 2002K rwdata, 4616K rodata, 3968K init, 975K bss, 171236K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40528 entries in 159 pages
[    0.000000] ftrace: allocated 159 pages with 6 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[... 503 lines cut ...]
[    9.951676] usbserial: USB Serial support registered for siemens_mpi
[    9.958202] usb_serial_simple 3-1.1:1.0: novatel_gps converter detected
[    9.965274] usb 3-1.1: novatel_gps converter now attached to ttyUSB0
[    9.972296] usb 3-1.1: novatel_gps converter now attached to ttyUSB1
[    9.979092] usb 3-1.1: novatel_gps converter now attached to ttyUSB2
[   10.136827] audit: type=1334 audit(1655482234.971:4): prog-id=7 op=LOAD
[   10.143559] audit: type=1334 audit(1655482234.971:5): prog-id=8 op=LOAD
[   10.471255] audit: type=1325 audit(1655482235.287:6): table=filter family=2 entries=0 op=xt_register pid=319 comm="iptables-restor"
[   10.486593] audit: type=1300 audit(1655482235.287:6): arch=c00000b7 syscall=209 success=yes exit=0 a0=5 a1=0 a2=40 a3=7fd9ea16b0 items=0 ppid=1 pid=319 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables-restor" exe="/usr/sbin/xtables-legacy-multi" key=(null)
[   10.515347] audit: type=1327 audit(1655482235.287:6): proctitle=2F7573722F7362696E2F69707461626C65732D726573746F7265002D77002D2D002F6574632F69707461626C65732F69707461626C65732E72756C6573
[   10.533471] audit: type=1325 audit(1655482235.303:7): table=filter family=2 entries=4 op=xt_replace pid=319 comm="iptables-restor"
[   10.548369] audit: type=1300 audit(1655482235.303:7): arch=c00000b7 syscall=208 success=yes exit=0 a0=5 a1=0 a2=40 a3=558bef0b10 items=0 ppid=1 pid=319 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables-restor" exe="/usr/sbin/xtables-legacy-multi" key=(null)
[   10.591537] audit: type=1327 audit(1655482235.303:7): proctitle=2F7573722F7362696E2F69707461626C65732D726573746F7265002D77002D2D002F6574632F69707461626C65732F69707461626C65732E72756C6573
[... cut ...]
[   12.116586] lo ate my IP address
[   12.361439] lo ate my IP address
[   12.364028] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.616953] igb 0000:04:00.0 ethsensor0: igb: ethsensor0 NIC Link is Up 10 Mbps Half Duplex, Flow Control: None
[   13.631931] igb 0000:04:00.0: EEE Disabled: unsupported at half duplex. Re-enable using ethtool when at full duplex.
[   13.642873] IPv6: ADDRCONF(NETDEV_CHANGE): ethsensor0: link becomes ready
[   33.764005] cam-dummy-reg: disabling
```

lategoodbye, 2024-11-26:
I was able to reproduce a very similiar issue with Mainline on a Raspberry Pi 3A+ and a USB hub after suspend 2 idle.

The following branch contains the relevant fix ( usb: dwc2: Fix HCD resume ):
https://github.com/lategoodbye/linux-dev/commits/v6.12-dwc_hcd_fixes/

Maybe this also fixes the issue here.

lategoodbye, 2024-11-26:
Please see at https://github.com/raspberrypi/linux/issues/6247#issuecomment-2501797331 for a test build, which should fix the dwc2 IRQ issue.