raspberrypi/linux #5068: Corrupt f2fs root partition after update to 5.15.48

neo1973, 2022-06-24:
### Describe the bug

After updating from kernel 5.15.45 (linux-rpi-5.15.45-3-armv7h.pkg.tar.xz) to 5.15.48 (linux-rpi-5.15.48-1-armv7h.pkg.tar.xz) my f2fs formatted root partition gets corrupted. The problem seems to be related to compression because the update is fine with no compression enabled. If I then enable compression while on 5.15.48 the corruption starts again.

No other packages were updated during my testing. Unfortunately I can't narrow the version further down as .46 and .47 were not packaged for Arch Linux ARM.

Once the partition is corrupted the same errors appear on my desktop pc when reading the SD card. The test was also done on two different SD cards with the same outcome.

### Steps to reproduce the behaviour

Update to kernel 5.15.48 with `rootflags=compress_algorithm=lz4,compress_extension=*` on the kernel command line. First time when booting with .48 the errors start to appear in the log.

Old packages are available from http://tardis.tiny-vps.com/aarm/repos if someone wants to test.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

OS: Arch Linux ARM

Firmware:
```
Mar  1 2022 14:23:32
Copyright (c) 2012 Broadcom
version 969fb9b1521fc7ac2b88b15a3a9e942da7678c4d (clean) (release) (start)
```
Kernel: `Linux alarmpi 5.15.48-1-rpi-ARCH #1 SMP Sat Jun 18 03:30:55 MDT 2022 armv7l GNU/Linux`

Kernel command line: 
```
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 rootfstype=f2fs rootflags=compress_algorithm=lz4,compress_extension=*
```
Command to create the partition: `mkfs.f2fs -f -l rpi-root -O extra_attr,inode_checksum,sb_checksum,compression /dev/mmcblk0p2`

### Logs

dmesg is full of these errors and files start to become inaccessible:
```
[   10.395372] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=85f5, mode=33188) should not have inline_data, run fsck to fix
[   10.451170] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=8fa1, mode=33188) should not have inline_data, run fsck to fix
[   11.115445] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=8fa1, mode=33188) should not have inline_data, run fsck to fix
[   11.422398] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=8fa1, mode=33188) should not have inline_data, run fsck to fix
[   11.424113] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.437344] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.443613] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.452340] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.458474] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.480721] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.486459] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.494090] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.499908] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.506331] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.513823] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
[   11.520979] F2FS-fs (mmcblk0p2): sanity_check_inode: inode (ino=932a, mode=33188) should not have inline_data, run fsck to fix
```

neo1973, 2022-06-24:
It looks like there is is a fix upstream: https://lore.kernel.org/lkml/20220617223106.3517374-1-jaegeuk@kernel.org/t/

popcornmix, 2022-06-27:
I don't believe the fix made it to v5.15.50. When the fix reaches 5.15 I'll get it into rpi-update.

neo1973, 2022-07-10:
I tested Archs 5.15.52 kernel which does include the patch I believe and the problem is still there :(

neo1973, 2022-07-12:
Yes, it looks like somebody already did: https://bugzilla.kernel.org/show_bug.cgi?id=216162. I think this ticket can be closed and we need to wait for a fix to trickle down from upstream.

closed 2022-07-12 by neo1973