raspberrypi/linux #5072: qmi_wwan_simcom

xevilstar, 2022-06-30:
### Describe the bug

root@raspberrypi:/usr/src# tar xvpfz qmi_wwan_simcom.tar.gz 

qmi_wwan_simcom/qmi_wwan_simcom.c

qmi_wwan_simcom/Makefile

qmi_wwan_simcom/

root@raspberrypi:/usr/src# cd qmi_wwan_simcom

root@raspberrypi:/usr/src/qmi_wwan_simcom# make

rm -rf *.o *~ core .depend .*.cmd *.ko *.mod.c .cache.mk .tmp_versions Module.* modules.order

make -C /lib/modules/5.15.50-v8+/build M=/usr/src/qmi_wwan_simcom modules

make[1]: Entering directory '/usr/src/linux'

  CC [M]  /usr/src/qmi_wwan_simcom/qmi_wwan_simcom.o

/usr/src/qmi_wwan_simcom/qmi_wwan_simcom.c: In function ‘qmi_wwan_register_subdriver’:

/usr/src/qmi_wwan_simcom/qmi_wwan_simcom.c:227:13: error: incompatible type for argument 4 of ‘usb_cdc_wdm_register’

  227 |       4096, &qmi_wwan_cdc_wdm_manage_power);

      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

      |             |

      |             int (*)(struct usb_interface *, int)

In file included from /usr/src/qmi_wwan_simcom/qmi_wwan_simcom.c:21:

./include/linux/usb/cdc-wdm.h:20:39: note: expected ‘enum wwan_port_type’ but argument is of type ‘int (*)(struct usb_interface *, int)’

   20 |      int bufsize, enum wwan_port_type type,

      |                   ~~~~~~~~~~~~~~~~~~~~^~~~

/usr/src/qmi_wwan_simcom/qmi_wwan_simcom.c:226:14: error: too few arguments to function ‘usb_cdc_wdm_register’

  226 |  subdriver = usb_cdc_wdm_register(info->control, &dev->status->desc,

      |              ^~~~~~~~~~~~~~~~~~~~

In file included from /usr/src/qmi_wwan_simcom/qmi_wwan_simcom.c:21:

./include/linux/usb/cdc-wdm.h:18:27: note: declared here

   18 | extern struct usb_driver *usb_cdc_wdm_register(struct usb_interface *intf,

      |                           ^~~~~~~~~~~~~~~~~~~~

make[2]: *** [scripts/Makefile.build:288: /usr/src/qmi_wwan_simcom/qmi_wwan_simcom.o] Error 1

make[1]: *** [Makefile:1871: /usr/src/qmi_wwan_simcom] Error 2

make[1]: Leaving directory '/usr/src/linux'

make: *** [Makefile:7: all] Error 2

root@raspberrypi:/usr/src/qmi_wwan_simcom# uname -a

Linux raspberrypi 5.15.50-v8+ #1 SMP PREEMPT Wed Jun 29 16:20:10 CEST 2022 aarch64 GNU/Linux

### Steps to reproduce the behaviour

1. wget https://gitlab.freedesktop.org/mobile-broadband/ModemManager/uploads/1f0005f203000f38f6925d78c1839e27/qmi_wwan_simcom.tar.gz
2. tar xvpfz qmi_wwan_simcom.tar.gz
3. cd qmi_wwan_simcom
4. make

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4
vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
 uname -a
Linux raspberrypi 5.15.50-v8+ #1 SMP PREEMPT Wed Jun 29 16:20:10 CEST 2022 aarch64 GNU/Linux

### Additional context

I am trying to make this work 
https://www.amazon.it/gp/product/B09WRF4986/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1
on arm64

6by9, 2022-06-30:
Duplicated in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/455

6by9, 2022-06-30:
And appears to be duplicated in https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/589, which is the correct place to be asking for support on an out-of-tree driver.

xevilstar, 2022-06-30:
I have posted there not knowing that the two sites were connected sorry,
but if you could try to patch the source I would appreciate it

xevilstar, 2022-07-06:
works out of the box with kernel uname -a
Linux raspberrypi 5.19.0-rc4-v8 #1 SMP PREEMPT Thu Jun 30 16:52:23 CEST 2022 aarch64 GNU/Linux

xevilstar, 2022-10-11:
> tried that no answers
> It looks like it's the Waveshare product -
> See their wiki page for installation instructions, or ask them for
> https://www.waveshare.com/wiki/SIM8202G-M2_5G_HAT_(B)#Working_with_Raspberry
> Then again it looks like they've done their normal and are shipping
> binaries only for GPL code. 5.4.51 was a good while ago.

they say I must use the 32 bit version of the so which I cannot do for my
project

xevilstar, 2022-10-11:
tried that no answers