raspberrypi/linux #5076: Asmedia ASM235CM Issue

symbios24, 2022-07-02:
### Describe the bug

Hello

I have a USB 3.0 NATEC NKZ-0448 3.5" sata hard drive enclosure

With Asmedia ASM235CM Chipset

In a Raspberry Pi 4B 4GB rev1.4
With latest firmware/bootloader 2022-04-26

And upon using a mechanical 3.5" hard disk drive to boot from the perfomance is very poor/slow

I disabled UASP with quirks but that didn't help
I changed the case USB cable and the PSU that didn't help either.

Tried many different hard drives the results were always the same

Tried another case i have that is USB 2.0 and the hard drive there worked way faster/normal even if its USB 2.0

hdparm is reporting speeds of 80/100mb/sec but the Chipset/Case is like its delaying/lagging the hdd operations the perfomance feels like 5/10mb/sec
Its slower than the sdr25 card reader in previous raspberry pi models.

The same case in my windows computer works normal without delaying.

Is this a known incomparability with Asmedia ASM235CM Chipset ??

Thanks

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspbian Buster 

### Logs

lsub/lsusb -t / usb-devices info under:

raspberrypi:~ $ lsusb
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0c45:7403 Microdia Foot Switch
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb -t
/: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
|__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
|__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
|__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
|__ Port 4: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
pi@raspberrypi:~ $ usb-devices

T: Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#= 1 Spd=480 MxCh= 1
D: Ver= 2.00 Cls=09(hub ) Sub=00 Prot=01 MxPS=64 #Cfgs= 1
P: Vendor=1d6b ProdID=0002 Rev=05.10
S: Manufacturer=Linux 5.10.43v64 xhci-hcd
S: Product=xHCI Host Controller
S: SerialNumber=0000:01:00.0
C: #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I: If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub ) Sub=00 Prot=00 Driver=hub

T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 2 Spd=480 MxCh= 4
D: Ver= 2.10 Cls=09(hub ) Sub=00 Prot=01 MxPS=64 #Cfgs= 1
P: Vendor=2109 ProdID=3431 Rev=04.21
S: Product=USB2.0 Hub
C: #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I: If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub ) Sub=00 Prot=00 Driver=hub

T: Bus=01 Lev=02 Prnt=02 Port=03 Cnt=01 Dev#= 3 Spd=1.5 MxCh= 0
D: Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs= 1
P: Vendor=0c45 ProdID=7403 Rev=00.01
S: Manufacturer=SONiX
S: Product=USB Device
C: #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=100mA
I: If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID ) Sub=01 Prot=01 Driver=usbhid
I: If#=0x1 Alt= 0 #EPs= 1 Cls=03(HID ) Sub=01 Prot=02 Driver=usbhid

T: Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#= 1 Spd=5000 MxCh= 4
D: Ver= 3.00 Cls=09(hub ) Sub=00 Prot=03 MxPS= 9 #Cfgs= 1
P: Vendor=1d6b ProdID=0003 Rev=05.10
S: Manufacturer=Linux 5.10.43v64 xhci-hcd
S: Product=xHCI Host Controller
S: SerialNumber=0000:01:00.0
C: #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I: If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub ) Sub=00 Prot=00 Driver=hub

T: Bus=02 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#= 2 Spd=5000 MxCh= 0
D: Ver= 3.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs= 1
P: Vendor=174c ProdID=55aa Rev=01.00
S: Manufacturer=ASMedia
S: Product=USB 3.1 Storage Device
S: SerialNumber=0000000000C7
C: #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=0mA
I: If#=0x0 Alt= 1 #EPs= 4 Cls=08(stor.) Sub=06 Prot=62 Driver=uas
pi@raspberrypi:~ $

tazz4843, 2022-12-17:
I can reproduce this issue to an extent. I have the same SATA adapter chipset, AsMedia ASM235CM, connected to a Samsung 860 EVO SSD (250gb). Normal performance is fine, but read ops are delayed by a few seconds. This issue is only noticeable on the Raspberry Pi 4 it's hooked up to, my x86 laptop is perfectly fine. At some point inevitably (usually takes a few days) a bunch of read ops will get built up and then it snowballs into a complete system lockup due to extreme IOWait. There is almost definitely some sort of underlying bug here in the Pi kernel, given the Pi is the only device to suffer from this issue.

phbcanada, 2023-02-10:
I am also having problems with this device (same chipset): https://www.amazon.ca/dp/B0B6PMVZHG
Works for a while but when a lot of data is being written I get write errors and the file system remounts as read-only.
Tried looking for firmware updates but the update tools I tried did not work with this device.

symbios24, 2023-02-10:
Try that do an rpi-update i think they fix some issues with some sata controllers

And if its the same after the update use quirks