raspberrypi/linux #5077: Kernel 5.18: Faulty playback of HEVC videos (KODI)

mkreisl, 2022-07-02:
### Describe the bug

I already noticed it with 5.18.2, and today I had another chance with 5.18.7 to play 1080p HEVC videos with Kodi without errors.
Unfortunately with the same result: the playback is simply broken. AVC videos, on the other hand, are played back flawlessly.

The same kernel settings were used that worked perfectly up to and including 5.17. So my question is, are any magical new settings required or is this simply a bug?

### Steps to reproduce the behaviour

Play HEVC video correctly

### Device (s)

Raspberry Pi 4 Mod. B

### System

XBian (based on Debian Bullseye)
Kodi 19.4
vcgencmd version:
Jun 15 2022 20:02:09 
Copyright (c) 2012 Broadcom
version ede5f7d365ef42ebbd4aa144ee5b51ea75604c82 (clean) (release) (start_x)

popcornmix, 2022-07-05:
The state structure has changed (used for communicating between kernel and ffmpeg).
You need to rebuild ffmpeg with updated headers.
Check updated branches on @jc-kynesim 's ffmpeg repo. e.g. https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.3.4/hevc_api_4

mkreisl, 2022-07-05:
@popcornmix, FYI:

I have now done this with Kodi 20 Alpha (dev/4.4/rpi_import_1 branch) and with kernel 5.18.7 the playback of HEVC works again without problems. Fortunately, it also still works with 5.15.45 kernel.

For Kodi 19 I still have to create the FFmpeg patch to test it further.

jc-kynesim, 2022-07-05:
FFmpeg does try to check that the structures match its expectations and it should cope correctly with all "released" versions of the linux driver but it can be fooled (rubbish) by some intermediate versions.  If it knows that it can't work then it won't start (black screen).

mkreisl, 2022-07-06:
Kodi 19/20 with kernel 5.18 and 5.19 is working again. Time to close this issue here.