raspberrypi/linux #5078: Rpi H265 driver generates warning when build with clang14

zhanghua000, 2022-07-03:
### Describe the bug

When build kernel with clang14, there will be a warning says `unannotated fall-through between switch labels` when build Rpi H265 driver, it will become an error if enabled `Compile the kernel with warnings as errors` option.

### Steps to reproduce the behaviour

1. Prepare a host and the source code, branch is `rpi.5.19.y`. Note: Using `clang14` for building.
2. Build the kernel source code.

### Device (s)

Raspberry Pi 4 Mod. B

### System

N/A

### Logs

Clang Output:
```
  CC [M]  drivers/staging/media/rpivid/rpivid_h265.o
drivers/staging/media/rpivid/rpivid_h265.c:1714:3: error: unannotated fall-through between switch labels [-Werror,-Wimplicit-fallthrough]
                case RPIVID_DECODE_ERROR_CONTINUE:
                ^
drivers/staging/media/rpivid/rpivid_h265.c:1714:3: note: insert '__attribute__((fallthrough));' to silence this warning
                case RPIVID_DECODE_ERROR_CONTINUE:
                ^
                __attribute__((fallthrough)); 
drivers/staging/media/rpivid/rpivid_h265.c:1714:3: note: insert 'break;' to avoid fall-through
                case RPIVID_DECODE_ERROR_CONTINUE:
                ^
                break; 
1 error generated.
```
I have enabled `Compile the kernel with warnings as errors` option so this warning is treated as an error.

### Additional context

N/A

popcornmix, 2022-07-04:
I believe this should be fixed with https://github.com/raspberrypi/linux/pull/5081

zhanghua000, 2022-07-05:
Yes, the latest code has fixed this, but it seems that this problem appears at `VideoCore Shared Memory (CMA) Driver`.
```
  CC [M]  drivers/staging/vc04_services/vc-sm-cma/vc_sm_cma_vchi.o
drivers/staging/vc04_services/vc-sm-cma/vc_sm_cma_vchi.c:281:2: error: unannotated fall-through between switch labels [-Werror,-Wimplicit-fallthrough]
        default:
        ^
drivers/staging/vc04_services/vc-sm-cma/vc_sm_cma_vchi.c:281:2: note: insert 'break;' to avoid fall-through
        default:
        ^
        break; 
1 error generated.
```
Should I create a new issue or continue on this?

pelwell, 2022-07-05:
Thanks for the heads-up - it's now fixed: https://github.com/raspberrypi/linux/commit/df1e772886002b0ccf273102cc5fc0f7edfb459a

zhanghua000, 2022-07-05:
Thanks for your help, this problem now disappears.

closed 2022-07-05 by zhanghua000