raspberrypi/linux #5085: fuzzy black text on the red background in a 15.6 touchscreen on raspberry kernel 5.10,but kernel 4.19 works well

gitgonewithwind, 2022-07-06:
### Describe the bug

HI all,
our application need to show black text on the red background as error,

but fuzzy black text on the red background in a 15.6 touchscreen (Lilliput. The model is tk156a ) on raspberry kernel 5.10 ( ycoto meta-raspberrypi https://github.com/agherzan/meta-raspberrypi/commit/e1014d41ef459af807f95cb7ff9105f077bc6daa), the pi 4 hardware versions are 1.4 and 1.2

at same time, I tested updated raspbian image 2022-04-04-raspios-bullseye-arm64.img.xz(not yocto image) from here: https://www.raspberrypi.com/software/ (kernel 5.15) it has the same issue on both pi4 hardware 1.4 and 1.2

we have pi 4 hardware version 1.2 ,so I tested older yocto image (kernel 4.19 , meta-raspberrypi https://github.com/agherzan/meta-raspberrypi/commit/b112816e95ec59447a82a3a0ee7842c547dda335) ,it works well!( of course ,this image can not startup in hardware 1.4,so we can not use 1.2 boards for our application )

therefore, I think kernel 4.19 work well for the touchscreen ,but updated kernel (5.10,5.15) not work well. Maybe the issue is from HDMI drivers in higher kernel?
so do you know where are the HDMI driver files in kernel? how to modified thems (I mean where is the key code ？）
any help?
I change different resolutions ,but same error.

### Steps to reproduce the behaviour

see the pictures here for more info
https://github.com/agherzan/meta-raspberrypi/issues/1069

### Device (s)

Raspberry Pi 4 Mod. B

### System

Which kernel version:
kernel  5.10
&
kernel 5.15

popcornmix, 2022-07-07:
It sounds like issue occurs with 2022-04-04-raspios-bullseye-arm64 (with 5.15 kernel), so focus on that.
Post any changes (if any) you made to config.txt and cmdline.txt.

By default that image uses the kms driver (`dtoverlay=vc4-kms-v3d`). Can you confirm if the issue occurs with kms changed to fkms on that line?

With kms, when you see fuzzy text can you post output of `/sys/kernel/debug/dri/0/state`.
If that file doesn't exist then try `/sys/kernel/debug/dri/1/state`.

gitgonewithwind, 2022-07-08:
> [torvalds/linux@f741b28](https://github.com/torvalds/linux/commit/f741b28fb299263d2d03a0fb701bfe648927cd47) introduced in 5.2 dropped the framebuffer default from 32bpp to 16bpp assuming you're running vc4-kms-v3d or vc4-fkms-v3d.
> 
> If your text renderer is using anti-aliasing to try and smooth the font out, then the loss of colour definition may cause that sort of issue.

@6by9 
my yocto image kernel is 5.10.  drivers/gpu/drm/vc4/vc4_drv.c is already changed to drm_fbdev_generic_setup(drm, 16); 
so the issue is NOT here.

6by9, 2022-07-08:
16bpp will be blurrier than 32bpp if using anti-aliased fonts.

gitgonewithwind, 2022-07-08:
> /sys/kernel/debug/dri/1/state

my debug info for Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux  2022-04-04-raspios-bullseye-arm64 ( I am going to try @6by9 solutions. let will post the result)
dtoverlay=vc4-kms-v3d &  dtoverlay=vc4-fkms-v3d

1.   vc4-fkms-v3d
cat /sys/kernel/debug/dri/1/state
plane[31]: plane-0
        crtc=crtc-0
        fb=99
                allocated by = Xorg
                refcount=2
                format=XR24 little-endian (0x34325258)
                modifier=0x0
                size=1024x768
                layers:
                        size[0]=1024x768
                        pitch[0]=4096
                        offset[0]=0
                        obj[0]:
                                name=0
                                refcount=3
                                start=001006f5
                                size=3145728
                                imported=no
        crtc-pos=1024x768+0+0
        src-pos=1024.000000x768.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.709 YCbCr
        color-range=YCbCr limited range
plane[38]: plane-1
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=1
        color-encoding=ITU-R BT.709 YCbCr
        color-range=YCbCr limited range
plane[45]: plane-2
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=1
        color-encoding=ITU-R BT.709 YCbCr
        color-range=YCbCr limited range
plane[52]: plane-3
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=1
        color-encoding=ITU-R BT.709 YCbCr
        color-range=YCbCr limited range
plane[59]: plane-4
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=1
        color-encoding=ITU-R BT.709 YCbCr
        color-range=YCbCr limited range
plane[66]: plane-5
        crtc=(null)
        fb=0
[... 373 lines cut ...]
        connector_mask=2
        encoder_mask=2
        mode: "": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x0 0x5
crtc[116]: crtc-5
        enable=0
        active=0
        self_refresh_active=0
        planes_changed=0
        mode_changed=0
        active_changed=0
        connectors_changed=0
        color_mgmt_changed=0
        plane_mask=0
        connector_mask=0
        encoder_mask=0
        mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
connector[32]: HDMI-A-1
        crtc=(null)
        self_refresh_aware=0
connector[41]: HDMI-A-2
        crtc=crtc-4
        self_refresh_aware=0
connector[50]: Writeback-1
        crtc=(null)
        self_refresh_aware=0

gitgonewithwind, 2022-07-08:
@6by9 
my cmdline.txt:

video=HDMI-A-1:1920x1080M-32@60 console=serial0,115200 console=tty1 root=PARTUUID=f3d36243-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

config.txt :
dtoverlay=vc4-kms-v3d    or dtoverlay=vc4-fkms-v3d 

---
but  the  issue is same as before

popcornmix, 2022-07-08:
The state posted for kms looks to have:
```
mode: "": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x0 0x5
format=XR24 little-endian (0x34325258)
crtc-pos=1920x1080+0+0
src-pos=1920.000000x1080.000000+0.000000+0.000000
```
So, a 1920x1080 hdmi resolution. 24bpp framebuffer. And no scaling. So that all looks as expected.

popcornmix, 2022-07-08:
Although fkms:
```
mode: "": 75 78750 1024 1040 1136 1312 768 769 772 800 0x0 0x5
```
kms:
```
mode: "": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x0 0x5
```
are picking different hdmi resolutions. Do both appear fuzzy or only one?

gitgonewithwind, 2022-07-08:
@popcornmix 
I tried different hdmi resolutions.  both appear fuzzy.

6by9, 2022-07-08:
I'd asked in [your forum post](https://forums.raspberrypi.com/viewtopic.php?p=2018438) for a link to this display as a search only revealed a TK1560-T - you never answered.

What is the native resolution of this display?

How are you generating your text? It vaguely looks like it's running in a web browser.
If you go back to your 4.19 image, then what does ```tvservice -s``` report?

Both vc4-kms-v3d and vc4-fkms-v3d will be rendering the display through V3D and OpenGL. 4.19 is likely to be working directly with the framebuffer with no acceleration. 

Does it look blurry on a different HDMI display?

gitgonewithwind, 2022-07-08:
@6by9 

I'd asked in [your forum post](https://forums.raspberrypi.com/viewtopic.php?p=2018438) for a link to this display as a search only revealed a TK1560-T - you never answered.
--
I did not received the email (the system shows will send a email to me if posted ), sorry I did not reply.

What is the native resolution of this display?
native resolution:1920x1080
this is the webpage for the screen :https://lilliputweb.net/tk1560-t-15-6-inch-touch-screen-monitor/ 

How are you generating your text? It vaguely looks like it's running in a web browser.
If you go back to your 4.19 image, then what does tvservice -s report?
yes,   it's running in a web browser for our application.
what does tvservice -s report? I will get it later

Both vc4-kms-v3d and vc4-fkms-v3d will be rendering the display through V3D and OpenGL. 4.19 is likely to be working directly with the framebuffer with no acceleration.
4.19 works well for my yocto image.

Does it look blurry on a different HDMI display?

not. we use more that 5 kinds HDMI display. there is no this kind issue.
 but one has another issue here:  https://github.com/agherzan/meta-raspberrypi/issues/1027 for  13.3" LCD Lilliput TK1330-NP/C/T  that we solve with hardware

gitgonewithwind, 2022-07-11:
@6by9 
If you go back to your 4.19 image, then what does tvservice -s report? 
--
in my yocto image:

tvservice -s
state 0x6 [DVI CUSTOM RGB full 16:9], 1280x720 @ 60.00Hz, progressive
root@raspberrypi4-64:~# uname -a
Linux raspberrypi4-64 4.19.66 #1 SMP PREEMPT Wed Apr 13 18:11:21 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

6by9, 2022-07-11:
Well that appears to answer your own question - run at 1280x720 60Hz.
Either modify the ```video=HDMI-A-1...``` line if booting to the console, or use ```xrandr --output HDMI-1 --mode 1280x720 --refresh 60``` if using X.

gitgonewithwind, 2022-07-11:
> Well that appears to answer your own question - run at 1280x720 60Hz. Either modify the `video=HDMI-A-1...` line if booting to the console, or use `xrandr --output HDMI-1 --mode 1280x720 --refresh 60` if using X.

you mean I run  xrandr --output HDMI-1 --mode 1280x720 --refresh 60   in kernel 5.10 at hardware version 1.4 , it will be fix ?

popcornmix, 2022-07-11:
> you mean I run xrandr --output HDMI-1 --mode 1280x720 --refresh 60 in kernel 5.10 at hardware version 1.4 , it will be fix ?

That's the suggestion. Try it. You can also use preferences/screen configuration from desktop menu for a gui way of setting this.

gitgonewithwind, 2022-07-11:
@6by9 
I tired before ,it's better but 
fuzzy  still  has a little bit.
my colleagues think it seems OK.( you know the fuzzy is not very heavy, Maybe my colleagues or QC can accept it )

/etc/X11/xorg.conf.d/10-monitor.conf 
..
Identifier "Screen0"
Device "HDMI-1"
Monitor "HDMI-1"
DefaultDepth 24
SubSection "Display"
Depth 24
Modes "1920x1080" "1280x720_60.00"
EndSubSection
EndSection
..

gitgonewithwind, 2022-07-11:
> The other thing from your tvservice output is that it is sending full range RGB in DVI mode, when I suspect that 5.10+ will be using limited range HDMI mode (depends on the EDID). You can force full range via `xrandr --output HDMI-1 --set "Broadcast RGB" "Full"`. (Options are `Automatic`, `Full`, or `Limited 16:235`).
> 
> This all sounds like it is down to a dodgy EDID. `xrandr --verbose` will print it out as a block of 16 lines (along with the modes that it has parse from it). Can you post that section please?
which kernel I run xrandr --verbose ? kernel 5.10?

gitgonewithwind, 2022-07-11:
> Do you have overscan on? I believe that can change the scaling away from the native resolution.

I think I do not have overscan on(I do not know where to switch it on). we only open  chromium(no frame,no address bar) pages to as a menu as GUI to display infos

popcornmix, 2022-07-11:
> Do you have overscan on? I believe that can change the scaling away from the native resolution.

There was no scaling (see https://github.com/raspberrypi/linux/issues/5085#issuecomment-1179040023)

6by9, 2022-07-11:
> which kernel I run xrandr --verbose ? kernel 5.10?

Yes, 5.10 or 5.15.

gitgonewithwind, 2022-07-11:
@6by9 
my yocto image:

5.10.83-v8 #1 SMP PREEMPT Tue Dec 7 13:42:57 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
root@raspberrypi4-64:~# xrandr --verbose
Can't open display 
root@raspberrypi4-64:~# export DISPLAY=:0
root@raspberrypi4-64:~# xrandr --verbose
Screen 0: minimum 320 x 200, current 1280 x 720, maximum 7680 x 7680
HDMI-1 connected primary 1280x720+0+0 (0x47) normal (normal left inverted right x axis y axis) 170mm x 110mm
        Identifier: 0x44
        Timestamp:  12680
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
        Brightness: 1.0
        Clones:    
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        EDID: 
                00ffffffffffff003190502102000000
                2d1c010380301b788a2895a7554ea326
                0f5054bfef8081c090407140b3008180
                a9c081008140023a801871382d40582c
                4500aa6e0000001e662156aa51001e30
                46f33300aa6e0000001e000000fc004d
                6f6e69746f720a2020202020000000fd
                00184c1e511e000000000000000001b8
                020330c24d010304050790121314169f
                2022260907070700008301000072030c
                001000b84420c0840102030401410000
                8c0ad08a20e02d10103e9600c48e2100
                00188c0ad090204031200c405500c48e
                21000018011d00bc52d01e20b8285540
                c48e2100001e011d80d0721c1620102c
                2580c48e2100009e0000000000000010
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 32 
                supported: 32
        non-desktop: 0 
                range: (0, 1)
  1280x720_60.00 (0x47) 74.500MHz -HSync +VSync *current +preferred
        h: width  1280 start 1344 end 1472 total 1664 skew    0 clock  44.77KHz
        v: height  720 start  723 end  728 total  748           clock  59.86Hz
  1920x1080 (0x48) 148.500MHz +HSync +VSync +preferred
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock  67.50KHz
        v: height 1080 start 1084 end 1089 total 1125           clock  60.00Hz
  4096x2160 (0x49) 297.000MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  54.00KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  24.00Hz
  4096x2160 (0x4a) 296.703MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  53.95KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  23.98Hz
  3840x2160 (0x4b) 297.000MHz +HSync +VSync
[... 137 lines cut ...]
        Subpixel:   unknown
        Clones:    
        CRTCs:      4
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 40 
                supported: 40
        non-desktop: 0 
                range: (0, 1)

gitgonewithwind, 2022-07-11:
@6by9 
root@raspberrypi4-64:~# xrandr --output HDMI-1 --set "Broadcast RGB" "Full"
X Error of failed request:  BadName (named color or font does not exist)
  Major opcode of failed request:  139 (RANDR)
  Minor opcode of failed request:  11 (RRQueryOutputProperty)
  Serial number of failed request:  39
  Current serial number in output stream:  39

6by9, 2022-07-11:
"Broadcast RGB" was only added in 5.15 via #4850 and updated by #5021 in May. You'll need a fairly recent kernel for that (5.15.40 or more recent).

gitgonewithwind, 2022-07-11:
> "Broadcast RGB" was only added in 5.15 via #4850 and updated by #5021 in May. You'll need a fairly recent kernel for that (5.15.40 or more recent).

do you know it has used in https://github.com/agherzan/meta-raspberrypi  for yocto ? if have done, I  will clone it to test it

gitgonewithwind, 2022-07-11:
@6by9 
I changed to 1920x1080, I used same test html file, so the display is smaller than 1280x720, the fuzzy is still there.(it's smaller, so Maybe my colleagues think it seems OK)
I am going to clone the newest 5.15 to try. Thanks.
root@raspberrypi4-64:~#  xrandr --verbose
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (0x47) normal (normal left inverted right x axis y axis) 170mm x 110mm
        Identifier: 0x44
        Timestamp:  12849
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
        Brightness: 1.0
        Clones:    
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        EDID: 
                00ffffffffffff003190502102000000
                2d1c010380301b788a2895a7554ea326
                0f5054bfef8081c090407140b3008180
                a9c081008140023a801871382d40582c
                4500aa6e0000001e662156aa51001e30
                46f33300aa6e0000001e000000fc004d
                6f6e69746f720a2020202020000000fd
                00184c1e511e000000000000000001b8
                020330c24d010304050790121314169f
                2022260907070700008301000072030c
                001000b84420c0840102030401410000
                8c0ad08a20e02d10103e9600c48e2100
                00188c0ad090204031200c405500c48e
                21000018011d00bc52d01e20b8285540
                c48e2100001e011d80d0721c1620102c
                2580c48e2100009e0000000000000010
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 32 
                supported: 32
        non-desktop: 0 
                range: (0, 1)
  1920x1080 (0x47) 148.500MHz +HSync +VSync *current +preferred
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock  67.50KHz
        v: height 1080 start 1084 end 1089 total 1125           clock  60.00Hz
  4096x2160 (0x48) 297.000MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  54.00KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  24.00Hz
  4096x2160 (0x49) 296.703MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  53.95KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  23.98Hz
  3840x2160 (0x4a) 297.000MHz +HSync +VSync
        h: width  3840 start 4016 end 4104 total 4400 skew    0 clock  67.50KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  30.00Hz
  3840x2160 (0x4b) 297.000MHz +HSync +VSync
        h: width  3840 start 4896 end 4984 total 5280 skew    0 clock  56.25KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  25.00Hz
  3840x2160 (0x4c) 297.000MHz +HSync +VSync
        h: width  3840 start 5116 end 5204 total 5500 skew    0 clock  54.00KHz
[... 133 lines cut ...]
        Subpixel:   unknown
        Clones:    
        CRTCs:      4
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 40 
                supported: 40
        non-desktop: 0 
                range: (0, 1)

gitgonewithwind, 2022-07-11:
@6by9 
yes, I find   I can use https://github.com/agherzan/meta-raspberrypi for 5.15
I will test it.
Thank you very much.

@[kraj](https://github.com/agherzan/meta-raspberrypi/commits?author=kraj)
@[agherzan](https://github.com/agherzan/meta-raspberrypi/commits?author=agherzan)
kraj and agherzan [linux-raspberrypi: Upgrade to 5.15.38](https://github.com/agherzan/meta-raspberrypi/commit/62a84833d90dfc18f8057bc5e99e96eff6798f12)
…

gitgonewithwind, 2022-07-12:
@6by9 
I am using 5.15 to test. if I display my test.htm file in chromium on the touchscreen  directly as before. there is a little fuzzy as before. but I enlarge it with fingers on the touchscreen. it's clear!
so is it scaling issue ?
.
.
.

uname -a
Linux raspberrypi4-64 5.15.38-v8 #1 SMP PREEMPT Wed May 11 19:05:52 UTC 2022 aarch64 GNU/Linux
"oot@raspberrypi4-64:~# xrandr --output HDMI-1 --set "Broadcast RGB" "Full"
root@raspberrypi4-64:~# xrandr --verbose
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (0x47) normal (normal left inverted right x axis y axis) 170mm x 110mm
        Identifier: 0x44
        Timestamp:  440582
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
        Brightness: 1.0
        Clones:    
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        EDID: 
                00ffffffffffff003190502102000000
                2d1c010380301b788a2895a7554ea326
                0f5054bfef8081c090407140b3008180
                a9c081008140023a801871382d40582c
                4500aa6e0000001e662156aa51001e30
                46f33300aa6e0000001e000000fc004d
                6f6e69746f720a2020202020000000fd
                00184c1e511e000000000000000001b8
                020330c24d010304050790121314169f
                2022260907070700008301000072030c
                001000b84420c0840102030401410000
                8c0ad08a20e02d10103e9600c48e2100
                00188c0ad090204031200c405500c48e
                21000018011d00bc52d01e20b8285540
                c48e2100001e011d80d0721c1620102c
                2580c48e2100009e0000000000000010
        Broadcast RGB: Full 
                supported: Automatic, Full, Limited 16:235
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGBr
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 32 
                supported: 32
        non-desktop: 0 
                range: (0, 1)
  1920x1080 (0x47) 148.500MHz +HSync +VSync *current +preferred
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock  67.50KHz
        v: height 1080 start 1084 end 1089 total 1125           clock  60.00Hz
  4096x2160 (0x48) 297.000MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  54.00KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  24.00Hz
  4096x2160 (0x49) 296.703MHz +HSync +VSync
        h: width  4096 start 5116 end 5204 total 5500 skew    0 clock  53.95KHz
[... 371 lines cut ...]
        CRTCs:      4
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter: 
        Broadcast RGB: Automatic 
                supported: Automatic, Full, Limited 16:235
        max bpc: 8 
                range: (8, 12)
        bottom margin: 0 
                range: (0, 100)
        top margin: 0 
                range: (0, 100)
        right margin: 0 
                range: (0, 100)
        left margin: 0 
                range: (0, 100)
        Colorspace: Default 
                supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGBr
        link-status: Good 
                supported: Good, Bad
        CONNECTOR_ID: 41 
                supported: 41
        non-desktop: 0 
                range: (0, 1)

gitgonewithwind, 2022-07-12:
Hi @6by9  I tried kernel  5.10 and  I enlarge the webpage with fingers on the touchscreen. it's clear.
but for our app ,we can not show so huge size infos . so we can not void the issue with it