raspberrypi/linux #5089: CPU governor defaults to powersave?

solskogen, 2022-07-09:
### Describe the bug

I'm not sure if this is a bug or a feature, but it comes to me as a bug as the CPU governor defaults to powersave, which means that it runs at 600MHz all the time unless I tell it not to. 

```
 $ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
powersave
```
I'd rather expected it to be ondemand. 

### Steps to reproduce the behaviour

Install latest 64bit distro, and upgrade it.

### Device (s)

Raspberry Pi 400

### System

```
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage5

$ vcgencmd version
Mar 24 2022 13:19:26
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

```

popcornmix, 2022-07-09:
The init script `etc/init.d/raspi-config` will switch the governor to performance (assuming you are not pressing the shift key to disable it).
Are you sure this is failing on a clean (and updated) install. I'd expect to hear a lot more complaints if it were.

solskogen, 2022-07-09:
My fault, sorry! For some strange reason that script was disabled on my system. Enabling it, did the trick!

closed 2022-07-09 by solskogen