raspberrypi/linux #5096: SPI Transfer time out with RT Patch when using MCP2518fd together with other SPI Devices via spidev

framet, 2022-07-18:
### Describe the bug

When using an MCP2518FD on SPI0 and using another SPI-Device on either SPI3 oder SPI4 and accessing the SPI-Device with spidev and iotctl an "spidev spix.x: SPI transfer timed out" occures which results in an IRQ spin/deadlock. The first timeout in visible in dmesg can be either SPI0 or SPI3/4.

See attached Log for dmesg out put.

Picture of htop when irq spin happens after error.
[image]

This Problem does not occure when:
 - can0 is down (ifdown can0) or no transfer are happening on the other SPIDev on eiter spi3 or spi4
 - no realtime patch is applied
 - using linux kernel 4.19.y-rt
 - using can0 (MCP2518FD on SPI0) and using an SPI-Device on spi1 (mini SPI Controller) but not spi3 or spi4

What I've already tried and not fixed the problem
 - cpu isolution for spidev calls, smp_affinity for spi irq's
 - spi->rt = false when the mcp251xfd driver calls spi_setup
 - spi->rt = true when spidev calls spi_setup
 - changed the spi_async call in spidev.c to spi_async_locked

I have testet everything on the following versions:
5.10.27:
https://github.com/raspberrypi/linux/commits/b460129fcbebe39b8ce1a3dfc18129998d12c40b
http://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.10/patch-5.10.27-rt36.patch.xz
5.15.49:
https://github.com/raspberrypi/linux/commits/ee039006371a0b1d64d825a59f0eed8627bb3c91
https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.15/patch-5.15.49-rt47.patch.xz
5.15.40-rt: 
https://github.com/kdoren/linux/tree/rpi-5.15.40-rt
4.19.y-rt: 
https://github.com/raspberrypi/linux/tree/rpi-4.19.y-rt

For the Kernel 4.19.y-rt I've used the mcp25xxfd driver from https://github.com/TKI-Automotive-GmbH/linux-rpi 

Everything was built with buildroot.

Hardware is a CM4 Module 1GB Ram without Wifi Module. It was tested on our own Motherboard and on the default Raspberry Pi IO Board.

### Steps to reproduce the behaviour

We are using an 20Mhz Oscillator instead of the 40Mhz with the MCP2518fd
can0 setup:
ip link set can0 up type can bitrate 1000000 sample-point 0.8 dbitrate 4000000 dsample-point 0.8 fd on restart-ms 100
no tx traffic
rx traffic: one message with 400hz and 16 Byte payload. Payload is sent with BRS on.

#enable spi0 and mcphttps://github.com/TKI-Automotive-GmbH/linux-rpi 2518fd
dtoverlay=spi0-1cs
dtoverlay=mcp251xfd,spi0-0,oscillator=20000000,speed=8500000,interrupt=24

Use any SPI-Device on either spi3 or spi4 and transfer Data with 400Hz with spidev ioctls.

### Device (s)

Raspberry Pi CM4

### System

Jan 17 2022 19:20:34 
Copyright (c) 2012 Broadcom
version bd34f55ef7b01b0a367f131060b561a2a58b80bb (clean) (release) (start)

Linux ss-sent-0 5.15.40-rt43-v7l #7 SMP PREEMPT_RT Fri Jul 15 18:37:21 CEST 2022 armv7l GNU/Linux

### Logs

[   50.994850] spidev spi4.1: SPI transfer timed out
[   50.994864] 8<--- cut here ---
[   50.994866] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   50.994871] pgd = 29a07681
[   50.994876] [00000000] *pgd=80000000004003, *pmd=00000000
[   50.994888] Internal error: Oops: 207 [#1] PREEMPT_RT SMP ARM
[   50.994893] Modules linked in: can_raw can cfg80211 rfkill ipv6 imx219 usb_f_ecm g_ether usb_f_rndis u_ether libcomposite i2c_dev raspberrypi_hwmon bcm2835_isp(C) bcm2835_codec(C) v4l2_mem2mem mcp251xfd can_dev bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_vmalloc spidev dwc2 roles spi_bcm2835 spi_bcm2835aux pwm_bcm2835 nvmem_rmem i2c_bcm2835 i2c_mux_pinctrl i2c_mux rpivid_mem bcm2835_unicam videobuf2_dma_contig v4l2_dv_timings v4l2_fwnode v4l2_async videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc uio_pdrv_genirq uio [last unloaded: imx219]
[   50.994972] CPU: 2 PID: 170 Comm: spi4 Tainted: G         C        5.15.40-rt43-v7l #1
[   50.994977] Hardware name: BCM2711
[   50.994980] PC is at bcm2835_spi_handle_err+0x20/0xe0 [spi_bcm2835]
[   50.994993] LR is at spi_transfer_one_message+0x454/0x6ec
[   50.995003] pc : [<bf0e754c>]    lr : [<c08fd0b0>]    psr: a0000013
[   50.995007] sp : c2c67e70  ip : c2c67e90  fp : c2c67e8c
[   50.995010] r10: c15495e0  r9 : c2d39000  r8 : 003da540
[   50.995012] r7 : c3df7eac  r6 : 00000000  r5 : c181e000  r4 : c181e3c0
[   50.995015] r3 : bf0e752c  r2 : 00000000  r1 : c3df7eac  r0 : c181e000
[   50.995018] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   50.995023] Control: 30c5383d  Table: 02be0480  DAC: 55555555
[   50.995025] Register r0 information: slab kmalloc-2k start c181e000 pointer offset 0 size 2048
[   50.995038] Register r1 information: non-slab/vmalloc memory
[   50.995044] Register r2 information: NULL pointer
[   50.995048] Register r3 information: 5-page vmalloc region starting at 0xbf0e6000 allocated at load_module+0x9dc/0x2878
[   50.995059] Register r4 information: slab kmalloc-2k start c181e000 pointer offset 960 size 2048
[   50.995068] Register r5 information: slab kmalloc-2k start c181e000 pointer offset 0 size 2048
[   50.995077] Register r6 information: NULL pointer
[   50.995081] Register r7 information: non-slab/vmalloc memory
[   50.995084] Register r8 information: non-paged memory
[   50.995088] Register r9 information: slab kmalloc-1k start c2d39000 pointer offset 0 size 1024
[   50.995096] Register r10 information: non-slab/vmalloc memory
[   50.995101] Register r11 information: non-slab/vmalloc memory
[   50.995104] Register r12 information: non-slab/vmalloc memory
[   50.995108] Process spi4 (pid: 170, stack limit = 0x7a72e8cc)
[   50.995112] Stack: (0xc2c67e70 to 0xc2c68000)
[   50.995116] 7e60:                                     ffffff92 c181e000 c1b16e00 c3df7eac
[   50.995121] 7e80: c2c67eec c2c67e90 c08fd0b0 bf0e7538 c2d39000 c109abb4 003da540 00000000
[   50.995125] 7ea0: 000000c8 00000000 c2d39000 c140689c 00000000 ffffe000 c2d39240 c181e2f8
[   50.995129] 7ec0: c15438a0 c181e000 00000000 bf0e6a1c c3df7eac c3df7ed0 c15438a0 c181e274
[   50.995133] 7ee0: c2c67f2c c2c67ef0 c08fd80c c08fcc68 c0247610 c15438a0 c2c67f2c c181e210
[   50.995137] 7f00: c0341044 00000000 c2842b80 c2842b84 c2c66000 c157f18c c15438a0 c181e274
[   50.995140] 7f20: c2c67f3c c2c67f30 c08fdc3c c08fd354 c2c67f74 c2c67f40 c0247640 c08fdc28
[   50.995144] 7f40: ffffe000 c08fdc1c c2c67f74 c2842bc0 c2ccf940 c0247538 c2842b80 c2d75280
[   50.995148] 7f60: c2c66000 c2c7da7c c2c67fac c2c67f78 c024905c c0247544 c2ccf960 c2ccf960
[   50.995152] 7f80: 00000000 c2842bc0 c0248e84 00000000 00000000 00000000 00000000 00000000
[   50.995155] 7fa0: 00000000 c2c67fb0 c02000d4 c0248e90 00000000 00000000 00000000 00000000
[   50.995159] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   50.995163] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   50.995165] Backtrace: 
[   50.995168] [<bf0e752c>] (bcm2835_spi_handle_err [spi_bcm2835]) from [<c08fd0b0>] (spi_transfer_one_message+0x454/0x6ec)
[   50.995182]  r7:c3df7eac r6:c1b16e00 r5:c181e000 r4:ffffff92
[   50.995184] [<c08fcc5c>] (spi_transfer_one_message) from [<c08fd80c>] (__spi_pump_messages+0x4c4/0x8d4)
[   50.995194]  r10:c181e274 r9:c15438a0 r8:c3df7ed0 r7:c3df7eac r6:bf0e6a1c r5:00000000
[   50.995196]  r4:c181e000
[   50.995198] [<c08fd348>] (__spi_pump_messages) from [<c08fdc3c>] (spi_pump_messages+0x20/0x24)
[   50.995208]  r10:c181e274 r9:c15438a0 r8:c157f18c r7:c2c66000 r6:c2842b84 r5:c2842b80
[   50.995210]  r4:00000000
[   50.995212] [<c08fdc1c>] (spi_pump_messages) from [<c0247640>] (kthread_worker_fn+0x108/0x2f0)
[   50.995220] [<c0247538>] (kthread_worker_fn) from [<c024905c>] (kthread+0x1d8/0x214)
[   50.995229]  r10:c2c7da7c r9:c2c66000 r8:c2d75280 r7:c2842b80 r6:c0247538 r5:c2ccf940
[   50.995231]  r4:c2842bc0
[   50.995233] [<c0248e84>] (kthread) from [<c02000d4>] (ret_from_fork+0x14/0x20)
[   50.995240] Exception stack(0xc2c67fb0 to 0xc2c67ff8)
[   50.995244] 7fa0:                                     00000000 00000000 00000000 00000000
[   50.995247] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   50.995251] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   50.995255]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0248e84
[   50.995257]  r4:c2842bc0
[   50.995260] Code: e8bd4000 e5906388 e5904040 e1a05000 (e5963000) 
[   50.995300] ---[ end trace 0000000000000002 ]---
[   51.954808] sched: RT throttling activated
[   52.481886] mcp251xfd spi0.0: SPI transfer timed out
[   52.481907] spi_master spi0: failed to transfer one message from queue
[   52.481914] mcp251xfd spi0.0 can0: IRQ handler mcp251xfd_handle_rxif() returned -110.
[   52.481922] mcp251xfd spi0.0 can0: IRQ handler returned -110 (intf=0x3f1a0002).

pelwell, 2022-07-21:
A fix for shared SPI interrupts went into the downstream Raspberry Pi kernel recently: 46feb7d7241b3f568910a054c1c39a41ab515477
This appeared in our rpi-firmware release (available via `rpi-update`) as of June 9th, which presents as kernel version 5.15.45. That might make you think that your 5.15.49 kernel (https://github.com/raspberrypi/linux/commits/ee039006371a0b1d64d825a59f0eed8627bb3c91) would include the patch, but unfortunately that commit is in the upstream tree, which lacks our downstream changes. If instead you used the 5.15.49 merge commit (f0da4e5685af) you would include the fixing commit, but then you'd be likely to have problems applying the -rt patch set.

There's a pair of commits in the upstream tree that add the flawed shared interrupt support then remove it:
```
ecfbd3cf3b8b spi: bcm2835: Enable shared interrupt support
d62069c22eda spi: bcm2835: Remove shared interrupt support
```
I suggest you try re-applying the "Enable" commit and then cherry-pick the fix (46feb7d7241b3f568910a054c1c39a41ab515477) on top of that. Unfortunately there has been a change to the SPI driver in the meantime that means that cherry-picking the "Enable" commit hits a merge conflict, but the combination of that patch and the fix amounts to two simple changes:
```diff
index 775c0bf2f923..9bdb1b85ae08 100644
--- a/drivers/spi/spi-bcm2835.c
+++ b/drivers/spi/spi-bcm2835.c
@@ -372,6 +372,10 @@ static irqreturn_t bcm2835_spi_interrupt(int irq, void *dev_id)
        struct bcm2835_spi *bs = dev_id;
        u32 cs = bcm2835_rd(bs, BCM2835_SPI_CS);
 
+       /* Bail out early if interrupts are not enabled */
+       if (!(cs & BCM2835_SPI_CS_INTR))
+               return IRQ_NONE;
+
        /*
         * An interrupt is signaled either if DONE is set (TX FIFO empty)
         * or if RXR is set (RX FIFO >= ¾ full).
@@ -1365,8 +1369,8 @@ static int bcm2835_spi_probe(struct platform_device *pdev)
        bcm2835_wr(bs, BCM2835_SPI_CS,
                   BCM2835_SPI_CS_CLEAR_RX | BCM2835_SPI_CS_CLEAR_TX);
 
-       err = devm_request_irq(&pdev->dev, bs->irq, bcm2835_spi_interrupt, 0,
-                              dev_name(&pdev->dev), bs);
+       err = devm_request_irq(&pdev->dev, bs->irq, bcm2835_spi_interrupt,
+                              IRQF_SHARED, dev_name(&pdev->dev), bs);
        if (err) {
                dev_err(&pdev->dev, "could not request IRQ: %d\n", err);
                goto out_dma_release;
```
Try just applying those changes on top of your most recent -rt tree and see if it helps.

marckleinebudde, 2022-07-21:
Hello Phil,

exactly, the uses kernel https://github.com/kdoren/linux/tree/rpi-5.15.40-rt is missing https://github.com/raspberrypi/linux/commit/46feb7d7241b3f568910a054c1c39a41ab515477.

I've cherry picked it to https://github.com/marckleinebudde/linux-rpi/tree/rpi-5.15.40-rt and created a new v5.15.x-rt here: https://github.com/marckleinebudde/linux-rpi/tree/rpi-5.15.55-rt

Sorry for not updating this issue. Feel free to close.

pelwell, 2022-07-21:
Thanks, Marc. for maintaining the -rt branches and supporting other users.

@framet I'm going to close this issue in anticipation of a successful outcome, but feel free to comment or reopen it if you still have difficulties.

closed 2022-07-21 by pelwell

marckleinebudde, 2022-07-21:
FYI: I'm currently mainlining the fixed shared IRQ support for the `spi-bcm2835.c` https://lore.kernel.org/all/20220719105305.3076354-1-mkl@pengutronix.de

pelwell, 2022-07-21:
Thanks - you were kind enough to credit and CC me.

marckleinebudde, 2022-07-21:
Sure, I was just picking up the bits an peaces :)