raspberrypi/linux #5097: ext4 regression causes huge load

lategoodbye, 2022-07-19:
### Describe the bug

[Recently](https://archlinuxarm.org/forum/viewtopic.php?f=67&t=16061) users of Arch Linux ARM complained about huge load on IO task like unpacking bigger tar archives starting with Linux 5.18. Because the error descriptions was mixed with SD cards issues it was hard to identify and i suspect the ext4 partition must be in a "used" state.

Last week i found a good scenario to bisect the issue. Here is the related upstream bug report:
https://marc.info/?l=linux-ext4&m=165815090631395&w=3

Unfortunately the offending change 3c65b7309d2e0dd8d134e7813cd95debd91ea07b has been backported to Linux 5.15.33, but a revert of the commit should fix the problem.

### Steps to reproduce the behaviour

- find a SD card with Raspberry Pi OS a used ext4 rootfs
- make sure Linux kernel 5.15.33 (RPI and mainline are affected) is installed
- run "sudo rpi-update"

Expected behavior:
- update is download and installed

Observed behavior:
- update hangs during download and abort or takes very long

### Device (s)

Raspberry Pi 4 Mod. B

### System

Kernel version: 5.15.55

pelwell, 2022-07-19:
Thanks - that's reverted on the current branches.

closed 2022-07-20 by pelwell

lategoodbye, 2022-09-11:
Hi, just a short update on this issue after a lot of work. There is a patch series which hopefully will be applied soon:
https://lore.kernel.org/all/20220908091301.147-1-jack@suse.cz/

popcornmix, 2022-09-11:
Thanks for info. I'm guessing it will be backported to linux-5.15.y and I'll know to revert the revert when the conflict comes.

Phoenix591, 2022-09-26:
Looks like it wasn't reverted but reportedly fixed in the latest ext4 changes in 6.0-rc7 https://github.com/torvalds/linux/commit/5e049663f678938dd56812ba870a471060f8ce28

lategoodbye, 2022-11-11:
@popcornmix No, need to revert this patch in rpi-5.15.y anymore. The fixes has been backported at the end of September:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/fs/ext4/mballoc.c?h=v5.15.78&id=fd8b829195495a0b3d6b3cc88c3b77f2c9c115e4

popcornmix, 2022-11-12:
Reverted the revert. Thanks for heads up.