raspberrypi/linux #5100: Kernels 5.18 and 5.19 lock up several seconds after booting on 3B+

matejdro, 2022-07-21:
### Describe the bug

I've been trying to build newer kernels for my Pi 3B+ (I mostly want [New IR toy driver](https://github.com/torvalds/linux/commits/master/drivers/media/rc/ir_toy.c)), but I've been unsuccessful so far.

Building and installing kernel 5.15 (from `rpi-5.15.y` branch) works perfectly fine, but attempting to build kernels 5.18 or 5.19 results in Pi freezing several seconds after boot. Freeze time appears to be random.

Sometimes it freezes just after "welcome" line: 
[image]

and sometimes it will work long enough for me to SSH into it, but it will still eventually freeze. Cursor on HDMI display will stop blinking and SSH connection will be terminated. Both LEDs on the PI will glow solid. 

Is Pi 3+ still supposed to have newer kernel support or is it EOL?

### Steps to reproduce the behaviour

1. Have working linux on Raspberry PI 3+ (with Kernel 5.15)
2. Build and install kernel according to [Official instructions](https://www.raspberrypi.com/documentation/computers/linux_kernel.html) from either branches `rpi-5.18.y` or `rpi-5.19.y`

### Device (s)

Raspberry Pi 3 Mod. B+

### System

I've tried building:

5.18 https://github.com/raspberrypi/linux/commit/ab60b2d25f223738745dee81c02db12e9866f1c3

5.19 https://github.com/raspberrypi/linux/commit/e4765ec5832efe7258a85e74d54c29e4da032170

Firmware:

```
Mar  1 2022 14:23:32 
Copyright (c) 2012 Broadcom
version 969fb9b1521fc7ac2b88b15a3a9e942da7678c4d (clean) (release) (start)
```

OS is Arch Linux ARM (with latest dependencies as of writing this).

### Logs

There are no visible entries in dmesg, device just locks up at random points after several seconds

### Additional context

I have a 5V 2.5A power supply. That Pi has been running 24/7 for several years now under that power supply with zero issues, plus, when on kernel 5.15, Pi has no issues with demanding tasks like building entire kernel, so I don't think it's the power supply

pelwell, 2022-07-21:
If you manage to log in once more would you please capture the dmesg output to a file somewhere it can be extracted, or use ssh from another computer, i.e. `ssh Pi dmesg > dmesg.txt`. It might show a difference between your system and one that doesn't freeze.

matejdro, 2022-07-22:
I've tried that and there were no dmesg messages at the time of freezing

pelwell, 2022-07-22:
I mean all the messages from the kernel boot process. Are you saying that there are none at all?

pelwell, 2022-07-22:
And I should have said that 3B+ is definitely not EOL - we are still supporting all Pis, including the original Model B from 2012.

pelwell, 2022-07-22:
Testing hasn't shown any problems so far with either 5.18.12-v7+ or 5.19.0-rc7-v7+ and Bullseye RPiOS on 3B+.

Looking at your screenshot, I don't like the look of the following error message:
```
Warning: /lib/modules/5.19.0-rc7-v7+/modules.devname: not found - ignoring
```
Are you sure you are installing the kernels correctly?

matejdro, 2022-07-24:
Here is full dmesg up until it freezes:

[dmesg.txt](https://github.com/raspberrypi/linux/files/9175957/dmesg.txt)

After some testing, once kernel gets past the booting stage (and I can SSH into it), freeze only appears to happen when I start some sort of demanding operation (such as building kernel again). If I leave pi idle, it seems to work for extended amounts of time.

Could it be power supply after all? But building on 5.15 works without a hitch. Maybe power management is somehow different?

> Are you sure you are installing the kernels correctly?

Totally possible that I'm mucking something up, that's the first time I'm doing this. But not sure what could I be doing wrong, I'm just following commands, according to the official manual:

```bash
git clone -b rpi-5.19.y --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make bcm2709_defconfig
make -j4 zImage modules dtbs
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
sudo reboot
```

pelwell, 2022-07-24:
The build method looks OK to me.

Does adding `over_voltage=2` to config.txt make it more stable?

matejdro, 2022-07-30:
Unfortunately, that made it even worse. I could not even get it to the point where SSH would be available.

Here are snapshots of some of the freezes:

[image]
[image]

[image]
[image]

pelwell, 2022-07-30:
Random failures in bits of common code like this are almost always due to power problems. Booting is extremely power hungry because of so many subsystems being started simultaneously.

If you have a USB keyboard, try holding down shift once the kernel messages start to appear to prevent the switch to the on-demand governor. Alternatively, add `arm_freq=900` to config.txt.

matejdro, 2022-07-31:
Huh, does that mean that newer kernels are that much power hungry than the 5.15?

I guess  I will stay at 5.15 for now and get a better power supply later.

Thanks a lot for your help.

closed 2022-07-31 by matejdro

pelwell, 2022-08-09:
I don't think they are more power hungry as such, but code improvements and newer compilers can make code more efficient, meaning that it runs more quickly, using more power in the moment but less power overall because the task takes less time. It can also be that a seemingly unrelated change results in the movement of some instructions in memory, which can cause different power demands in some part of the SoC - enough to tip a marginal system over the edge.

badbat75, 2022-09-01:
Same issue for me on a RPi 1.
With an USB keyboard it hangs after USB initialization, disconnecting it, the system go further until the USB wifi starts, disconnecting USB wifi the system boot until the end, but without keyboard I'm not able to check if it is stuck.
Of course connecting the usb keyboard after the boot, it hangs.
Tested 5.18 (all versions) and 5.19.0 -> 5.19.5
I guess it's related to usb initialization, but I can't debug further.