raspberrypi/linux #5102: wm8960 audio codec not detected

naguirre, 2022-07-22:
### Describe the bug

I"m using rpi linux kernel 5.15.55 on a Raspbery PI Zero 2 W with a ReSpeaker 2-Mics Pi HAT. I'm trying to get the sound card detected, but it's not.

### Steps to reproduce the behaviour

Build a kernel with bcm2709_defconfig.

Append to config.txt : 
```
dtparam=i2s=on
dtparam=i2c_arm=on
dtparam=audio=on
dtparam=i2c1=on
dtoverlay=wm8960-soundcard
```
Plug Raspbery PI Zero 2 W with a ReSpeaker 2-Mics Pi HAT
Boot on the kernel with a minimal distro (buildroot)

```
modprobe snd_soc_wm8960
modprobe i2c-dev
```

launch i2cdetect
```
# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
# 
```

wm8960 is detected at 0x1a address, but iit's not owned by the wm8960 driver. `UU` should be displayed.
After adding few dev_dbg in wm8960 kernel module, it seems that i2c driver is never probed. 

After changing the dtb overlay (https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/wm8960-soundcard-overlay.dts#L32) and adding `@1a`, i2c driver is correctly loaded.

Is `@1a` necessary for the i2c driver to be probed ? Is `reg = <0x1a>;` not sufficient ? 

However, even after codec were correctly handled by i2c, I was not able to get a sound card, with the dtbo provided. 
I tried to load simple-card driver
```
modprobe snd_soc_simple_card
```

Nothing happen.
Is there something wrong with the dtbo ?

### Device (s)

Raspberry Pi Zero 2 W

### System

```
# cat /etc/os-release 
NAME=Buildroot
VERSION=2022.05-14-g3f7d62b5c3
ID=buildroot
VERSION_ID=2022.05
PRETTY_NAME="Buildroot 2022.05"

```
[... 334 lines cut ...]
002584.438: Loaded overlay 'miniuart-bt'
002587.776: dtparam: i2s=on
002599.340: dtparam: i2c_arm=on
002611.124: dtparam: audio=on
002624.699: dtparam: i2c1=on
002690.830: brfs: File read: 1819 bytes
002700.013: brfs: File read: /mfs/sd/overlays/wm8960-soundcard.dtbo
002730.445: Loaded overlay 'wm8960-soundcard'
002792.652: brfs: File read: 2295 bytes
002796.402: brfs: File read: /mfs/sd/cmdline.txt
002801.228: Read command line from file 'cmdline.txt':
002807.119: 'root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200'
002827.263: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
003194.824: brfs: File read: 65 bytes
005605.028: brfs: File read: /mfs/sd/zImage
005608.536: Loading 'zImage' to 0x8000 size 0x2811d40
005614.358: Device tree loaded to 0x19be7e00 (size 0x81c6)
005621.387: uart: Set PL011 baud rate to 103448.300000 Hz
005628.250: uart: Set PL011 baud rate to 103448.300000 Hz
005628.296: uart: Baud rate change done...
005631.710: uart: Baud rate change done...
005637.132: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
010252.739: vchiq_core: vchiq_init_state: slot_zero = 0xd5980000, is_master = 1

```

pelwell, 2022-07-22:
1. When you say you have a minimal, buildroot distro, does that mean there is no udev and you are expecting to have to modprobe kernel modules explicitly?
2. If the answer to 1 is "no", when you had changed the node name in the overlay to `wm8960@1a`, did you find that the sound_soc_wm8960 module was loaded automatically?
3. snd_soc_simple_card is the correct soundcard driver in this case. Does modprobe load it successfully?

My understanding of i2c device node naming is that the `@<a>` suffix was a convention, not a requirement, but perhaps that has changed recently. I'll take a look.

pelwell, 2022-07-22:
Testing with a modified i2c-rtc overlay confirms that the lack of an address value in the i2c device node name makes no difference. From the modified overlay:
```
	fragment@3 {
		target = <&i2cbus>;
		__dormant__ {
			#address-cells = <1>;
			#size-cells = <0>;

			ds3231: ds3231 {
				compatible = "maxim,ds3231";
				reg = <0x68>;
			};
		};
	};
```
which when loaded gives:
```
pi@raspberrypi:~$ ls /proc/device-tree/soc/i2c@7e804000/
'#address-cells'   compatible   name        pinctrl-names   status
 clock-frequency   ds3231       phandle     reg
 clocks            interrupts   pinctrl-0  '#size-cells'
pi@raspberrypi:~$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

naguirre, 2022-07-22:
Ok, my bad, I was not mounting /sysfs at startup, so module were not modprobe by udev. Now everything is fine, and as you said, no need to add @addr it works without.
I'm now able to play and record sound with wm8960

Thanks, i close this issue.

closed 2022-07-22 by naguirre