raspberrypi/linux #5108: kernel git 5.19-rc8 is out

xevilstar, 2022-07-28:
### Describe the bug

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git

### Steps to reproduce the behaviour

look at the makefile in the actual raspberrypi git and the version is still -rc7

### Device (s)

Raspberry Pi 4 Mod. B

### System

 cat /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4
vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
 uname -a
Linux raspberrypi 5.19.0-rc7-v8+ #1 SMP PREEMPT Tue Jul 19 14:43:47 CEST 2022 aarch64 GNU/Linux

pelwell, 2022-07-28:
This kind of Issue just creates work for us - it's not like we ever don't pull the latest versions.

closed 2022-07-28 by pelwell

pelwell, 2022-07-28:
You mean you have more to say on the matter?

xevilstar, 2022-10-11:
No, I'm not.