raspberrypi/linux #5110: The BT driver does not work or does not exist

girl131, 2022-07-31:
### Describe the bug

My RPi4b is in an aluminum box and BT has big problems with latency.

I buy: ASUS USB-BT500
Driver: https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-BT500/HelpDesk_Download/

But the drivers won't install and won't work for me: https://pastebin.com/raw/ycmByCBg

I keep getting the error: No default controller available

### Steps to reproduce the behaviour

After connecting the USB BT dongle (ASUS USB-BT500) does not work

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspios bullseye arm64

parheliamm, 2022-08-02:
Why don't leverage built-in BT directly?

popcornmix, 2022-08-02:
I'd suggest returning that dongle and getting one supported by (upstream) linux directly.
Any driver that requires you to compile the modules is going to be a pain
(you will have to rebuild modules after any kernel update from apt).
And that relies on them constantly updating the driver source to match newer kernels.

girl313, 2022-08-02:
> Why don't leverage built-in BT directly?
can you read?

I repeat again: My RPi4b is in an aluminum box and BT has big problems with latency.

girl313, 2022-08-02:
> I'd suggest returning that dongle and getting one supported by (upstream) linux directly. Any driver that requires you to compile the modules is going to be a pain (you will have to rebuild modules [...]

That's what I thought too, unfortunately I have absolutely no idea how to find out which BTs are supported according to chipset or firmware, where and how can I find it?

Or can you recommend to me a supported bluetooth 5.0+?

popcornmix, 2022-08-02:
> Or can you recommend to me a supported bluetooth 5.0+?

Sorry no, I don't use an external BT dongle.
In general I'd always search for make/model of device and linux and see what it says.
If there is a link to a zip file with some source files in, or no references to linux at all, then I'd avoid.
Ideally you'll find comments like "works out of the box on linux" (or they may even mention on raspberry pi).

girl313, 2022-08-02:
Can someone smarter explain to me what this is:
https://github.com/raspberrypi/linux/pull/4864
https://github.com/raspberrypi/linux/commit/33404381c5e875cbd57eec6d9bbacd3b13b404c9

Here should be the drivers for Wifi and Bluetooth:

Wifi :
https://github.com/raspberrypi/linux/tree/aeaa2460db088fb2c97ae56dec6d7d0058c68294/drivers/net/wireless

Bluetooth:
https://github.com/raspberrypi/linux/tree/aeaa2460db088fb2c97ae56dec6d7d0058c68294/drivers/bluetooth

I searched for a long time and didn't find any list (USB BT dongle for RPi4 which devices are supported) interestingly, according to the chipset I found only wifi adapters.....

Please do not close until the drivers are added to the RPi (can someone advise me how to do this).

PS: Meantime,, can anyone recommend a BT Dongle that is 100% supported, thanks!

parheliamm, 2022-08-03:
Try google, you can find the below link:
https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-BT500/HelpDesk_Download/
Compile ko by yourself.

popcornmix, 2022-08-03:
Using a standard kernel (not one you've built yourself),
and the BT500 plugges in, what does:
```
uname -a
lsusb
```
report?

girl313, 2022-08-03:
uname -a
`Linux rpitest 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`

lsusb
`Bus 001 Device 004: ID 0b05:190e ASUSTek Computer, Inc. ASUS USB-BT500`

girl313, 2022-08-03:
@parheliamm 

> Try google, you can find the below link: https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-BT500/HelpDesk_Download/ Compile ko by yourself.

I tried and I don't know how.

If I had known, I wouldn't have posted here.
Do you have more any extra useless advice?

girl313, 2022-08-03:
@popcornmix 

I turned off the internal bluetooth using the command: (dtoverlay=disable-bt in the config.txt file)

I plug in (USB-BT Dongle) I reboot

sudo bluetoothctl scan on
`No default controller available`

I turned off the internal bluetooth using the command: (dtoverlay=disable-bt in the config.txt file)
I plug in (USB-BT Dongle) I reboot

https://i.postimg.cc/yNxdRbqr/image.png

popcornmix, 2022-08-03:
Does `lsmod` show any module related to BT500? Can you see any reference in `dmesg`?

girl313, 2022-08-03:
lsmod: https://pastebin.com/raw/zRJsU325

dmesg: https://pastebin.com/raw/kifpnBpp

`[    2.196477] usb 1-1.1: Product: ASUS USB-BT500
[    2.196491] usb 1-1.1: Manufacturer: Realtek`\

I don't see anything else :-(

girl313, 2022-08-03:
does not contain command: (dtoverlay=disable-bt in the config.txt file)

(with internal BT turned on) lsmod: https://pastebin.com/raw/jLqaF8jb

(with internal BT turned on) dmesg: https://pastebin.com/raw/HqNkZiy8

parheliamm, 2022-08-04:
> @parheliamm
> 
> > Try google, you can find the below link: https://www.asus.com/Networking-IoT-Servers/Adapters/All-series/USB-BT500/HelpDesk_Download/ Compile ko by yourself.
> 
> I tried and I don't know how.
> 
> If I had known, I wouldn't have posted here. Do you have more any extra useless advice?

If you think compile a out of tree ko is useless option at this point, I have to say you cannot enable your  BT device any more.

girl313, 2022-08-04:
@parheliamm Stop writing stupid and spam answers and saying something I didn't say.

I know what your next answer will be create your own driver...

I repeat! I tried installing the driver and it didn't work for my (Yes, I'm probably doing something wrong and that's why I'm writing here).

648643579, 2022-08-05:
Asus has two bluetooth usb adapter BT400 and BT500.
Even if these are popular models, they don't have support, that's just Linux.
But I have to admit that support for these models should be added.

@parheliamm
zamysli se než začneš psát nesmysly.

parheliamm, 2022-08-05:
If you guys want to enable BT module directly without doing anything, RPI4 and Linux are not a smart choice for your guys.
I repeat one more time, trying to compile KO based on out of tree source code. That is the only way you could enable your BT device.
If you guys could study how to leverage dkms, that would be better than compiling KO directly.

parheliamm, 2022-08-05:
One more hint:
If you want to try to build a kernel, refer to the below link.
https://www.raspberrypi.com/documentation/computers/linux_kernel.html

648643579, 2022-08-05:
@parheliamm

I was hoping you had sense, but you are seriously stupid.
Send her instructions that absolutely do not explain the driver assembly problem.

Please, find another hobby :-)

popcornmix, 2022-08-08:
@pelwell do you know status of BT500
It looks like you cherry-picked an upstream commit in https://github.com/raspberrypi/linux/pull/4864 , although I believe the commit is now in upstream 5.15 anyway. But it appears to not work. 

ID is `0b05:190e` which seems to match cherry-picks. Does the source provided driver provide something not available upstream? Or is it just not being enabled correctly (from user side)?

pelwell, 2022-08-08:
The commits I cherry-picked have indeed been replaced by upstream merges, so they can be ignored.

I can see what is happening, and suggest workarounds, but I'm lacking an understanding of _why_ you are in this situation.

The kernel source file drivers/bluetooth/btusb.c declares a large number of USB Bluetooth controllers. The list is split into two groups - one called `btusb_table`, and one called `blacklist_table`. The primary difference between those two seems to be that the former includes a `MODULE_DEVICE_TABLE` declaration that causes its entries to end up in the modules.alias table, while the latter doesn't. An alias allows a matching module to be loaded when a USB device is detected, but there are many devices in the blacklist_table list that can't all be broken, so I can't believe that this partitioning is accidental.

Another observation is that at the top of `btusb_table` is the following:
```
 	/* Generic Bluetooth USB device */
	{ USB_DEVICE_INFO(0xe0, 0x01, 0x01) },

	/* Generic Bluetooth AMP device */
	{ USB_DEVICE_INFO(0xe0, 0x01, 0x04), .driver_info = BTUSB_AMP },

	/* Generic Bluetooth USB interface */
	{ USB_INTERFACE_INFO(0xe0, 0x01, 0x01) },
```
I interpret this to be matching on the device's class, rather than it's vendor/device pair. The probe function for the btusb module explicitly searches blacklist_table for the device, so provided the device matches one of the above declarations the kernel should be able to load a driver for it.

Debugging is hard without one of the devices, but fortunately I have a very old BT dongle that is listed in blacklist_table, not btusb_table, and it's driver is loaded:
```
pi@raspberrypi:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Try these:
1. Run `lsusb` as above on your system with the dongle attached. Then take the Bus (1, in my case) and Device number (3, in my case) and use them (with a separating colon) in the following command, replying here with your output:
```
pi@raspberrypi:~$ lsusb -v -s 1:3 | head -16
Couldn't open device, some information will be missing

Bus 001 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0a12 Cambridge Silicon Radio, Ltd
  idProduct          0x0001 Bluetooth Dongle (HCI mode)
  bcdDevice            5.25
  iManufacturer           0
  iProduct                0
  iSerial                 0
```
2. What do the following commands return:
```
pi@raspberrypi:~$ sudo modprobe btusb
pi@raspberrypi:~$ hcitool dev
```

girl313, 2022-08-09:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:190e ASUSTek Computer, Inc. ASUS USB-BT500
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`
Bus 001 Device 003: ID 0b05:190e ASUSTek Computer, Inc. ASUS USB-BT500
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x190e
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 ASUS USB-BT500
  iSerial                 3 00E04C131687
`

https://pastebin.com/raw/Ah1MawHy

girl313, 2022-08-09:
@pelwell 
If you wanted I can give you access to the RPi4 via RealVNC.

just write to this email: 6284862468@email.cz

pelwell, 2022-08-09:
And the other commands?
```
pi@raspberrypi:~$ sudo modprobe btusb
pi@raspberrypi:~$ hcitool dev
```

girl313, 2022-08-09:
sudo modprobe btusb > nothing

hcitool dev
`Devices:
        hci0    DC:A6:32:66:E1:6B
`

pelwell, 2022-08-09:
Does the modprobe cause btusb to be loaded? `lsmod | grep btusb`

girl313, 2022-08-09:
```
btusb                  45056  0
btrtl                  24576  1 btusb
btintel                40960  1 btusb
btbcm                  28672  2 hci_uart,btusb
bluetooth             450560  35 btrtl,btintel,hci_uart,btbcm,bnep,btusb,rfcomm

```

pelwell, 2022-08-09:
Try this:
1. Unplug the dongle and reboot.
2. In a console winfow run `sudo udevadm monitor -k -p | tee udevlog.txt`.
3. Plug in the dongle.
4. When console activity stops (it shouldn't be more than a second or so), hit Ctrl-C and upload udevlog.txt somewhere.

girl313, 2022-08-09:
https://pastebin.com/raw/DYYKHsaG

pelwell, 2022-08-09:
That output is almost identical to mine - note the `DRIVER=btusb` and `add /module/btusb`. Did that really not cause the btusb module to be loaded?

girl313, 2022-08-09:
I performed a clean installation so as not to affect any other logs or information that I am sending you.

And my BT is not working or am I doing something wrong..
Maybe I'm doing something wrong and I'm really stupid :-D

I don't want to bother you and waste your time, but if you would like to check problem yourself, you can write to me on that email and I will send you the information for RealVNC. (In my opinion, the best solution)

pelwell, 2022-08-09:
I appreciate the offer, but any sort of debugging is likely to require custom kernel builds. I think it will be more efficient to acquire one of the dongles (and perhaps a few others) for my collection.

girl313, 2022-08-09:
Thank you and I appreciate your efforts to help me :-)

pelwell, 2022-08-10:
The dongle arrived, and when I plugged it in the btusb module was immediately loaded, and the following messages appeared in the kernel log:
```
[23119.999152] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[23120.001126] Bluetooth: hci1: RTL: rom_version status=0 version=1
[23120.001163] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[23120.002485] bluetooth hci1: Direct firmware load for rtl_bt/rtl8761bu_fw.bin failed with error -2
[23120.002532] Bluetooth: hci1: RTL: firmware file rtl_bt/rtl8761bu_fw.bin not found
[23120.040048] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[23120.043026] Bluetooth: hci1: RTL: rom_version status=0 version=1
[23120.043037] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[23120.043079] bluetooth hci1: Direct firmware load for rtl_bt/rtl8761bu_fw.bin failed with error -2
[23120.043088] Bluetooth: hci1: RTL: firmware file rtl_bt/rtl8761bu_fw.bin not found
```
The missing firmware can be found in the upstream kernel firmware repo: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtl_bt
Installing the necessary files gave me a working BT interface (it can scan, at least).

Download the following files:
* https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtl_bt/rtl8761bu_fw.bin
* https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtl_bt/rtl8761bu_config.bin

Copy them to /lib/modules/rtl_bt/ and try again.

girl313, 2022-08-11:
It seems to work but I used a different path "/lib/firmware/rtl_bt/"

How it actually works
Can two bluetooth work together?

When the signal for one BT is weak, will the others automatically switch?

So what is the best way to use USB bluetooth to be set as main?

pelwell, 2022-08-12:
> It seems to work but I used a different path "/lib/firmware/rtl_bt/"

Yes, that's correct - it's what I wanted to type, but my muscle memory had other ideas.

> How it actually works
> Can two bluetooth work together?

Sadly I think the answer is "not very well", or "not easily".

> When the signal for one BT is weak, will the others automatically switch?

No, I don't think so. Unlike WiFi (and wired networking), Bluetooth doesn't have a routing layer (IP, for networks) that can choose which interface to use. Bluetooth appears as independent interfaces, and the application software (which includes the GUI) has to choose which interface to use. My colleague who creates and maintains the Raspberry Pi desktop environment doubted that the Bluetooth plugin would cope with an additional interface.

> So what is the best way to use USB bluetooth to be set as main?

Disable the onboard interface - put `dtoverlay=disable-bt` in config.txt and reboot.

pelwell, 2022-08-12:
The help for that overlay says:
```
pi@raspberrypi:~$ dtoverlay -h disable-bt
Name:   disable-bt

Info:   Disable onboard Bluetooth on Pi 3B, 3B+, 3A+, 4B and Zero W, restoring
        UART0/ttyAMA0 over GPIOs 14 & 15.
        N.B. To disable the systemd service that initialises the modem so it
        doesn't use the UART, use 'sudo systemctl disable hciuart'.

Usage:  dtoverlay=disable-bt

Params: <None>
```
So you may also want to run `sudo systemctl disable hciuart`.

girl313, 2022-08-12:
> Yes, that's correct - it's what I wanted to type, but my muscle memory had other ideas.
>
>
> Sadly I think the answer is "not very well", or "not easily".
>
>
> No, I don't think so. Unlike WiFi (and wired [...]

THANK YOU FOR EVERYTHING, I LOVE YOU :-D

PS:

I somehow don't understand why it's good to turn off sudo systemctl disable hciuart?

This is a service that is connected precisely with USB BT, isn't it?

pelwell, 2022-08-12:
HCI is the main Bluetooth control interface, but the "uart" part of `hciuart` refers to the serial interface with which the onboard BT modem is connected. Because there is no way to auto-detect devices at the end of a UART (unlike USB), something has to explicitly configure the BT modem - that is what the `hciuart` service does in RPiOS (it can also be done using Device Tree).

USB supports device discovery and device classes, so the initialisation is automatic once you have the correct driver (and firmware!) installed.

pelwell, 2022-08-12:
Please close the issue if you are satisfied with the answers.

girl313, 2022-08-12:
**thank you!!!** and close :-)

closed 2022-08-12 by girl313