raspberrypi/linux #5117: Enable CONFIG_MT7921U

leezu, 2022-08-06:
### Describe the bug

WiFi-6 USB Adapters with Linux kernel support are on the market now. But rpi does not support them out of the box. https://github.com/morrownr/USB-WiFi/issues/87

Please add CONFIG_MT7921U=m for 5.18+.

### Steps to reproduce the behaviour

Verify CONFIG_MT7921U=m is not mentioned in https://github.com/raspberrypi/linux/blob/40ee256/arch/arm64/configs/bcm2711_defconfig

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

pelwell, 2022-08-08:
Done.

leezu, 2022-08-09:
Thank you, @pelwell. As follow-up, would you mind including the latest upstream firmware for these USB Wifi cards? The currently packaged versions cause hangs and much time will be wasted for RPi users if each one needs to update the firmware manually. I've opened a new issue over at RPi-Distro: https://github.com/RPi-Distro/firmware-nonfree/issues/30 Thank you!

closed 2022-08-09 by leezu