raspberrypi/linux #5119: raspberrypi 4 don't show unicode between 8602(↚) and 8703(⇿)

dty717, 2022-08-09:
### Describe the bug

Raspberrypi 4 just show □ when unicode is between 8602(↚) and 8703(⇿).

### Steps to reproduce the behaviour

Copy the unicode between 8602(↚) and 8703(⇿), it will show  □.

### Device (s)

Raspberry Pi 4 Mod. B

### System

 OS and version:Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage5
firmware :version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
kernel :Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

### Logs

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.32-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP Thu Mar 31 19:39:41 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 503104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=E4:5F:01:BF:F0:11 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=5d697b61-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000015a00000-0x0000000019a00000] (64MB)
[    0.000000] Memory: 1587568K/2019328K available (10240K kernel code, 1390K rwdata, 3292K rodata, 2048K init, 590K bss, 104080K reserved, 327680K cma-reserved, 1232896K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35355 entries in 104 pages
[    0.000000] ftrace: allocated 104 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x574/0x758 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000299] Console: colour dummy device 80x30
[    0.000363] printk: console [tty1] enabled
[    0.000429] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[... 395 lines cut ...]
[   13.985664] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   13.985696]  MAI: soc_pcm_open() failed (-19)
[   13.986025] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   13.986043]  MAI: soc_pcm_open() failed (-19)
[   13.986328] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   13.986345]  MAI: soc_pcm_open() failed (-19)
[   15.763813] Bluetooth: Core ver 2.22
[   15.763911] NET: Registered PF_BLUETOOTH protocol family
[   15.763923] Bluetooth: HCI device and connection manager initialized
[   15.763945] Bluetooth: HCI socket layer initialized
[   15.763962] Bluetooth: L2CAP socket layer initialized
[   15.763990] Bluetooth: SCO socket layer initialized
[   15.779685] Bluetooth: HCI UART driver ver 2.3
[   15.779707] Bluetooth: HCI UART protocol H4 registered
[   15.779797] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.781063] Bluetooth: HCI UART protocol Broadcom registered
[   16.043526] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.043539] Bluetooth: BNEP filters: protocol multicast
[   16.043552] Bluetooth: BNEP socket layer initialized
[   16.064239] NET: Registered PF_ALG protocol family
[   16.084994] cryptd: max_cpu_qlen set to 1000
[   16.169636] Bluetooth: RFCOMM TTY layer initialized
[   16.169664] Bluetooth: RFCOMM socket layer initialized
[   16.169682] Bluetooth: RFCOMM ver 1.11
[   31.833193] cam-dummy-reg: disabling

pelwell, 2022-08-09:
Which application are you trying to display these characters in:
* Chromium web browser?
* A Terminal window in the GUI?
* A bare framebuffer console on the display (e.g. boot to console, or after pressing Ctrl-F1)?

For me, viewing this issue in Chromium shows the expected characters, and cutting/pasting them into Terminal also works. Only the third option, the framebuffer console, doesn't. This is expected behaviour - the virtual console only supports 256- or 512-character fonts - type `showconsolefont` to see which characters are available.

dty717, 2022-08-10:
Matchbox-keyboard works well in OS64, but has the problem in OS32, but I am not sure if it's the application problem.

elsaco, 2022-09-18:
@dty717 please check your locale settings. With:

```
[tux@raspberrypi ~]$ localectl
   System Locale: LANG=en_US.UTF-8
       VC Keymap: us
      X11 Layout: us
```

Unicode characters are being shown as expected. Sample `gnome-terminal` output:

[image]