raspberrypi/linux #5123: Box hdd connected to USB 3 hangs on reboot (with USB2 no problem).

0202ru, 2022-08-14:
### Describe the bug

Orico 3.5" Box Ver. 3139U3 Rev. 1.0 with support UASP 18Tb
https://aliexpress.ru/item/1005003621202839.html

HDD Seagate Exos 18Tb with last firmware ver. SN02.

Hdd box connect to USB3 and reboot Raspbian OS x32 or x64 (same in Ubuntu) with reboot command or from GUI. HDD box hangs.

If you connect to USB 2, there is no freezing when restarting the Raspbian OS.

Is there a problem with USB3 drivers in Raspbian OS and Ubuntu? How to fix it?

### Steps to reproduce the behaviour

Detail video and description https://0202.ru/BugOricoExternalBox3.5USB3.0/

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi OS with desktop x32
Release date: April 4th 2022

Raspberry Pi OS with desktop and recommended software x32
Release date: April 4th 2022

Raspberry Pi OS Lite x32
Release date: April 4th 2022

Raspberry Pi OS with desktop x64
Release date: April 4th 2022

Raspberry Pi OS Lite x64
Release date: April 4th 2022

Ubuntu Desktop 22.04.1 LTS x64

Ubuntu Server 22.04.1 LTS x64

P33M, 2022-08-15:
What happens if you apply the steps listed in this thread?
https://forums.raspberrypi.com/viewtopic.php?t=245931

0202ru, 2022-08-16:
Thank you very much! After disabling the "USB Attached SCSI" protocol, the Orico 3.5" Ver. 3139U3 Rev. 1.0 no hangs on reboot.

In order to disable "USB Attached SCSI" protocol I've added to the beginning of the file

/boot/cmdline.txt

the text "usb-storage.quirks=0080:0578:u" with a space at the end and no line break at the end. In my case "0080:0578" - is the id of the Orico 3.5" external box Ver. 3139U3 Rev. 1.0. To see the id of connected USB devices in linux you could run the command lsusb.

I plan to write to Orico support to get them to fix the "USB Attached SCSI" protocol command processing errors for the Orico 3.5" box. 3139U3 Rev. 1.0.

P33M, 2022-08-17:
Using usb-storage instead of UAS will result in a minor reduction in performance. It's on the vendor to provide a UAS-compatible firmware that will work with a Pi 4 - there won't be any changes made to Pi 4 firmware or software here.

closed 2022-08-17 by P33M