raspberrypi/linux #5128: USB 3.0 to SATA adapter causes problems mechanical disk edition

symbios24, 2022-08-15:
### Describe the bug

Hello 

I've been having this problems booting from mechanical drive/s from my Raspbery pi 4B rev1.4

Tested with various usb 3.0 to sata 2.5" / 3.5" controllers and mechanical drives.

Asmedia Asm235cm : very slow performance of the hdd drive even with uasp disabled with quirks

Asmedia Asm1153 : very slow performance of the hdd drive even with uasp disabled with quirks

Initio inic3639 : very slow performance of the hdd drive even with uasp disabled with quirks

Norelsys NS1066: fast perfomance of the hdd drive chipset does not support uasp

Via VLI700 : fast perfomance of the hdd drive chipset does not support uasp

I will update the adapters/chipset list as im testing more of them.

So far the quirks in the problematic/incompatible chipsets does not help the hdd drive to work normal,they work ok though with ssds with uasp enabled.

Does anyone has a solution for asm235cm/asm1153/inic3639 to work good/fast with mechanical hard drives??

Thanks 

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspbian Buster latest version

andrum99, 2022-08-15:
ASM1153 is a "known good" chipset: by that I mean that it is know to work correctly when implemented correctly on a board. Not all adaptor boards are created equal. Can you tell me the make and model of the ASM1153-based adaptor you are using?

symbios24, 2022-08-15:
Hi there

Its SP - The Silicon Power Armor A30 Case

andrum99, 2022-08-15:
> Its SP - The Silicon Power Armor A30 Case

I had a quick google of that: the fact they're using a USB type A connector on both ends of the cable does not inspire confidence that they have a well designed product from an electronic point of view. I'm sure it works with most other computers, but for reasons that are not understood the Pi 4 seems to be more picky about how well the USB 3 circuitry is implemented on attached devices.

I would try with the equivalent Startech product - see https://www.startech.com/en-us/hdd/s251bru33.

It is also possible that the ASM1153 chip needs the E revision to be reliable on the Pi 4 - the Startech adaptor I've linked above uses the E revision, as does the 3.5" enclosure I have from them.

symbios24, 2022-08-15:
I have a no name case also with that type of cable
With vli700 it works great with mechanical hdds over 100mv/sec

JamesH65, 2022-08-16:
We've had a couple of devices sent in to RPi recently with problems on the Pi4's USB. Both devices (I won't name and shame) did NOT adhere to the USB specification, which is why they did not work as expected on the Pi4, despite working on other boxes.

symbios24, 2022-08-16:
I see is there another modern chipset/brand that will work ok with mechanical drives??

What about this one under did you test it before?

https://www.reichelt.com/de/en/external-usb-3-0-hdd-case-sata-3-5-aluminium-black-logilink-ua0107-p107618.html

It has the Asm1153E

Thanks

andrum99, 2022-08-16:
I should not have said that the Pi 4 is 'more picky' about the USB 3 devices it works with - this is not the case. What I was trying to say is that sometimes the Pi 4 has problems with USB 3 devices which are badly designed, and which therefore do not conform to the USB 3 specification. This is not because there is anything wrong with the Pi 4, but because it cannot be expected to operate correctly with devices which do not conform to the USB 3 specification.

You cannot pick a device from a random non-reputable brand and expect it to work correctly all of the time. Sometimes you will be successful, sometimes not, as you have discovered. Startech is a known good brand that makes good quality products, which is why I recommended you try one of their products.

symbios24, 2022-08-16:
Startech looks good but its also expensive so its not in my radar area,it apperas that non uasp cases works better with mechanical hdds,but how i can find an old case/chipset like that in 2022 ??

andrum99, 2022-08-16:
Well in my book expensive and working is better than cheap and not working.

symbios24, 2022-08-16:
You are right, with my bugdet its not an option though,the strange thing is that all the above cases work good in windows,is it possible that a future kernel update will correct the issues?
Or a vl805 firmware??

andrum99, 2022-08-16:
> You are right, with my bugdet its not an option though,the strange thing is that all the above cases work good in windows,is it possible that a future kernel update will correct the issues? Or a vl805 firmware??

Windows on Raspberry Pi, or Windows on a PC? Either way, once you're outside the USB 3 spec you are in the territory of 'undefined behaviour' so it's pot luck whether it works or not. There is nothing to fix on the Pi - the fault lies with the badly designed devices you are trying to attach.

symbios24, 2022-08-16:
Windows for PC
As i said before the above adapters work ok with ssds in the raspberry pi and the pc with mechanical hdds

I believe The problem is most in the mechanical hdds in the raspberry pi and most the uasp support even if its disabled with quirks the hdd is not working correct its like uasp is still there.

Usb2.0 controllers are not a problem at all they all work at max speed 30-32mb/sec without delay.

andrum99, 2022-08-16:
I have two mechanical hard disks in 3.5" UASP USB 3 enclosures running fine on the Pi 4. As I've said before, the root cause of the problem you have is most likely to be that you are using devices which do not conform to the USB 3 spec.

andrum99, 2022-08-16:
> There's almost no point using UASP on a USB to SATA bridge with a mechanical hard disk anyway

I'm using the UASP drives as NAS storage on Pi 4, so anything which reduces CPU usage is a good thing. The Ethernet interface almost saturates one of the ARM cores running interrupt handlers at times.

symbios24, 2022-08-16:
So you are using mechanical drives as nas storage correct?

If yes have you checked their perfomance?

For example hdparm reports over 100mb/sec but the real time performance of the drive is slower as hell in my case with uasp enabled cases

andrum99, 2022-08-16:
> So you are using mechanical drives as nas storage correct?
> 
> If yes have you checked their perfomance?

Correct. I have no problems with the performance of these drives connected to the Pi 4 with UASP enabled.

symbios24, 2022-08-16:
And what cases/chipsets are you using in them?

andrum99, 2022-08-16:
> And what cases/chipsets are you using in them?

Sorry - I seem to have mis-remembered which enclosures had which chipset. The two mechanical hard disks are in 3.5" Startech S3510BMU33 enclosures which have ASM1053 chips. The boot drive is an SSD which is in a 2.5" Startech S2510BMU33 with an ASM1153 chip. The ASM1053 does not do UASP, the ASM1153 does. So I'm not using UASP with a mechanical hard drive - apologies.

symbios24, 2022-08-16:
No problem

I tested 3.5" drives at 7200rpm as boot drives in the raspi in a usb2.0 case the performance was noticeable faster than the same setup with a 2.5" hdd at 5400rom even if the speed was the same 30mb/sec is that because of the extra rpm?

Or because the drive is full sized and not a mobile one?

What i want to say is if i get a 2.5" drive at 7200rpm will i see the same performance as the 3.5" drive at the same rpm?

Thanks

andrum99, 2022-08-16:
In my limited experience, 3.5" desktop and laptop hard disks (i.e. not enterprise server drives) tend to be faster than equivalent 2.5" drives. I've always tried to get 7200rpm 2.5" drives rather than 5400rpm ones,  but I'm not sure there is much of a difference.

symbios24, 2022-08-16:
I found 2 2.5" drives in the recycle bin surprisingly in 100% Health but they were both at 5400 :(

I even boot my Rpi3 from 2.5" hdd no more mem card bottleneck

I use to had many hdds i only keep the largest ones..

I have a couple of ssds but never use them for booting just took them for testing

Just like conventional hdds 

When you have time can you try a normal 3.5" hdd as the boot drive in the raspi with the Asm1153E chipset?

And if its not working correct with uasp can you disable it with quirks and try one more time,just normal operations to verify the realitime drive performance not with hdparm but install something with apt to see the speed when its Installing
For example i install xorg in a non compatible chipset it takes 4-5 minutes in a compatible it takes less than a minute

Thanks

andrum99, 2022-08-16:
Unfortunately I can't test a 3.5" drive with the 2.5" enclosure which has the ASM1153 chip as it is an unpowered enclosure, so won't be able to power the 3.5" drive. I don't have a 3.5x enclosure with that chip. I suggest you ask on one of the forums (at https://forums.raspberrypi.com/) to see if anyone else has got the ASM1153 working properly with an HDD and UASP on the Pi 4.

symbios24, 2022-08-16:
Great thanks i will ask

I want to take a used small size (160-320gb) 2.5" 
 drive at 7200 from ebay but im hesitant about the health of the drive no one of the sellers is mentioning this,i cant take a drive and will have bad sectors.

andrum99, 2022-08-16:
> Great thanks i will ask
> 
> I want to take a used small size (160-320gb) 2.5" drive at 7200 from ebay but im hesitant about the health of the drive no one of the sellers is mentioning this,i cant take a drive and will have bad sectors.

If it's for a boot drive, I personally always use SSDs now - they're so much faster.

symbios24, 2022-08-16:
Yes i know

Im an old school drive user as you can see 😉

andrum99, 2022-08-19:
Just a quick note to say that I just checked on my Raspberry Pi-based NAS and the two HDDs **are** using UASP. They do have ASM1053 chips, and they are using UASP quite happily. My mistake.

symbios24, 2022-08-20:
Just to prevent failures there is hard disk sentinel for raspberry special for nas

https://www.hdsentinel.com/hdslin/hdsentinel-armv8.bz2

andrum99, 2022-08-20:
> Hitachi makes very good 2.5" drives

Officially at least, they haven't been called Hitachi for a few years now - it's HGST - whic was Hitachi Global Storage Technology. Probably Hitachi sold it off and the new owners don't have rights to the Hitachi trademarks. Or something.

symbios24, 2022-08-22:
HGST drives are the best for durability
Seagate are the worst though.

Most of my 2.5" drives are hitach and HGST

I even have 3 ide 2.5" Hitachi drives 20/30/40GB with still 100% health😸

The 40GB is the boot drive in Rpi3

jglathe, 2022-09-29:
I'd suggest trying out the newest kernels, 6.0.0-rc7 has good fixes for the USB issues. https://github.com/raspberrypi/linux/issues/5088#issuecomment-1262714260 Maybe this helps. I can provide .deb packages.

symbios24, 2022-10-05:
Great,when it will be available in raspbian??

pelwell, 2022-10-06:
The fix is also in our 5.15 kernel branch, and has been available in beta kernel builds for a few weeks. If you have a non-critical system and have backed up any valuable data then you can install the latest pre-release kernel using `sudo rpi-update`.

The kernel apt package is updated irregularly and less frequently, but I'll be surprised if it hasn't been bumped by the end of October.