raspberrypi/linux #5130: Raspberry Pi 4 does not start with a clean installation after updating the eeprom

jxrse, 2022-08-17:
### Describe the bug

Raspberry Pi 4 does not start with a clean installation after updating the eeprom,
If I connect an sd or ssd with which I am currently working, yes, it works as it progresses, but if I install a new system on a USB or sd card from scratch, totally clean, it does not advance and when it reaches a few lines, it stops showing the screen and does not respond by ssh does not show anything on the screen

### Steps to reproduce the behaviour

With the Raspberry Imager application I install it on a usd or sdcard raspbian 64 bits desktop, configure it and install it.
I process to insert the sd in my raspberry and you show codes on the screen but when you reach a specific one it always stops working.

### Device (s)

Raspberry Pi 4 Mod. B

### System

This is the current one with which it works
*-*-*
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4

Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

### Logs

sudo rpi-eeprom-update
*-*--**-
BOOTLOADER: up to date
   CURRENT: Thu Mar 10 11:57:12 UTC 2022 (1646913432)
    LATEST: Thu Mar 10 11:57:12 UTC 2022 (1646913432)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

### Additional context

I also tell you that although I try with several sd, usb or ssd and installing other versions that appear in pi imager of the operating systems they do not work and it is due to updating the eeprom

pelwell, 2022-08-17:
It also helps to write a list of all the steps you take, in order.

jxrse, 2022-08-17:
I send you a test video with what happens
Thanks you

https://user-images.githubusercontent.com/55897061/185116200-349bd9b2-606d-413b-bbe6-5c48944f86d5.mp4

timg236, 2022-08-17:
This seems to be a long way into running Linux so the bootloader has been replaced in memory. I'm guessing this is a display issue

jxrse, 2022-08-17:
I will comment again, it is a clean installation from 0 and recent, suddenly it stops giving an image and my router does not detect me, it is not connected to the internet or anything, I do not have ssh access either, I have also tried the mini hdmi with several screens and nothing, it is still not seen and with other sds that previously worked it can be seen perfectly and does not give screenshots indicating no signal

timg236, 2022-08-18:
Since the system is running you could try enabling hdmi_safe=1 to restrict the display resolutions. 

N.B. There's no interaction between the Linux display stack and the bootloader and since we can see that the latest version of both work together (I'm running this right now) we would need SSH / UART logs to help you any further.

jxrse, 2022-08-23:
Sorry if I don't express myself well but I'm Spanish and I still have the problem
I uncommented what I indicated about hdmi safe in 1 , but nothing remains exactly the same.

What exactly do you need me to send you in order to help me because, as I told you, what happened earlier in the video starts and happens. therefore I can't send the new installation any configuration via ssh

DomTel, 2022-09-08:
I have the same problem on the same raspberry pi 4b and I manage to fix it in that way:
After you flash your SD-card or USB-disk with Raspberry PI OS (I use Pi OS Lite 64) open config.txt on boot drive and disable a string dtoverlay=vc4-kms-v3d with #  -> #dtoverlay=vc4-kms-v3d  That's all, raspberry boots as it suppose to.