raspberrypi/linux #5131: Unable to get console

Zarla, 2022-08-17:
### Describe the bug

Hello All,
After booted a 5.19.0 or 5.19.1 kernel into rpi2, rpi3 or rpi4, I'm not able to get a console from the HDMI video cable.
I noticed the following issue with the framebuffer :
```
kernel: bcm2708_fb soc:fb: FB found 1 display(s)
kernel: bcm2835-dma 3f007000.dma: failed to get irq for DMA channel 0
kernel: bcm2708_fb soc:fb: Couldn't allocate a DMA channel
kernel: bcm2708_fb soc:fb: probe failed, err -2
kernel: bcm2708_fb: probe of soc:fb failed with error -2
```
No problem to login through a ssh session.
Thanks in advance for your support.

### Steps to reproduce the behaviour

Example of how I compiled the kernel with the following steps for a rpi2 :
KERNEL=kernel7;make bcm2709_defconfig;make -j4 zImage modules dtbs;sudo make modules_install;sudo rm -f /boot/firmware/*.dtb;sudo cp arch/arm/boot/dts/*.dtb /boot/firmware/;sudo rm -f /boot/firmware/overlays/*.dtb*;sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/firmware/overlays/;sudo cp arch/arm/boot/dts/overlays/README /boot/firmware/overlays/
sudo mv /boot/firmware/kernel7.img /boot/firmware/kernel7-`uname -r`.img;sudo scripts/mkknlimg arch/arm/boot/zImage /boot/firmware/$KERNEL.img

### Device (s)

Raspberry Pi 2 Mod. B, Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B

### System

The OS is a Debian Bullseye 11.4 64bits. 

Aug  9 2022 13:46:06 
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

Linux 5.19.1-v7+ #1 SMP Tue Aug 16 10:38:07 CEST 2022 armv7l GNU/Linux

pelwell, 2022-08-17:
What's in your `config.txt`?

pelwell, 2022-08-17:
I'm guessing you have neither `dtoverlay=vc4-kms-v3d` nor `dtoverlay=vc4-fkms-v3d`. I can confirm the same failure with that configuration.

pelwell, 2022-08-17:
Ahh - the dangers of (not) maintaining a downstream driver. We were bitten by an upstream change removing support for the creation of IRQ resources, a reasonable thing to do since drivers are meant to use platform_get_irq.

See e7b8b60.

Zarla, 2022-08-17:
> Ahh - the dangers of (not) maintaining a downstream driver. We were bitten by an upstream change removing support for the creation of IRQ resources, a reasonable thing to do since drivers are meant [...]

Thanks pelwell for your comments. I will try your advice.

Zarla, 2022-08-19:
It's fixed for me after recompiling the kernel with the 7cfedc2 commit. Thanks Phil.

> kernel: bcm2708_fb soc:fb: FB found 1 display(s)
> kernel: Console: switching to colour frame buffer device 240x67
> kernel: bcm2708_fb soc:fb: Registered framebuffer for display 0, size 1920x1080
> kernel: bcm2835-rng 3f104000.rng: hwrng registered

closed 2022-08-19 by Zarla