raspberrypi/linux #5134: Bluetooth mouse lag after update to raspberrypi-kernel 1:1.20220811-1

numetaler, 2022-08-19:
### Describe the bug

Since the update to this kernel version via the update manager my Logitech MX Master 3 is very laggy/delayed. THe mouse is connected over Bluetooth. A different mouse wich is connected with an USB-Dongle works fine. I had to go back to the old version because i didn't find a fix. Editing the /boot/cmdline.txt didn't help.

I'm running Bullseye 64Bit with every update available.

### Steps to reproduce the behaviour

Install latest raspberrypi-kernel 1:1.20220811-1 package

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue:
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

vcgencmd version:
Aug  9 2022 13:44:40 
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

uname -a:
Linux raspihomeserver 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

pelwell, 2022-08-24:
> Editing the /boot/cmdline.txt didn't help.

1. What edits did you try?
2. Are you using firmware-kms (i.e. is `dtoverlay=vc4-fkms-v3d` in config.txt)?
3. Can you try `usbhid.mousepoll=20` (as a diagnostic)?

pelwell, 2022-08-24:
On second thoughts, the `usbhid.mousepoll` setting is unlikely to change anything since this is a pure Bluetooth mouse, but there's no harm in trying.

numetaler, 2022-08-24:
Hello,

1.What edits did you try?
I tried usbhid.mousepoll=20 and this didn't do anything

2.Are you using firmware-kms (i.e. is dtoverlay=vc4-fkms-v3d in config.txt)?
Yes this line is in my /boot/config.txt. But i think i deactivated this for testing an it didn't make a change. This is something what i doublecheck.

pelwell, 2022-08-24:
There's a new kernel build available that may improve things for you. You can install it using `sudo rpi-update`. You should be aware that this effectively a beta-release which has had little testing, so you might want to back up your data before updating. However, there have been few significant changes since the release you are running, and certainly not ones that I would suspect could cause data loss.

numetaler, 2022-08-26:
I deactivated dtoverlay=vc4-fkms-v3d an the mouse is working fine again. Thank you for your Help!

Ps.: I also did the sudo rpi-update but unfortunatly this broke everything completly and my Pi didn't boot at all. Not anything was showing up on my monitor, so i went back to my backup.

popcornmix, 2022-08-26:
> Nothing was showing up on my monitor at all, so i went back to my backup.

Can you describe your backup? i.e. output of "vcgencmd version", "uname -a" and whether it was an RPiOS bullseye image, or something else?

numetaler, 2022-08-26:
Hello,

My backup was exactly the same as  schon in the initial post: 

System

cat /etc/rpi-issue:
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

vcgencmd version:
Aug 9 2022 13:44:40
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

uname -a:
Linux raspihomeserver 5.15.32-v8+ https://github.com/raspberrypi/linux/issues/1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux.

From this state, i installed the package raspberrypi-kernel 1:1.20220811-1, deactivated dtoverlay=vc4-fkms-v3d and dtoverlay=rpivid-v4l2 (since it said on boot on the right upper corner it is no longer necessary). This state of the System works currently without any problem. 

But when i did sudo rpi-update at this point the pi wont't boot anymore

running System:

System:
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

vcgencmd version:
Aug  9 2022 13:44:40 
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

uname -a:
Linux raspihomeserver 5.15.56-v8+ #1575 SMP PREEMPT Fri Jul 22 20:31:26 BST 2022 aarch64 GNU/Linux

popcornmix, 2022-08-26:
Ah:
> Raspberry Pi 4 Mod. B 8GB booting from Intenso m.2 Sata III 128GB USB3.0 SSD via Geekworm X862 V2.0 Expansion Board

sounds like https://github.com/raspberrypi/firmware/issues/1744

numetaler, 2022-08-26:
Thank you for the information! I think i just wait for now since everything works fine at my current state