raspberrypi/linux #5136: x265 decoding under rpi-5.19.y does not work (but it does work under rpi-5.15.y)

graysky2, 2022-08-20:
### Describe the bug

Using @popcornmix [gbm_matrix](https://github.com/popcornmix/xbmc/tree/gbm_matrix) of kodi to playback x265 video booted into a kernel built from the rpi-5.19.y tree fails to playback.  Downgrading to a kernel build from the rpi-5.15.y branch works as expected.

### Steps to reproduce the behaviour

1. Boot into the rpi-5.19.y kernel
2. Start kodi
3. Try to playback the test file, [jellyfish-90-mbps-hd-hevc-10bit.mkv](https://www.larmoire.info/jellyfish/media/jellyfish-90-mbps-hd-hevc-10bit.mkv)

### Device (s)

Raspberry Pi 4 Mod. B

### System

* Arch ARM aarch64, kernel 5.19.1-1 build from [this](https://github.com/graysky2/PKGBUILDs/commits/rpi-5.19.y).
* `Aug  9 2022 13:44:40 
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)`
* `Linux workbench 5.19.1-1-rpi-ARCH #1 SMP PREEMPT Fri Aug 19 08:46:32 EDT 2022 aarch64 GNU/Linux`

### Logs

* Kodi [debug log](https://gist.github.com/graysky2/9dd2f834ab8a14355ad9cce932677441)

### Additional context

Interestingly, if I build kodi from Dom's [gbm](https://github.com/popcornmix/xbmc/tree/gbm) branch which is the Nexus code base, I am able to play back the same test video under the rpi-5.19.y kernel just fine.  The issue only affects the gbm_matrix build of kodi when it is paired with the newer kernel.

| Kernel branch | Kodi branch/version | x265 playback |
| :-: | --- | :-: | 
| rpi-5.15.y (latest commit) | gbm_matrix (latest commit)/matrix | yes |
| rpi-5.15.y (latest commit) | gbm (latest commit)/nexus | yes |
| rpi-5.19.y (latest commit) | gbm_matrix (latest commit)/matrix | no |
| rpi-5.19.y (latest commit) | gbm (latest commit)/nexus | yes |

For reference, `/boot/config.txt`
```
display_auto_detect=1

dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1
hdmi_enable_4kp60=1
force_turbo=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1
```

popcornmix, 2022-08-30:
Kodi (the matrix version from RPiOS) was what I used when testing bring up of 5.19 and it worked for me.
This uses system ffmpeg which has been updated fairly recently (so I believe it contains some logic for working with older and newer kernels).

Is the ffmpeg version the same in both cases, or does it change with kodi version?
When not working, does `/dev/video19` exist?

graysky2, 2022-08-30:
I use the tagged ffmpeg for matrix (4.3.2) and nexus (4.4.1) building them internally so they are different.  I just built matrix with the tagged version of ffmpeg for nexus (4.4.1) and it works as expected.

For the record:
```
% ls -lh /dev/video19
crw-rw----+ 1 root video 81, 19 Aug  9 19:29 /dev/video19
```

popcornmix, 2022-08-30:
Okay, so this is an ffmpeg issue rather than a linux issue?

switching to upstream hevc v4l2 support will break userland ABI.

@jc-kynesim has some hacks in recent ffmpeg trees to detect kernel version and try to support both,
but that may not have been done on the older 4.3.2 tree.

RPiOS uses his [test/4.3.4/rpi_main](https://github.com/jc-kynesim/rpi-ffmpeg/tree/test/4.3.4/rpi_main) I believe (and I think did work with 5.19/5.20).

graysky2, 2022-08-30:
> Okay, so this is an ffmpeg issue rather than a linux issue?

The experiment of building matrix with nexus' ffmpeg is consistent with that.  I am OK running matrix with the 4.4.1 tagged version of ffmpeg on 5.19 even though I try to honor the ffmpeg versions for the stable relases.  Thanks for the thoughts to troubleshoot this.

graysky2, 2022-08-30:
I am ok closing this since we determined the issue was not with the kernel.  Reopen if you disagree.

closed 2022-08-30 by graysky2