raspberrypi/linux #5138: BUG: Bad page map in process / BUG: KASAN: wild-memory-access; suspected DMA issue

kralo, 2022-08-22:
### Describe the bug

I have an application that does image analysis from an ov9281 sensor. Sometimes it would hang and produce traces like the following.

I have then tried to isolate and written a reproducer (below).
This happens on Kernels 5.10.110 and 6.0.0-rc1.

I suspect this has to do with the DMA/vc_sm_cma part of the camera image aquisition, because it does **not happen** with a "virtual" camera from the [vivid driver](https://www.kernel.org/doc/html/v4.9/media/v4l-drivers/vivid.html)

Very often the bad page map is around pmd:800000001801003 . PTE does always seem to be different.

The crashes with the reproducer happen around 1-3 times/hour, with the vivid driver it runs for 10 hours straight, when I terminated due to lack of patience.

```
Aug 21 17:25:41 kernel: BUG: Bad page map in process Xorg  pte:80706090a070705 pmd:800000001801003
Aug 21 17:25:41 kernel: addr:0000005599400000 vm_flags:200100073 anon_vma:ffffff804a2f20d8 mapping:0000000000000000 index:5599400
Aug 21 17:25:41 kernel: file:(null) fault:0x0 mmap:0x0 read_folio:0x0
Aug 21 17:25:41 kernel: CPU: 0 PID: 466 Comm: Xorg Tainted: G         C         6.0.0-rc1-v8-gc8f41281d1f4 #7
Aug 21 17:25:41 kernel: Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
Aug 21 17:25:41 kernel: Call trace:
Aug 21 17:25:41 kernel: dump_backtrace.part.0 (arch/arm64/kernel/stacktrace.c:184) 
Aug 21 17:25:41 kernel: show_stack (arch/arm64/kernel/stacktrace.c:191) 
Aug 21 17:25:41 kernel: dump_stack_lvl (lib/dump_stack.c:107 (discriminator 4)) 
Aug 21 17:25:41 kernel: dump_stack (lib/dump_stack.c:114) 
Aug 21 17:25:41 kernel: print_bad_pte (mm/memory.c:567 (discriminator 12)) 
Aug 21 17:25:41 kernel: vm_normal_page (mm/memory.c:638) 
Aug 21 17:25:41 kernel: do_wp_page (mm/memory.c:3384) 
Aug 21 17:25:41 kernel: __handle_mm_fault (mm/memory.c:4929 mm/memory.c:5053) 
Aug 21 17:25:41 kernel: handle_mm_fault (mm/memory.c:5151) 
Aug 21 17:25:41 kernel: do_page_fault (arch/arm64/mm/fault.c:502 arch/arm64/mm/fault.c:602) 
Aug 21 17:25:41 kernel: do_translation_fault (arch/arm64/mm/fault.c:689) 
Aug 21 17:25:41 kernel: do_mem_abort (arch/arm64/mm/fault.c:818) 
Aug 21 17:25:41 kernel: el0_da (arch/arm64/kernel/entry-common.c:133 arch/arm64/kernel/entry-common.c:142 arch/arm64/kernel/entry-common.c:503) 
Aug 21 17:25:41 kernel: el0t_64_sync_handler (arch/arm64/kernel/entry-common.c:646) 
Aug 21 17:25:41 kernel: el0t_64_sync (arch/arm64/kernel/entry.S:581) 
Aug 21 17:25:41 kernel: Disabling lock debugging due to kernel taint
Aug 21 17:25:41 kernel: ------------[ cut here ]------------
Aug 21 17:25:41 kernel: WARNING: CPU: 0 PID: 466 at mm/memory.c:2923 wp_page_copy (mm/memory.c:2923 mm/memory.c:3117) 
Aug 21 17:25:41 kernel: Modules linked in: rfkill bcm2835_unicam v4l2_dv_timings videobuf2_dma_contig rtc_pcf85063 videobuf2_memops ov9281 videobuf2_v4l2 regmap_i2c videobuf2_co>
Aug 21 17:25:41 kernel: CPU: 0 PID: 466 Comm: Xorg Tainted: G    B    C         6.0.0-rc1-v8-gc8f41281d1f4 #7
Aug 21 17:25:41 kernel: Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
Aug 21 17:25:41 kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 21 17:25:41 kernel: pc : wp_page_copy (mm/memory.c:2923 mm/memory.c:3117) 
Aug 21 17:25:41 kernel: lr : wp_page_copy (mm/memory.c:2917 mm/memory.c:3117) 
Aug 21 17:25:41 kernel: sp : ffffffc00be77a30
Aug 21 17:25:41 kernel: x29: ffffffc00be77a30 x28: ffffff805faec000 x27: ffffff804bc35e58
Aug 21 17:25:41 kernel: x26: ffffffc00be77cb0 x25: ffffff804a7ac680 x24: ffffffc00be77c88
Aug 21 17:25:41 kernel: x23: 1ffffff8017cef5e x22: ffffff804bc35e18 x21: 0000000000000000
Aug 21 17:25:41 kernel: x20: fffffffe017ebb00 x19: ffffffc00be77c70 x18: 0000000000000000
Aug 21 17:25:41 kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000005599400000
Aug 21 17:25:41 kernel: x14: 0000000000000000 x13: 746e696174206c65 x12: ffffffb00bf5da00
Aug 21 17:25:41 kernel: x11: 1ffffff00bf5d9ff x10: ffffffb00bf5d9ff x9 : dfffffc000000000
Aug 21 17:25:41 kernel: x8 : ffffff805faecfff x7 : 0000000000000000 x6 : ffffff805faec000
Aug 21 17:25:41 kernel: x5 : ffffff805faed000 x4 : 0000000000000000 x3 : 0000005599400000
Aug 21 17:25:41 kernel: x2 : 0000000000000f80 x1 : 0000005599400000 x0 : 0000000000001000
Aug 21 17:25:41 kernel: Call trace:
Aug 21 17:25:41 kernel: wp_page_copy (mm/memory.c:2923 mm/memory.c:3117) 
Aug 21 17:25:41 kernel: do_wp_page (mm/memory.c:3404) 
Aug 21 17:25:41 kernel: __handle_mm_fault (mm/memory.c:4929 mm/memory.c:5053) 
Aug 21 17:25:41 kernel: handle_mm_fault (mm/memory.c:5151) 
Aug 21 17:25:41 kernel: do_page_fault (arch/arm64/mm/fault.c:502 arch/arm64/mm/fault.c:602) 
Aug 21 17:25:41 kernel: do_translation_fault (arch/arm64/mm/fault.c:689) 
Aug 21 17:25:41 kernel: do_mem_abort (arch/arm64/mm/fault.c:818) 
Aug 21 17:25:41 kernel: el0_da (arch/arm64/kernel/entry-common.c:133 arch/arm64/kernel/entry-common.c:142 arch/arm64/kernel/entry-common.c:503) 
Aug 21 17:25:41 kernel: el0t_64_sync_handler (arch/arm64/kernel/entry-common.c:646) 
Aug 21 17:25:41 kernel: el0t_64_sync (arch/arm64/kernel/entry.S:581) 
Aug 21 17:25:41 kernel: ---[ end trace 0000000000000000 ]---
Aug 21 17:25:47 systemd[1]: systemd-fsckd.service: Succeeded.

```
[... 387 lines cut ...]
18:20:47 kernel: el0t_64_sync (arch/arm64/kernel/entry.S:581) 
18:20:47 kernel: Mem-Info:
18:20:47 kernel: active_anon:298 inactive_anon:251827 isolated_anon:0
18:20:47 kernel: Node 0 active_anon:1192kB inactive_anon:1007308kB active_file:181180kB inactive_file:143764kB unevictable:80kB isolated(anon):0kB isolated(file):128kB mapped:233440kB dirty:3788kB writeback:0kB shmem:31372kB writeback_tmp:0kB kernel_stack:14624kB pagetables:11512kB all_unreclaimable? no
18:20:47 kernel: DMA free:19708kB boost:4096kB min:12972kB low:15188kB high:17404kB reserved_highatomic:0KB active_anon:364kB inactive_anon:645092kB active_file:37920kB inactive_file:87732kB unevictable:0kB writepending:2540kB present:970752kB managed:876496kB mlocked:0kB bounce:0kB free_pcp:184kB local_pcp:0kB free_cma:2460kB
18:20:47 kernel: lowmem_reserve[]: 0 724 724 724
18:20:47 kernel: DMA32 free:32496kB boost:16884kB min:24388kB low:26264kB high:28140kB reserved_highatomic:4096KB active_anon:828kB inactive_anon:362216kB active_file:142688kB inactive_file:55984kB unevictable:80kB writepending:1248kB present:1048576kB managed:741396kB mlocked:80kB bounce:0kB free_pcp:248kB local_pcp:0kB free_cma:0kB
18:20:47 kernel: lowmem_reserve[]: 0 0 0 0
18:20:47 kernel: DMA: 199*4kB (UMEC) 177*8kB (UMEC) 178*16kB (UMEC) 241*32kB (UMEC) 86*64kB (UMEC) 7*128kB (UEC) 2*256kB (C) 1*512kB (C) 0*1024kB 0*2048kB 0*4096kB = 20196kB
18:20:47 kernel: DMA32: 245*4kB (UMEH) 149*8kB (UMH) 149*16kB (UMH) 665*32kB (UMEH) 113*64kB (UE) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 33068kB
18:20:47 kernel: 89083 total pagecache pages
18:20:47 kernel: 0 pages in swap cache
18:20:47 kernel: Free swap  = 0kB
18:20:47 kernel: Total swap = 0kB
18:20:47 kernel: 504832 pages RAM
18:20:47 kernel: 0 pages HighMem/MovableOnly
18:20:47 kernel: 100359 pages reserved
18:20:47 kernel: 81920 pages cma reserved
18:20:47 kernel: unicam fe801000.csi: dma alloc of size 921600 failed

```

### Additional context

+cc @naushir @pelwell

pelwell, 2022-08-22:
Interesting - could this be due to the same underlying issue as https://github.com/raspberrypi/linux/issues/5133?

olokos, 2022-08-22:
I've read this entire issue and the logs, and I couldn't find anything that relates to #5133 
The tainted modules are different, the stack traces are completely different, the kernel is also different (5.15 for me, 5.10 and 6 here), I also don't use CM4, just Pi + microSD + IMX477/HQ Camera.

I'm not entirely sure how the kernel structure is built for Pi, so don't take my words as something 100% correct, I could be wrong here, but this doesn't seem related to me, considering that my logs were twice almost exactly the same, while here the logs structure itself is vastly different.

I could be wrong though and those 2 issues could be related, I just couldn't find any common error messages with my issue.

pelwell, 2022-08-22:
The common factors are "camera" and "memory corruption".

olokos, 2022-08-22:
Right, this could be related, as I said, don't take my word for it, I just wanted to show my perspective, but those 2 issues could be directly related aswell, I have also updated my issue with additional data.

kralo, 2022-08-23:
I set up a third device, CM4, Kernel 5.19.1, reproducer does not take long.

Can I do something to produce more meaningful debug traces?

```
A 23 11:30:45 rpidmareproducer[911]: now calling system()
A 23 11:30:54 rpidmareproducer[911]: done...closing
A 23 11:30:54 : BUG: Bad page map in process rpidmareproduce pte:f5ffedfefbfefcf1 pmd:800000001c01003
A 23 11:30:54 : addr:0000007f9fa01000 vm_flags:00100073 anon_vma:ffffff8059578a98 mapping:0000000000000000 index:7f9fa01
A 23 11:30:54 : [file:(null)](file:///(null)) fault:0x0 mmap:0x0 read_folio:0x0
A 23 11:30:54 : CPU: 0 PID: 911 Comm: rpidmareproduce Tainted: G C        5.19.1-v8-ge7b8b60fb78b #10 8451487651094e40323d5adcf3324a1f1ff7f9cc
A 23 11:30:54 : Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
A 23 11:30:54 : Call trace:
A 23 11:30:54 : dump_backtrace.part.0 (arch/arm64//stacktrace.c:200)
A 23 11:30:54 : show_stack (arch/arm64//stacktrace.c:207)
A 23 11:30:54 : dump_stack_lvl (lib/dump_stack.c:107 (discriminator 4))
A 23 11:30:54 : dump_stack (lib/dump_stack.c:114)
A 23 11:30:54 : print_bad_pte (mm/memory.c:567 (discriminator 12))
A 23 11:30:54 : vm_normal_page (mm/memory.c:630)
A 23 11:30:54 : unmap_page_range (mm/memory.c:1425 mm/memory.c:1567 mm/memory.c:1596 mm/memory.c:1617 mm/memory.c:1638)
A 23 11:30:54 : unmap_vmas (mm/memory.c:1722)
A 23 11:30:54 : exit_mmap (mm/mmap.c:3163)
A 23 11:30:54 : __mmput (/fork.c:1250 (discriminator 2) /fork.c:1189 (discriminator 2))
A 23 11:30:54 : mmput (/fork.c:1209)
A 23 11:30:54 : do_exit (./arch/arm64/include/asm/current.h:19 /exit.c:511 /exit.c:782)
A 23 11:30:54 : do_group_exit (/exit.c:906)
A 23 11:30:54 : __arm64_sys_exit_group (/exit.c:936)
A 23 11:30:54 : invoke_syscall (arch/arm64//syscall.c:38 arch/arm64//syscall.c:52)
A 23 11:30:54 : el0_svc_common.constprop.0 (./arch/arm64/include/asm/daifflags.h:28 arch/arm64//syscall.c:150)
A 23 11:30:54 : do_el0_svc (arch/arm64//syscall.c:207)
A 23 11:30:54 : el0_svc (arch/arm64//entry-common.c:133 arch/arm64//entry-common.c:142 arch/arm64//entry-common.c:625)
A 23 11:30:54 : el0t_64_sync_handler (arch/arm64//entry-common.c:643)
A 23 11:30:54 : el0t_64_sync (arch/arm64//entry.S:581)
A 23 11:30:54 : Disabling lock debugging due to  taint
A 23 11:30:54 : BUG: Bad page map in process rpidmareproduce pte:61768fb1e0feffff pmd:800000001c01003
A 23 11:30:54 : addr:0000007f9fa14000 vm_flags:00100073 anon_vma:ffffff8059578a98 mapping:0000000000000000 index:7f9fa14
A 23 11:30:54 : [file:(null)](file:///(null)) fault:0x0 mmap:0x0 read_folio:0x0
A 23 11:30:54 : CPU: 0 PID: 911 Comm: rpidmareproduce Tainted: G    B C        5.19.1-v8-ge7b8b60fb78b #10 8451487651094e40323d5adcf3324a1f1ff7f9cc
A 23 11:30:54 : Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
A 23 11:30:54 : Call trace:
A 23 11:30:54 : dump_backtrace.part.0 (arch/arm64//stacktrace.c:200)
A 23 11:30:54 : show_stack (arch/arm64//stacktrace.c:207)
A 23 11:30:54 : dump_stack_lvl (lib/dump_stack.c:107 (discriminator 4))
A 23 11:30:54 : dump_stack (lib/dump_stack.c:114)
A 23 11:30:54 : print_bad_pte (mm/memory.c:567 (discriminator 12))
A 23 11:30:54 : vm_normal_page (mm/memory.c:630)
A 23 11:30:54 : unmap_page_range (mm/memory.c:1425 mm/memory.c:1567 mm/memory.c:1596 mm/memory.c:1617 mm/memory.c:1638)
A 23 11:30:54 : unmap_vmas (mm/memory.c:1722)
A 23 11:30:54 : exit_mmap (mm/mmap.c:3163)
A 23 11:30:54 : __mmput (/fork.c:1250 (discriminator 2) /fork.c:1189 (discriminator 2))
A 23 11:30:54 : mmput (/fork.c:1209)
A 23 11:30:54 : do_exit (./arch/arm64/include/asm/current.h:19 /exit.c:511 /exit.c:782)
A 23 11:30:54 : do_group_exit (/exit.c:906)
A 23 11:30:54 : __arm64_sys_exit_group (/exit.c:936)
A 23 11:30:54 : invoke_syscall (arch/arm64//syscall.c:38 arch/arm64//syscall.c:52)
A 23 11:30:54 : el0_svc_common.constprop.0 (./arch/arm64/include/asm/daifflags.h:28 arch/arm64//syscall.c:150)
A 23 11:30:54 : do_el0_svc (arch/arm64//syscall.c:207)
A 23 11:30:54 : el0_svc (arch/arm64//entry-common.c:133 arch/arm64//entry-common.c:142 arch/arm64//entry-common.c:625)
A 23 11:30:54 : el0t_64_sync_handler (arch/arm64//entry-common.c:643)
A 23 11:30:54 : el0t_64_sync (arch/arm64//entry.S:581)
```

olokos, 2022-08-23:
I think `/var/log/messages.log` is a very good source of logs.
It contains dmesg logs from multiple reboots, with archiving also, so you probably have a lot of those aswell.
Try to look into that file and find line that contains `tainted`, usually there will be a stack trace nearby.
I suggest you upload a cropped file that contains ~100 logs before and after the stack trace and the full file, then we'll know for sure if Your bug is related to my #5133  - speaking of which, just crashed again.

kralo, 2022-08-23:
No, olokos. Spamming these issues with megabytes of logs won't get us anywhere. I am able to find stacktraces, and I can annotate the line numbers because I compile the kernel myself.

My issue has a working reproducer. What I was asking was if there was a hint to produce _better_ logs by enabling more kernel debug options, i.e. I tried the `cma_alloc` ones, but they do not add any insight.

I will wait for someone from the kernel team to hear if they can reproduce.

pelwell, 2022-08-23:
We're still in holiday season, but this issue will get some attention in the not too distant future.

olokos, 2022-08-24:
@kralo do keep in mind I have specifically NOT copy pasted my logs into my issue, exactly for the reason of not making it spammy. On top of that I provided both cropped version of dmesg and full one, so the issue is quickly found, but so it also contains the rest of events that happened in the system before and after the crash.

I've never said you are doing something wrong or you can't do something right, I didn't even think about that for a second. I wasn't questioning your skills nor was I trying to display my knowledge. That's not the point whatsoever.

Pelwell mentioned that your issue might be related to mine. That is the **only** reason why I'm commenting here, to resolve the bug.

I wanted to do apples-to-apples comparison between Your issue and my issue, especially since we do seem to run the same Pi and similar camera stack, but I can't do it because:
- Your logs differ in structure from a standard DMESG output (not sure why, perhaps it's not DMESG?)
- It's pretty hard to tell what is the order of the logs you provided (chronology info is lost )
- There's little to no information what happened before and after your stack traces

IMO it's better to provide a little bit too much (if it doesn't ruin readablity) then too little.

I can be wrong and might also have misunderstood something, so bear with me please, but honestly all I want to achieve is to have a stable running HQ Camera/IMX477 stream. 
Considering Your and my issue could be the result of the same underlying issue, why won't we cooperate instead and try to figure out as much as possible around the problem?

All I wanted from you is to post a bit more of logs, so I could have a read and see if it looks anything like my problem, nothing else, really!

Especially since you literally asked:
`
Can I do something to produce more meaningful debug traces?
`
Providing logs before and after stack trace, might sometimes bring more value than the stack trace itself, which you chose to ignore completely, even when asked for specifically.

Why fight instead of working together and bringing valuable info for kernel team to work with or finding the root cause of the problem ourselves?...

kralo, 2022-08-25:
I compiled the kernel with `CONFIG_CMA_DEBUG=y` and `CONFIG_CMA_DEBUGFS=y`. This led me to the bcm2835-unicam.c and 
I enabled more verbose debug output with 
```
$ cat /etc/modprobe.d/unicam.conf
# 3 is very verbose, 2 medium
options bcm2835_unicam debug=2
```

As my Application (and my reproducer, willfully) do, they "stall" the buffer pipeline (by `sleep()`'ing, because all the buffers get filled) and I get these messages
`[  104.206562] unicam fe801000.csi: ISR: [0] Dropping frame, buffer not available at FS`

In bcm2835-unicam.c, there is [this](https://github.com/raspberrypi/linux/blob/e6f0bf2a5706e2007ef91df6b40bfde1dfbd0029/drivers/media/platform/bcm2835/bcm2835-unicam.c#L133)
```
/*
 * Size of the dummy buffer. Can be any size really, but the DMA
 * allocation works in units of page sizes.
 */
#define DUMMY_BUF_SIZE		(PAGE_SIZE)
```

This buffer, which on my RPi4CM is 4K, is where the CSI-Frame-Handler's write address gets directed to, when there is no (vb2-)buf ready.

**But will the `UNICAM_IBEA0` really respect the Endaddr and never write past it's boundary?**
```
dma_addr_t endaddr = dmaaddr + buffer_size;

	if (pad_id == IMAGE_PAD) {
		reg_write(dev, UNICAM_IBSA0, dmaaddr);
		reg_write(dev, UNICAM_IBEA0, endaddr);
```

**I could not find clear documentation about this. Someone with access to more verbose videocore iv specification should check this.
In the meantime my working theory is that the image sensor data got written to a buffer that was to small and spilled over.**

As a quick test, I have increased the size of the dummy buffer, by `#define DUMMY_BUF_SIZE		(PAGE_SIZE*230)` which is sufficient for my image size. My reproducer ran for 6 hours without problems, I stopped it to lack of patience but will run again this night.

pelwell, 2022-08-26:
That's a useful finding. It would be even better if you could initialise the start of the second page of the dummy buffer with a fixed pattern, and check periodically that it hasn't changed - something like this:
```diff
--- a/drivers/media/platform/bcm2835/bcm2835-unicam.c
+++ b/drivers/media/platform/bcm2835/bcm2835-unicam.c
@@ -131,6 +131,11 @@ MODULE_PARM_DESC(media_controller, "Use media controller API");
  * allocation works in units of page sizes.
  */
 #define DUMMY_BUF_SIZE         (PAGE_SIZE)
+#define PADDED_DUMMY_BUF_SIZE  (PAGE_SIZE * 230)
+
+static const uint32_t guard_pattern[] = {
+       0x5a5a1234, 0xffffffff, 0xffffffff, 0x5678a5a5
+};
 
 enum pad_types {
        IMAGE_PAD,
@@ -852,6 +857,12 @@ static void unicam_schedule_dummy_buffer(struct unicam_node *node)
 static void unicam_process_buffer_complete(struct unicam_node *node,
                                           unsigned int sequence)
 {
+       if (!memcmp(node->dummy_buf_cpu_addr, guard_pattern, sizeof(guard_pattern)))
+       {
+               pr_err("%s: guard pattern corrupted:\n%*phN\n", __func__,
+                      sizeof(guard_pattern), node->dummy_buf_cpu_addr);
+               memcpy(node->dummy_buf_cpu_addr, guard_pattern, sizeof(guard_pattern));
+       }
        node->cur_frm->vb.field = node->m_fmt.field;
        node->cur_frm->vb.sequence = sequence;
 
@@ -3004,13 +3015,16 @@ static int register_node(struct unicam_device *unicam, struct unicam_node *node,
        media_entity_pads_init(&vdev->entity, 1, &node->pad);
 
        node->dummy_buf_cpu_addr = dma_alloc_coherent(&unicam->pdev->dev,
-                                                     DUMMY_BUF_SIZE,
+                                                     PADDED_DUMMY_BUF_SIZE,
                                                      &node->dummy_buf_dma_addr,
                                                      GFP_KERNEL);
        if (!node->dummy_buf_cpu_addr) {
                unicam_err(unicam, "Unable to allocate dummy buffer.\n");
                return -ENOMEM;
        }
+
+       memcpy(node->dummy_buf_cpu_addr, guard_pattern, sizeof(guard_pattern));
+
        if (!unicam->mc_api) {
                if (pad_id == METADATA_PAD ||
                    !v4l2_subdev_has_op(unicam->sensor, video, s_std)) {
@@ -3104,7 +3118,7 @@ static void unregister_nodes(struct unicam_device *unicam)
                struct unicam_node *node = &unicam->node[i];
 
                if (node->dummy_buf_cpu_addr) {
-                       dma_free_coherent(&unicam->pdev->dev, DUMMY_BUF_SIZE,
+                       dma_free_coherent(&unicam->pdev->dev, PADDED_DUMMY_BUF_SIZE,
                                          node->dummy_buf_cpu_addr,
                                          node->dummy_buf_dma_addr);
                }
```

pelwell, 2022-08-26:
[ Patch updated - it was missing the `DUMMY_BUF_SIZE` offsets for the guard pattern. ]

6by9, 2022-08-26:
IIRC The buffer end address in Unicam was checked at the end of a line.
With IMX477 in full res mode (4056x3040 12bpp) one line would be longer than a single page. I'm initially surprised it's an issue with OV9281.

@naushir is out of the office until Tuesday, but we'll discuss it then.
In the firmware the end address would have been set to a multiple of the stride, therefore a comparison at the end of the line of the start address for the next line would be wrap correctly. With the kernel buffer being a page size, the start of the line would be within the buffer, but the end of the line wouldn't be.
Allocating a dummy buffer of size ```DUMMY_BUF_SIZE * 2```, but passing ```DUMMY_BUF_SIZE``` to ```unicam_wr_dma_addr``` may be a solution.

It should be noted that the dummy buffer will only ever be written to in the event of a dropped frame due to buffers not being cycled correctly / fast enough.

naushir, 2022-08-30:
Some of these call stacks look suspiciously similar to those in https://github.com/raspberrypi/libcamera-apps/issues/246.  For that issue there was a suspicion it was power related, and using ``over_voltage=xx`` helped resolve the crash.

> But will the `UNICAM_IBEA0` really respect the Endaddr and never write past it's boundary?

This should be the expected behaviour of the hardware, according to the spec.  Clearly this may not be true based on your observations, so there may be some hardware bug that is in play.  Note that the dummy buffer will be used when there are frame drops occurring.  This is not entirely uncommon, so the overwrite does not seem to be happening on all occasions or we would see this much more often.  Perhaps there's some other interaction when the system/memory bus is heavily loaded?

> In the firmware the end address would have been set to a multiple of the stride, therefore a comparison at the end of the line of the start address for the next line would be wrap correctly. With [...]

I would have hoped that the Unicam HW checks the buffer overwrite possibility on every AXI burst operation rather than just at the line boundary, but there is no way to confirm this.  If that were the case, the image width would not matter for sizing the dummy buffer.

> Allocating a dummy buffer of size `DUMMY_BUF_SIZE * 2`, but passing `DUMMY_BUF_SIZE` to `unicam_wr_dma_addr` may be a solution.

Assuming the HW has a bug, this feels more like the right solution here - maybe even size the buffer ``* 4`` to be sure.

naushir, 2022-08-30:
Predictably, I cannot reproduce the crash (in the 4-5 hours it has been running so far) with the rpidmareproducer script! However, I am running a very different configuration base:

- 32-bit Raspberry Pi OS image.
- imx477 camera.
- Running on a Raspberry Pi 4B board, not a CM4.

I also have ``stress`` running in the background to make things busier just to see if that can provoke the crash.  I'll switch to a 64-bit OS tomorrow if I still have no luck with my current image.

naushir, 2022-08-31:
Right, so I definitely see the overruns in the dummy buffer occurring when adding/testing guard-words on the end of the buffer. I have not reproduced the overrun with libcamera, only with the provided ``rpidmareproducer.c`` test application.

The good news is that doubling the size of the dummy buffer and providing Unicam with half that size seems to "fix" the overrun - i.e. overruns still occur, but we do not trample outside of our allocation.

@kralo would you be able to try out this change from https://github.com/raspberrypi/linux/pull/5157 and let me know if you still get crashes with your setup?

kralo, 2022-09-01:
I can reproduce the errors when checking with the guard patterns, this was an excellent idea.

However, to anyone trying this I recommend checking for (memcpy () != 0)

```
diff --git a/drivers/media/platform/bcm2835/bcm2835-unicam.c b/drivers/media/platform/bcm2835/bcm2835-unicam.c
index 979c4bf37..dd265e2e5 100644
--- a/drivers/media/platform/bcm2835/bcm2835-unicam.c
+++ b/drivers/media/platform/bcm2835/bcm2835-unicam.c
@@ -855,10 +855,18 @@ static void unicam_schedule_dummy_buffer(struct unicam_node *node)
 static void unicam_process_buffer_complete(struct unicam_node *node,
                                           unsigned int sequence)
 {
-       if (!memcmp(node->dummy_buf_cpu_addr + DUMMY_BUF_SIZE, guard_pattern, sizeof(guard_pattern)))
+       if (memcmp(node->dummy_buf_cpu_addr + DUMMY_BUF_SIZE, guard_pattern, sizeof(guard_pattern)) != 0)
        {
                pr_err("%s: guard pattern corrupted:\n%*phN\n", __func__,
                       sizeof(guard_pattern), node->dummy_buf_cpu_addr + DUMMY_BUF_SIZE);
+
+                         pr_err("%s: guard pattern is :\n%*phN\n", __func__,
+                      sizeof(guard_pattern), guard_pattern);
+
+                       pr_err(" guard memcmp result %d",
+                      memcmp(node->dummy_buf_cpu_addr + DUMMY_BUF_SIZE, guard_pattern, sizeof(guard_pattern)));
+
+
                memcpy(node->dummy_buf_cpu_addr + DUMMY_BUF_SIZE, guard_pattern, sizeof(guard_pattern));

        }

```

This yields 
```
[  181.781915] unicam fe801000.csi: ISR: [0] Dropping frame, buffer not available at FS
[  181.793360] unicam_process_buffer_complete: guard pattern corrupted:
               ff00ff00ff00ff00ff00ff00ff00ff00
[  181.793435] unicam_process_buffer_complete: guard pattern is :
               34125a5affffffffffffffffa5a57856
[  181.793452]  guard memcmp result 203

```

which clearly is camera pixel data, as my ov9281 ist set to generate the testpattern, i.e. `v4l2-ctl -d /dev/v4l-subdev0 --set-ctrl test_pattern=1`

So I will now run the test for a couple hours and report in the pull req.

pelwell, 2022-09-01:
Thanks for the memcmp correction (the code wasn't even compiled) - I've updated the code fragment to avoid confusion.

closed 2022-11-02 by kralo