raspberrypi/linux #5142: v4l2jpegdec not outputting data for some cameras

engnfrc, 2022-08-23:
### Describe the bug

Using an ELP-USBFHD01M w/ OV2710 camera, v4l2jpegdec isn't sending anything downstream, terminal output stops at "Redistribute latency...". Using an "identity silent=false" before this element I see lots of output, but if I move it to after this element there's silence. If I switch from v4l2jpegdec to jpegdec the pipeline works as expected. Also, using a different camera (ELP-USBFHD06H w/ SONY IMX322/IMX323) the v4l2jpegdec element works as expected. I've tried numerous of the ELP-USBFHD01M cameras all with the same result.

Example pipeline for ELP-USBFHD01M (OVP) where v4l2jpegdec isn't working (identity_pre present at the framerate, identity_post is missing):
`GST_DEBUG=*:4 GST_DEBUG_DUMP_DOT_DIR=$PWD/pipelines gst-launch-1.0 -e -vvv v4l2src device=/dev/videoHEL1 ! image/jpeg,width=1280,height=720,framerate=60/1 ! jpegparse ! identity name=identity_pre silent=false ! v4l2jpegdec ! identity name=identity_post silent=false ! filesink location=ELP-USBFHD01M.raw`

Example pipeline for ELP-USBFHD01M (OVP) where jpegdec is working:
`GST_DEBUG=*:4 GST_DEBUG_DUMP_DOT_DIR=$PWD/pipelines gst-launch-1.0 -e -vvv v4l2src device=/dev/videoHEL1 ! image/jpeg,width=1280,height=720,framerate=60/1 ! jpegparse ! identity name=identity_pre silent=false ! jpegdec ! identity name=identity_post silent=false ! filesink location=ELP-USBFHD01M.raw`

Example pipeline for ELP-USBFHD06H (SONY) where v4l2jpegdec is working:
`GST_DEBUG=*:4 GST_DEBUG_DUMP_DOT_DIR=$PWD/pipelines gst-launch-1.0 -e -vvv v4l2src device=/dev/videoHEL1-1 ! image/jpeg,width=1280,height=720,framerate=30/1 ! jpegparse ! identity name=identity_pre silent=false ! v4l2jpegdec ! identity name=identity_post silent=false ! filesink location=ELP-USBFHD06H.raw`

Here are two attachments, the first grabs a single frame from the camera and attempts to convert it to raw using v4l2jpegdec and the second grabs a single frame from the camera and saves it straight to a file.
[Pipeline - Raw.zip](https://github.com/raspberrypi/linux/files/9409942/Pipeline.-.Raw.zip)
[Pipeline - JPEG.zip](https://github.com/raspberrypi/linux/files/9409943/Pipeline.-.JPEG.zip)

Discussion of this issue started on the RPi Forums (https://forums.raspberrypi.com/viewtopic.php?p=2031719) where **@6by9** has been helping and I've also shared it on the Gstreamer Devel Listserv [gstreamer-devel@lists.freedesktop.org](mailto:gstreamer-devel@lists.freedesktop.org) where **Nicolas Dufresne (ndufresne)** has been assisting.

### Steps to reproduce the behaviour

Attach an ELP-USBFHD01M camera to a USB port on an RPi 4B and run an adapted pipeline based on the following original pipeline:
`GST_DEBUG=*:4 GST_DEBUG_DUMP_DOT_DIR=$PWD/pipelines gst-launch-1.0 -e -vvv v4l2src device=/dev/videoHEL1 ! image/jpeg,width=1280,height=720,framerate=60/1 ! jpegparse ! identity name=identity_pre silent=false ! v4l2jpegdec ! identity name=identity_post silent=false ! filesink location=ELP-USBFHD01M.raw`

### Device (s)

Raspberry Pi 4 Mod. B

### System

System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

Linux hel-microdas 5.15.33-v8 #1 SMP PREEMPT Tue Apr 19 16:29:44 UTC 2022 aarch64 GNU/Linux
Revision        : c03114
Serial          : 100000007578f78c
Model           : Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 24 2022 13:19:26
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       10255636 5629140   4085824  58% /
devtmpfs         1590472       0   1590472   0% /dev
tmpfs            1723432       0   1723432   0% /dev/shm
tmpfs             689376   17740    671636   3% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/sda1         261868   31620    230248  13% /boot
tmpfs             344684      20    344664   1% /run/user/1000
[... 746 lines cut ...]
[20603.115670] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.115733]  MAI: soc_pcm_open() failed (-19)
[20603.116420] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.116494]  MAI: soc_pcm_open() failed (-19)
[20603.117090] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.117132]  MAI: soc_pcm_open() failed (-19)
[20603.123142] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.123197]  MAI: soc_pcm_open() failed (-19)
[20603.123619] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.123659]  MAI: soc_pcm_open() failed (-19)
[20603.123906] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.123943]  MAI: soc_pcm_open() failed (-19)
[20603.177004] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.177057]  MAI: soc_pcm_open() failed (-19)
[20603.177836] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.177890]  MAI: soc_pcm_open() failed (-19)
[20603.178754] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.178805]  MAI: soc_pcm_open() failed (-19)
[20603.185563] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.185621]  MAI: soc_pcm_open() failed (-19)
[20603.185913] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.185952]  MAI: soc_pcm_open() failed (-19)
[20603.187940] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[20603.187993]  MAI: soc_pcm_open() failed (-19)
[347912.815547] uvcvideo 1-1.3:1.1: Failed to resubmit video URB (-1).

engnfrc, 2022-08-23:
@6by9 posted this reply on the RPi forums just now, seems like he might be onto something...

> [quote=6by9 post_id=2032142 time=1661278452 user_id=92334]
> I've had a quick look at the data in camera1.mjpg. The JPEG data isn't fully valid.
> 
> I tend to use [url=https://github.com/ImpulseAdventure/JPEGsnoop]JPEGSnoop[/url] to look at JPEG bitstreams, and it complains of bad huffman codes and overreading scan segments.
> VLC also complains of No JPEG data found in image.
> 
> With camera2.mjpg, both JPEGSnoop and VLC are happy.
> 
> MJPEG is a very badly defined format in that there is no official specification for it. Some encoders drop the Huffman tables totally and rely on defaults, whilst MJPEG-B mangles the stream such that it isn't compatible with the JPEG JFIF standard.
> It'll take a moderate amount of analysis to work out what camera1 has done to the stream.
> [/quote]

ndufresne, 2022-08-24:
@6by9 note that instead of just skipping the buffer, the driver should use `vb2_buffer_done(vb, VB2_BUF_STATE_ERROR);` if it can't handle the frame due to these errors. That being said, this is not always easy to achieve if you have a ring buffer driven decoder.

6by9, 2022-08-24:
I'm aware that it should be returning ```VB2_BUF_STATE_ERROR```, but the JPEG codec code is pretty icky - logging and asserts appear to have offended the original author, so it contains neither.

The codec hardware is throwing an error, but I don't understand the cause at present.

I think the decode error flag is being discarded somewhere in the pipe though. Fixing that should be relatively simple, but of course doesn't solve the problem of it not decoding the data.

Examing the JPEG data in more detail, camera1.mjpg (from the forums) and Pipeline.-.JPEG/ELP-USBFHD01M_1frame.jpg are both YUV4:2:2 images. camera2.mjpg is YUV4:2:0. The hardware should cope with either, but it is a significant difference.

6by9, 2022-08-24:
Firmware fixed to return the decode error flag. It should be in the next rpi-update.

I'm still trying to work out why it doesn't like the JPEG block doesn't like the stream. I'm fairly sure that it's done YUV422 decode to YUV420.
The stream does have restart markers at the end of every line, but again that is standard and should be handled (the encode can be asked to insert restart markers, so easy to test).

popcornmix, 2022-08-26:
rpi-update contains potential fix for this issue.

engnfrc, 2022-08-27:
@popcornmix can you elaborate on that? I ran through ‘apt update’, ‘apt upgrade’, ‘rpi-update’, and ‘rpi-eeprom-update’ but the pipeline is still not outputting anything from v4l2jpegdec.

6by9, 2022-08-30:
The fix (only in rpi-update, not apt) is to report the error in decode, not to be able to decode the stream.
There's still something in that JPEG stream that the hardware doesn't like.

CarlFK, 2023-10-02:
I think I am bumping into the same thing.  my real source is a v4l2 video device (hdmi2usb) but I've come up with some tests that use a generated stream:  videotestsrc ! jpegenc ! jpegparse ! v4l2jpegdec
here you can see my test, and the script that generates it:

https://gist.github.com/CarlFK/53c82b739f7953a504699dcef116768d

sometimes it runs all 9000 frames, 
sometimes it stops and I have to hit ^c
`^Chandling interrupt.`

```
pi@pi4opsis1:~ $ ./a2.sh 
++ echo i=0 s=00000
i=0 s=00000
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw, '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:50.831124797
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=1 s=00001
i=1 s=00001
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,colorimetry=bt709 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:11.275387442
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=2 s=00010
i=2 s=00010
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,pixel-aspect-ratio=1/1 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:11.430691314
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=3 s=00011
i=3 s=00011
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,pixel-aspect-ratio=1/1,colorimetry=bt709 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:25.804701338
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=4 s=00100
i=4 s=00100
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,framerate=30/1 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:11.941029860
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=5 s=00101
i=5 s=00101
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,framerate=30/1,colorimetry=bt709 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:02.585810582
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=6 s=00110
i=6 s=00110
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,framerate=30/1,pixel-aspect-ratio=1/1 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:11.277416064
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=7 s=00111
i=7 s=00111
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,framerate=30/1,pixel-aspect-ratio=1/1,colorimetry=bt709 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:11.565386590
Setting pipeline to NULL ...
Freeing pipeline ...
++ echo i=8 s=01000
i=8 s=01000
++ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' videoconvert '!' video/x-raw,height=720 '!' fakesink num-buffers=9000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3132): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
pi@pi4opsis1:~ $ 
```

CarlFK, 2023-10-02:
The above tests might be over complicated.  It seems a simple case fails if I let it run long enough: 
```
$ gst-launch-1.0 videotestsrc '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' fakesink 
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:01:49.030130857
```

however: videotestsrc generates some "noise" - if I give it a static red frame, it seems to run forever:

```
pi@pi4opsis1:~ $ gst-launch-1.0 videotestsrc pattern=red '!' jpegenc '!' jpegparse '!' v4l2jpegdec '!' fakesink 
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
23:11:53. / 99:99:99.
```

I suspect a clever use of tee and filesink location=red.jpeg can capture the frame that is causing problems.  I gave it a shot, but my gst skills aren't up to it.

yurifedoseev, 2025-02-28:
I bumped to a similar issue when `v4l2jpegdec` freezes or stats lagging. It prevents to reduce RPI CPU load because the pipeline has to use `jpegdec` for proper work