raspberrypi/linux #5150: ffmpeg h264_vl42m2m encoded videos do not play back

JooJooBee666, 2022-08-29:
### Describe the bug

When trying to encode using h264_vl42m2m within ffmpeg, the resulting videos are useless.  However, by simply changing to libx264, and nothing else, videos are fine.  For example:

ffmpeg -i sample_1920x1080.mp4 -c:v h264_v4l2m2m -b:v 8M -c:a copy sample_1920x1080_test.mp4 - **VIDEO NO GOOD**
ffmpeg -i sample_1920x1080.mp4 -c:v libx264 -b:v 8M -c:a copy sample_1920x1080_test.mp4 - **VIDEO WORKS**

Here's the output from the h264_v4l2m2 encoder:
```
ffmpeg -i sample_1920x1080.mp4 -c:v h264_v4l2m2m -b:v 8M -c:a copy sample_1920x1080_test.mp4
ffmpeg version 4.3.4-0+deb11u1+rpt3 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'sample_1920x1080.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf57.83.100
  Duration: 00:00:28.24, start: 0.000000, bitrate: 10836 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1920x1080, 10833 kb/s, 23.98 fps, 23.98 tbr, 24k tbn, 47.95 tbc (default)
    Metadata:
      handler_name    : VideoHandler
File 'sample_1920x1080_test.mp4' already exists. Overwrite? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (h264_v4l2m2m))
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0x559c591f60]  <<< v4l2_encode_init: fmt=0/-1
[h264_v4l2m2m @ 0x559c591f60] Using device /dev/video11
[h264_v4l2m2m @ 0x559c591f60] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x559c591f60] requesting formats: output=YU12 capture=H264
[h264_v4l2m2m @ 0x559c591f60] Failed to set number of B-frames: Invalid argument
[h264_v4l2m2m @ 0x559c591f60] Failed to set gop size: Invalid argument
Output #0, mp4, to 'sample_1920x1080_test.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
    Stream #0:0(und): Video: h264 (h264_v4l2m2m) (avc1 / 0x31637661), yuv420p, 1920x1080, q=-1--1, 8000 kb/s, 23.98 fps, 24k tbn, 23.98 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      encoder         : Lavc58.91.100 h264_v4l2m2m
frame=  677 fps= 21 q=-0.0 Lsize=   22094kB time=00:00:28.19 bitrate=6419.5kbits/s speed=0.871x    
video:22091kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.016396%
```

Here's the output from the libx264 encoder:
```
ffmpeg -i sample_1920x1080.mp4 -c:v libx264 -b:v 8M -c:a copy sample_1920x1080_test.mp4 
ffmpeg version 4.3.4-0+deb11u1+rpt3 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
[... 53 lines cut ...]

Thanks!

### Steps to reproduce the behaviour

try:
`ffmpeg -i sample_1920x1080.mp4 -c:v h264_v4l2m2m -b:v 8M -c:a copy sample_1920x1080_test.mp4`
...and watch as it consumes power and outputs garbage.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi 4B, 8GB model

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

Aug 23 2022 14:23:34 
Copyright (c) 2012 Broadcom
version bf881c824bac214626a9f78094d614f7db462c34 (clean) (release) (start)

Linux pi5 5.15.61-v8+ #1578 SMP PREEMPT Wed Aug 24 11:51:18 BST 2022 aarch64 GNU/Linux

popcornmix, 2022-09-10:
@jc-kynesim can you reproduce this?

jc-kynesim, 2022-09-12:
I now understand what is going on - the mp4 muxer appears to require the SPS/PPS extradata at init time if it is going to take it from the context, but the h/w encoder won't provide it before it has been given its first frame (so after that). However it can also be passed along attached to the encoded packet as sidedata so I'll fix the encoder to do that and fix the mp4 muxer to respect it.

6by9, 2022-09-13:
> so I'll fix the encoder to do that and fix the mp4 muxer to respect it

Encoder being the FFmpeg encode side, or does it need a kernel change?

jc-kynesim, 2022-09-13:
Just ffmpeg - it all works fine if you have a raw h264 stream as the output type from ffmpeg - it is just separating off SPS/PPS for the mp4 muxer that has issues and that is all internal to ffmpeg (I do make the nasty assumption that SPS/PPS turns up on its own in the 1st capture frame which isn't a part of the V4L2 spec but I recall you saying was always true for our encoder).

6by9, 2022-09-13:
There is a V4L2 control V4L2_CID_MPEG_VIDEO_HEADER_MODE. FFmpeg chooses V4L2_MPEG_VIDEO_HEADER_MODE_SEPARATE to ask for the the header bytes to be separate from the 1st frame. The default is that they are joined with the first frame.
https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/ext-ctrls-codec.html

jc-kynesim, 2022-09-13:
Ah! Missed that. Excellent.

jc-kynesim, 2022-09-15:
Fix now pushed to my ffmpeg repo (https://github.com/jc-kynesim/rpi-ffmpeg.git branch  test/4.3.4/rpi_main if you want to build it yourself). It'll be in the next ffmpeg update. Also fixes the same issue in mkv generation.

JooJooBee666, 2022-09-15:
@jc-kynesim What are the build parameters used to build from source?

jc-kynesim, 2022-09-15:
You should find build/install instructions in pi-util/BUILD.txt - if they don't work or are unclear please tell me

JooJooBee666, 2022-09-15:
Ahh, didn't know that was there.  I'll try that and let you know.

JooJooBee666, 2022-09-15:
No luck.  Ran make celan and manually cleaned up the out folder but it's not working.  Not sure what to do about this error:
```
pi@pi5:~/source/rpi-ffmpeg $ pi-util/conf_native.sh           
Configure for native build
M/C aarch64
Shared libs
Destination directory: /home/pi/source/rpi-ffmpeg/out/arm64-bullseye-4.3.4-shared-rel
Out of tree builds are impossible with config.h in source dir.
```

jc-kynesim, 2022-09-15:
Run "git clean -dxf" - that will get rid of the old in-tree objects etc.

JooJooBee666, 2022-09-15:
Well, it completed compiling but after I ran a make install but it didn't actually install ffmpeg anywhere.  Now its just completely broken (i.e. no ffmpeg in path).  I tried both shared and static.  What did I miss?

JooJooBee666, 2022-09-16:
I just noted that 4.3.4 was up on the repos.  Is that this version already or do we need to wait a bit longer?

jc-kynesim, 2022-09-16:
a) I'm pretty sure that the BUILD.txt I pointed you at made it clear that you needed to get rid of the previous version and copy the result if you wanted it globally accessible.
b) 4.3.4 is the version used by PiOS and in the branch I pointed you at
c) Dunno about the camera issue

JooJooBee666, 2022-09-16:
Yeah, I didn't finish reading the build.txt but noted that afterward.  

Regarding the version number, I was just trying to clarify since that is also the version number on your repo but I confirmed that that version is also broken still.

Regarding the camera, I wasn't expecting you to know, but was hoping maybe something would make sense when you heard it.  I'm just reloading.....AGAIN.

closed 2022-09-17 by JooJooBee666

ViktuK359, 2022-09-28:
> Fix now pushed to my ffmpeg repo (https://github.com/jc-kynesim/rpi-ffmpeg.git branch test/4.3.4/rpi_main if you want to build it yourself). It'll be in the next ffmpeg update. Also fixes the same issue in mkv generation.

And how update for RPi4 HAOS?

pelwell, 2022-09-28:
That's a question for the HAOS devs, surely?

roger-, 2022-10-28:
Does anyone have a binary with this fix incorporated?

hackneyb, 2022-11-19:
> Fix now pushed to my ffmpeg repo (https://github.com/jc-kynesim/rpi-ffmpeg.git branch test/4.3.4/rpi_main if you want to build it yourself). It'll be in the next ffmpeg update.

@jc-kynesim Would you have an estimate on how long these releases take and which version will contain this fix?  My path to have this working is wait for the ffmpeg update and then for HAOS to uptake that ... I'd appreciate you shedding some light on this for me.  Thanks

popcornmix, 2022-11-20:
There was a recent ffmpeg update in RPiOS (timestamp shows it was built on Nov 7).

LaurenceGough, 2022-11-21:
Is there any way to get that ffmpeg update installed using apt on a Pi? I tried installing the deb files manually but it's one dependency after another missing and it was getting a bit crazy, obviously this was what apt was made for.

apt sources is the standard: `deb https://archive.raspberrypi.org/debian/ bullseye main`

Many thanks

jherby2k, 2022-11-21:
> Fix now pushed to my ffmpeg repo (https://github.com/jc-kynesim/rpi-ffmpeg.git branch test/4.3.4/rpi_main if you want to build it yourself). It'll be in the next ffmpeg update. Also fixes the same issue in mkv generation.

@jc-kynesim This does not appear to have trickled up, as nightly builds still don't include these patches. They also target a very old version of ffmpeg. Any update for us?

mantaalex, 2023-01-03:
still doesn't work for some people @jc-kynesim do you have some news? thanks on advance

hackneyb, 2023-01-06:
> When you say 'still does not work for some people' - does that imply that it does work for others? Assuming so, what are the differences between those it does work for and those it doesn't? I'll happily debug but I need an actual failing example on PiOS. The initial example given in this thread works for me with the current PiOS distribution ffmpeg.

Not a developer, but there are extensive threads detailing what doesn't work (here: https://github.com/blakeblackshear/frigate/issues/3780) and an official bug fix request at the ffmpeg page here: https://trac.ffmpeg.org/ticket/10060.  The specific case that locks "some people" out of using acceleration is an RPi4 64 bit scenario with ffmpeg packaged from your repository as part of Frigate.  Frigate runs as an add-on (in a container) in Home Assistant with the affected ffmpeg version (4.3.3?) and acceleration used by cameras does not work.   Add-on container images can be manually patched when they launch to fix but this is temporary until the next restart and beyond the hobbyist (me) capabilities.  I'm sure I could pull it off if I dedicated an evening to it but it's a band-aid.  Since Home Assistant can be ran in multiple configurations, only the typical HAOS + add-on install mode is affected.  If you run HA on non-restricted linux distribution then ffmpeg that Frigate uses can be patched easier.  

More info here, with many messy examples of what works and what doesn't: https://github.com/blakeblackshear/frigate/issues/3780

@jherby2k has a good handle on where it is reproducible and where it isn't, and took the initiative to open the ffmpeg request which in theory will trickle down to your project which will fix the ffmpeg version the Frigate project ultimately uses downstream.  I think the last idea here was if you can help influence the ffmpeg project bug fix request, or debug and fix it in parallel in your project.  The definition of the issue is obtainable, and I can help get you the info on what I'm running to help, but the Frigate project link above is teeming with examples.

6by9, 2023-01-06:
Is that FFmpeg bug report valid for this issue? It's referencing DECODE, whereas this is ENCODE.

Accurate bug reports are critical if you want developers to investigate them.

hackneyb, 2023-01-07:
> Is that FFmpeg bug report valid for this issue? It's referencing DECODE, whereas this is ENCODE.
> 
> Accurate bug reports are critical if you want developers to investigate them.

Frigate uses ffmpeg to decode https://docs.frigate.video/configuration/hardware_acceleration

6by9, 2023-01-07:
https://github.com/raspberrypi/linux/issues/5150#issuecomment-1244026046
> I now understand what is going on - the mp4 muxer appears to require the SPS/PPS extradata at init time if it is going to take it from the context, but the h/w encoder won't provide it before it has [...]

https://github.com/raspberrypi/linux/issues/5150#issuecomment-1245173832
> Just ffmpeg - it all works fine if you have a raw h264 stream as the output type from ffmpeg - it is just separating off SPS/PPS for the mp4 muxer that has issues and that is all internal to ffmpeg [...]

https://github.com/raspberrypi/linux/issues/5150#issuecomment-1247922065
> Fix now pushed to my ffmpeg repo (https://github.com/jc-kynesim/rpi-ffmpeg.git branch test/4.3.4/rpi_main if you want to build it yourself). It'll be in the next ffmpeg update. Also fixes the same issue in mkv generation.

That's the sum total of the fix for this issue related to encoding videos.

If using https://github.com/jc-kynesim/rpi-ffmpeg.git also fixes something else, then great, but it is not related to this issue regardless of what others think.
Telling the FFmpeg devs to look at this issue will not help them solve any decode related issue.

blakeblackshear, 2023-01-07:
As the frigate maintainer, I can confirm that this issue is not relevant.

hackneyb, 2023-01-07:
> If using https://github.com/jc-kynesim/rpi-ffmpeg.git also fixes something else, then great, but it is not related to this issue regardless of what others think. Telling the FFmpeg devs to look at this issue will not help them solve any decode related issue.

Thanks for taking the time to do this, then.  Clarity is good.

hackneyb, 2023-01-07:
> As the frigate maintainer, I can confirm that this issue is not relevant.

thank you