raspberrypi/linux #5151: firmware 20220811 - turn off power led not work

antonellocaroli, 2022-08-30:
### Describe the bug

With firmware 20220811, the power LED does not switch off.

### Steps to reproduce the behaviour

In the config.txt there are:

```
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
```

Reinstalling firmware 20220331 works again as it should.

### Device (s)

Raspberry Pi 4 Mod. B

### System

NAME=Gentoo
ID=gentoo
PRETTY_NAME="Gentoo Linux"
ANSI_COLOR="1;32"
HOME_URL="https://www.gentoo.org/"
SUPPORT_URL="https://www.gentoo.org/support/"
BUG_REPORT_URL="https://bugs.gentoo.org/"
VERSION_ID="2.8"

Mar 24 2022 13:19:43 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start_cd)

Linux GentooP 5.15.36+ #1 SMP PREEMPT Thu May 5 19:40:00 CEST 2022 aarch64 GNU/Linux

polojoe, 2022-08-30:
see here https://github.com/raspberrypi/firmware/issues/1742

antonellocaroli, 2022-08-30:
@polojoe thank you, I had missed the post...

This seems to work:

`dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off`

closed 2022-08-30 by popcornmix