raspberrypi/linux #5153: Unusual Uasp file

symbios24, 2022-08-30:
### Describe the bug

Hello there

Is it possible please to add to the file: https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/usb/storage/unusual_uas.h

Some usb 3.0 chipsets/controllers more??

The Jmicron Jms578

The Asmedia Asm1153E

The Asmedia Asm235CM

And The Initio Inic3639

Thanks in advance 

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspbian Buster latest version

pelwell, 2022-08-31:
Do you have the information required? A Pull Request would be ideal.

symbios24, 2022-08-31:
Whay information you need??

I dont know how to do pull requests

pelwell, 2022-08-31:
> What information you need??

The text to add to uasp.h.

symbios24, 2022-08-31:
You need the output of the lsusb command ??

pelwell, 2022-08-31:
lsusb would give the vendor and device IDs, but that doesn't tell us which quirks are needed for each one.

symbios24, 2022-08-31:
My asm1153e quirks are: 174c:55aa

symbios24, 2022-08-31:
How can i see that,with what command??

pelwell, 2022-08-31:
Why do you think the controllers you listed need to be added to `unusual_uas.h`? What have you been doing to set the quirk(s) up to now?

symbios24, 2022-08-31:
The jms567 that is included in the unusual_uas file works good with quirks enabled and both mechanical and ssds have great performance 

Im using both mechanical and ssds with that controller's most of them perform somehow ok with quirks enabled and ssds but with mechanical disks their performance is very bad with quirks or not there is no difference 

If you cannot add them all at least add the Asmedia Asm1153E
And the Initio inic3639

pelwell, 2022-08-31:
What I'm hearing is "I have several devices that use a number of SATA bridges. One of them works well, the others don't. The one that works well is included in unusual_uas.h, the others aren't, so please add the others to it to fix them."

What have I got wrong, or missed out?

symbios24, 2022-08-31:
If you cannot add any device,can you at least tell me how to add it my self manual to the file in my os directly??

pelwell, 2022-08-31:
That's a kernel header file - you'll have to recompile the kernel. There are instructions for kernel building here: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

pelwell, 2022-08-31:
You didn't answer my earlier questions:
> What have you been doing to set the quirk(s) up to now?

symbios24, 2022-08-31:
I set them from the cmdline.txt file
By put the device id

Eg: usb-storage.quirks=174c:55aa:u

pelwell, 2022-08-31:
OK, we're getting somewhere, the `:u` at the end translates to `US_FL_IGNORE_UAS`.

Have you done something similar with all the other devices? If so, tell me what the `quirks=...` entries were.

trejan, 2022-08-31:
[OP has a post on the forum saying disabling UAS didn't help for those adapters.](https://forums.raspberrypi.com/viewtopic.php?t=339591) unusual_uas.h doesn't do anything that can't be done via usb-storage.quirks in cmdline.txt.

symbios24, 2022-08-31:
Only the device id is changing in the other devices
The :u remains the same,are there other commands than :u to put in the end?

Do you want the other devices ids?

pelwell, 2022-08-31:
> Do you want the other devices ids?

Yes, of course.

What do you have to say about "OP has a post on the forum saying disabling UAS didn't help for those adapters"?

@trejan is correct in saying that unusual_uas.h doesn't do anything you can't do with quirks on the command line - it's just a convenience.

symbios24, 2022-08-31:
I asked there also maybe someone knew something.

Jms578: 152d:0578:u

Inic3639: 13FD:3960:u

Asm1153E: 174c:55aa:u

Asm235cm: 174c:55aa :u

Im not 100,% for the initio controller i have to take a look at it later.

asmello, 2022-11-10:
Ok, Phil redirected me here from the discussion in #5246. In summary, I also have a JMS578 device that required disabling UAS to work properly. I did so with a quirk override in `cmdline.txt`, same as the OP here, and wanted to contribute that back to the kernel - particularly because I saw there was a quirks entry for this device already, but it didn't seem to be doing its job.

Phil helped me identify that we just need to include `US_FL_IGNORE_UAS` in the flags set for this device's quirks entry. I can do that with a simple PR. However, the fact that the original contributor of the quirks entry didn't set this flag suggests he didn't have the same issues we're having. Which would mean not everyone with this device requires disabling UAS.

So the question is, do we want to set this flag in the kernel, even if it could degrade performance for some people? Do we prioritise the device working correctly in all cases, or performing best for those who _can_ take advantage of UAS?

Given general Linux philosophy I'd assume the former, but wanted to put this up for discussion here before making a PR.

symbios24, 2022-11-17:
Only sata 2 chipsets without uasp support work correct with mechanical hdds
95% of the chipsets that i tried with uasp support they work very slow with mechanical drives with huge delay in read/write operations even with uasp disabled,only the jms566/567 worked correct with uasp disabled.