raspberrypi/linux #5156: brcmfmac patch to include ref to 43436* firmware files?

macmpi, 2022-08-31:
### Describe the bug

With Pi Zero 2 W using 43436 SYN chip and `brcmfmac` driver, we can note that driver `modinfo` does not list relevant firmware file.
This may cause impractical issues to associate the 2 pieces together.
It seems automatic loading does not use that information, but some distros may use that to include firmware files, etc...

Would it make sense to add relevant firmware files references within `brcmfmac` driver for sake of consistency?
Not sure how to proceed, nor likelihood to have that accepted upstream...
Thoughts?

### Steps to reproduce the behaviour

```
$ modinfo -F firmware brcmfmac
brcm/brcmfmac*-sdio.*.bin
brcm/brcmfmac*-sdio.*.txt
brcm/brcmfmac43752-sdio.clm_blob
brcm/brcmfmac43752-sdio.bin
brcm/brcmfmac43012-sdio.clm_blob
brcm/brcmfmac43012-sdio.bin
brcm/brcmfmac4373-sdio.clm_blob
brcm/brcmfmac4373-sdio.bin
brcm/brcmfmac4359-sdio.bin
brcm/brcmfmac4356-sdio.clm_blob
brcm/brcmfmac4356-sdio.bin
brcm/brcmfmac4354-sdio.clm_blob
brcm/brcmfmac4354-sdio.bin
brcm/brcmfmac43456-sdio.bin
brcm/brcmfmac43455-sdio.clm_blob
brcm/brcmfmac43455-sdio.bin
brcm/brcmfmac43430b0-sdio.bin
brcm/brcmfmac43430-sdio.clm_blob
brcm/brcmfmac43430-sdio.bin
brcm/brcmfmac43430a0-sdio.bin
brcm/brcmfmac4339-sdio.bin
brcm/brcmfmac43362-sdio.bin
brcm/brcmfmac4335-sdio.bin
brcm/brcmfmac43341-sdio.bin
brcm/brcmfmac43340-sdio.bin
brcm/brcmfmac4334-sdio.bin
brcm/brcmfmac4330-sdio.bin
brcm/brcmfmac4329-sdio.bin
brcm/brcmfmac43241b5-sdio.bin
brcm/brcmfmac43241b4-sdio.bin
brcm/brcmfmac43241b0-sdio.bin
brcm/brcmfmac43143-sdio.bin
brcm/brcmfmac4373.bin
brcm/brcmfmac43569.bin
brcm/brcmfmac43242a.bin
brcm/brcmfmac43236b.bin
brcm/brcmfmac43143.bin
```

### Device (s)

Raspberry Pi Zero 2 W

### System

Linux pi32 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux

jchidley, 2022-08-31:
I own this specific system Raspberry Pi Zero 2 W.  I can confirm that this is an issue on it and I am willing to do tests and investigation. You will have to be really specific and be clear exactly what you want me to do…

macmpi, 2022-09-06:
@pelwell any thought?

pelwell, 2022-09-06:
It's not high on the list of thing to look at. A Pull Request might be the quickest way to a solution for you.

macmpi, 2022-09-06:
Well understood, thanks.
Adequately patching that [`brcmfmac/sdio.c`](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/net/wireless/broadcom/brcm80211/brcmfmac/sdio.c#L607-L662) is probably too touchy for me though, as it might be more involving than just adding few constants... Never mind.