raspberrypi/linux #5161: 64 bit - Kernel Oops PREEMP SMP

tennisparty, 2022-09-02:
### Describe the bug

Hello when I stop running a flask server I get the following error message

```Message from syslogd@raspberrypi at Sep  2 15:22:51 ...
 kernel:[  328.916992] Internal error: Oops: 96000004 [#1] PREEMPT SMP

Message from syslogd@raspberrypi at Sep  2 15:22:51 ...
 kernel:[  329.169279] Code: f8266804 91014044 eb04003f 540001a0 (f9401424)```

The pi is headless and disconnects when this behaviour happens. I am running the most recent Bullseye, 64bit. 

### Steps to reproduce the behaviour

Run flask server and then kill the python process with ctrl - c

### Device (s)

Raspberry Pi 3 Mod. A+

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

Aug 26 2022 14:04:36 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

Linux raspberrypi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

### Logs

```[   22.632349] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   35.807879] cam-dummy-reg: disabling
[  328.869264] Unable to handle kernel paging request at virtual address 40c53d3e3c3e6966
[  328.877279] Mem abort info:
[  328.880073]   ESR = 0x96000004
[  328.883159]   EC = 0x25: DABT (current EL), IL = 32 bits
[  328.888593]   SET = 0, FnV = 0
[  328.891656]   EA = 0, S1PTW = 0
[  328.894827]   FSC = 0x04: level 0 translation fault
[  328.899793] Data abort info:
[  328.902856]   ISV = 0, ISS = 0x00000004
[  328.906773]   CM = 0, WnR = 0
[  328.909780] [40c53d3e3c3e6966] address between user and kernel address ranges
[  328.916992] Internal error: Oops: 96000004 [#1] PREEMPT SMP
[  328.922566] Modules linked in: 8021q garp stp llc ov9281 snd_soc_hdmi_codec vc4 cec drm_kms_helper brcmfmac brcmutil cfg80211 snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops i2c_mux_pinctrl i2c_mux raspberrypi_hwmon rfkill bcm2835_unicam bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_dv_timings v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc v4l2_fwnode v4l2_async videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_bcm2835(C) snd_pcm snd_timer snd mc vc_sm_cma(C) i2c_bcm2835 uio_pdrv_genirq uio i2c_dev i2c_bcm2708 drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  328.981833] CPU: 3 PID: 11232 Comm: python3 Tainted: G         C        5.15.61-v8+ #1579
[  328.990014] Hardware name: Raspberry Pi 3 Model A Plus Rev 1.0 (DT)
[  328.996278] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  329.003240] pc : media_graph_walk_next.part.9+0xd0/0x198 [mc]
[  329.009013] lr : __media_pipeline_stop+0x48/0xd0 [mc]
[  329.014078] sp : ffffffc008e1ba90
[  329.017388] x29: ffffffc008e1ba90 x28: 0000000000000000 x27: 0000007fa1bf73f8
[  329.024528] x26: ffffffee3c5388c8 x25: 0000000000000009 x24: 0000000000100000
[  329.031669] x23: 0000000000000001 x22: ffffff80055c38e8 x21: ffffff8006bcc538
[  329.038809] x20: ffffff8006bccd38 x19: ffffff8006bccd40 x18: 000000000000000b
[  329.045949] x17: 0000000000000000 x16: ffffffee3b85a9c0 x15: ffffff8006ac0030
[  329.053089] x14: ffffff8003359be0 x13: 0000000000000000 x12: 0000000000000000
[  329.060230] x11: ffffff8003359b60 x10: 0000000000001a00 x9 : ffffffedf5643c68
[  329.067370] x8 : ffffff8006bccd60 x7 : 0000000000000001 x6 : 0000000000000020
[  329.074509] x5 : 0000000000000002 x4 : ffffff80055c3938 x3 : 0000000000000002
[  329.081649] x2 : ffffff80055c38e8 x1 : 40c53d3e3c3e693e x0 : ffffff8006bccd40
[  329.088790] Call trace:
[  329.091231]  media_graph_walk_next.part.9+0xd0/0x198 [mc]
[  329.096644]  media_pipeline_stop+0x34/0x50 [mc]
[  329.101188]  unicam_stop_streaming+0x78/0x178 [bcm2835_unicam]
[  329.107028]  __vb2_queue_cancel+0x3c/0x220 [videobuf2_common]
[  329.112794]  vb2_core_queue_release+0x2c/0x58 [videobuf2_common]
[  329.118814]  _vb2_fop_release+0x88/0xa8 [videobuf2_v4l2]
[  329.124137]  unicam_v4l2_release+0x58/0xe0 [bcm2835_unicam]
[  329.129713]  v4l2_release+0xc8/0x108 [videodev]
[  329.134308]  __fput+0x78/0x240
[  329.137363]  ____fput+0x18/0x28
[  329.140501]  task_work_run+0x90/0x168
[  329.144165]  do_exit+0x3a0/0xab8
[  329.147392]  do_group_exit+0x4c/0xa8
[  329.150966]  get_signal+0xd4/0x8d8
[  329.154368]  do_notify_resume+0x2e0/0x470
[  329.158379]  el0_svc+0x58/0x60
[  329.161433]  el0t_64_sync_handler+0x90/0xb8
[  329.165614]  el0t_64_sync+0x1a0/0x1a4
[  329.169279] Code: f8266804 91014044 eb04003f 540001a0 (f9401424) 
[  329.175371] ---[ end trace 6b255843d5e9f0d2 ]---
[  329.180241] Fixing recursive fault but reboot is needed!```

pelwell, 2022-09-02:
You say you a running "a flask server". Apart from the fact that you don't say what a "flask server" is, you also don't say what the "flask server" is doing - I suspect it isn't simply saying "Hello world!".

JamesH65, 2022-09-02:
It appears to be doing something with the camera? But not clear from the report.

JamesH65, 2022-09-02:
> I had rarely seen a "unable to handle kernel paging request" in #5138 myself (unfortunately none of which I saved for the ticket). Stacktrace there sometimes also had `do_exit`.
> 
> I suggest you wait for a new kernel release and try again

This seems entirely unrelated to the issue, have you posted on the wrong thread?

pelwell, 2022-09-02:
No - @kralo's probably correct.

tennisparty, 2022-09-02:
Hello, sorry I wasn't clear I am running a python file which runs picamera2, a server hosting a mjpeg stream, and some motion detection on a low res stream in the background - a detailed description is on this issue [picamera2 simultaneous](https://github.com/raspberrypi/picamera2/issues/255). 

Let me know if there is any other useful information which I can send over

lategoodbye, 2022-09-04:
@tennisparty could you please point directly to the script which triggers the issue (the issue contains multiple versions)? Is it possible to provide a minimal script without the webserver stuff?

tennisparty, 2022-09-17:
Thanks everyone, I am using a ov9281 and doing some processing in the background to detect motion so maybe the issue that is linked above is indeed the same. @kralo do you know if the kernel is now updated, sorry I am not sure exactly where to check?

pelwell, 2022-09-17:
It's not in the stable branch yet, but you can grab the beta kernel with `sudo rpi-update`.

tennisparty, 2022-09-18:
Thanks @pelwell I'll use that for the moment. What is the best way to track when that edit is merged into the stable branch?

pelwell, 2022-09-18:
The next update here will include the required patch: https://github.com/raspberrypi/rpi-firmware/commits/stable
Any kernel built from rpi-5.15.y labelled 5.15.63 or later has the fix.

tennisparty, 2022-09-18:
Thankyou, that's really useful

closed 2022-09-27 by tennisparty

kralo, 2022-10-11:
Yes, it had been updated roughly 10 days ago. Have a look at `rpi-update`