raspberrypi/linux #5162: Oops "undefined instruction"

gongsearch, 2022-09-02:
### Describe the bug

My RPi 4B (kernel and userland 32-bit) sometimes hangs completely. Only power-cycling helps in this case. On the hunt for the reason I found the following Oops in my logs, which seems to be the root-cause.

### Steps to reproduce the behaviour

Unknown. It just happens. Sometimes after 2 days, sometims after 8 weeks.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start_cd)575 SMP Fri Jul 22 20:29:46 BST 2022 armv7l GNU/Linux

kernel: 5.15.56-v7l+ #1Aug  9 2022 13:44:57

Previous kernel-versions also had this problem. 

### Logs

[582562.963366] BUG: unsupported FP instruction in kernel mode
[582562.963398] FPEXC == 0x40000700
[582562.963417] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
[582562.963440] Modules linked in: ip6t_REJECT nf_reject_ipv6 nft_counter ipt_REJECT nf_reject_ipv4 xt_set xt_multiport nft_compat nf_tables x_tables ip_set_hash_ip ip_set nfnetlink wireguard curve25519_neon libcurve25519_generic libchacha20poly1305 chacha_neon poly1305_arm ip6_udp_tunnel udp_tunnel cdc_acm btrfs blake2b_generic xor xor_neon zstd_compress raid6_pq sg rpivid_hevc(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev raspberrypi_hwmon mc nvmem_rmem uio_pdrv_genirq uio drm jitterentropy_rng fuse drm_panel_orientation_quirks backlight ipv6
[582562.963771] CPU: 0 PID: 18871 Comm: smbd Tainted: G         C        5.15.56-v7l+ #1575
[582562.963797] Hardware name: BCM2711
[582562.963812] PC is at vfp_reload_hw+0x30/0x44
[582562.963842] LR is at __und_usr_fault_32+0x0/0x8
[582562.963862] pc : [<c0203780>]    lr : [<c0200f80>]    psr: a0000013
[582562.963880] sp : ef445fb0  ip : ef445f30  fp : 00000000
[582562.963896] r10: ef4441f0  r9 : c0200fe4  r8 : 0012bec0
[582562.963913] r7 : 00000001  r6 : 2dff86c3  r5 : c0217990  r4 : cfe600f0
[582562.963931] r3 : c135b08c  r2 : b6755588  r1 : c02c843c  r0 : ed2d8b02
[582562.963948] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[582562.963973] Control: 30c5383d  Table: 1440d580  DAC: fffffffd
[582562.963989] Register r0 information: 0-page vmalloc region starting at 0xeac00000 allocated at iotable_init+0x0/0x118
[582562.964032] Register r1 information: non-slab/vmalloc memory
[582562.964055] Register r2 information: non-paged memory
[582562.964075] Register r3 information: non-slab/vmalloc memory
[582562.964096] Register r4 information: non-slab/vmalloc memory
[582562.964117] Register r5 information: non-slab/vmalloc memory
[582562.964137] Register r6 information: non-paged memory
[582562.964157] Register r7 information: non-paged memory
[582562.964176] Register r8 information: non-paged memory
[582562.964195] Register r9 information: non-slab/vmalloc memory
[582562.964216] Register r10 information: non-slab/vmalloc memory
[582562.964237] Register r11 information: NULL pointer
[582562.964257] Register r12 information: non-slab/vmalloc memory
[582562.964277] Process smbd (pid: 18871, stack limit = 0xe5dec054)
[582562.964297] Stack: (0xef445fb0 to 0xef446000)
[582562.964316] 5fa0:                                     01670ce0 01670cc8 a9ef9b14 00000008
[582562.964339] 5fc0: 01670cc0 01670cc8 01670d3c a9ef9b14 01670ce0 00000000 a9ef9b14 01a0b490
[582562.964363] 5fe0: b5e84f5c a9ef9aec b5e712f0 b6755588 60000010 ffffffff 00000000 00000000
[582562.964382] Backtrace: no frame pointer
[582562.964409] Code: 128aa080 e89a0162 e3110102 0a000003 (eee96a10)
[582562.964433] ---[ end trace 0c3584db2adb3d0f ]---

lategoodbye, 2022-09-04:
The trace suggests that you are running Samba on the Raspberry Pi. Could you please provide more information about this?

syss, 2023-02-25:
something similar happend on my raspberry as well.

```
Feb 23 06:05:28 rpi2 kernel: [671214.396225] BUG: unsupported FP instruction in kernel mode
Feb 23 06:05:28 rpi2 kernel: [671214.403266] FPEXC == 0x40000000
Feb 23 06:05:28 rpi2 kernel: [671214.407859] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
Feb 23 06:05:28 rpi2 kernel: [671214.416120] Modules linked in: nft_fib_ipv4 nft_ct nft_fib_ipv6 nft_fib wireguard curve25519_neon libcurve25519_generic libchacha20poly1305 chacha_neon poly1305_arm ip6_udp_tunnel udp_tunnel rpcsec_gss_krb5 sha256_generic cfg80211 rfkill 8021q garp stp llc nft_chain_nat xt_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter xt_multiport xt_tcpudp xt_comment nft_compat nf_tables nfnetlink btrfs blake2b_generic xor xor_neon zstd_compress raid6_pq sg uas bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) videodev snd_pcm raspberrypi_hwmon mc snd_timer snd vc_sm_cma(C) uio_pdrv_genirq uio fixed drm nfsd fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Feb 23 06:05:28 rpi2 kernel: [671214.505477] CPU: 1 PID: 9072 Comm: rtorrent main Tainted: G         C        5.15.61-v7+ #1579
Feb 23 06:05:29 rpi2 kernel: [671214.517263] Hardware name: BCM2835
Feb 23 06:05:29 rpi2 kernel: [671214.522280] PC is at vfp_reload_hw+0x30/0x44
Feb 23 06:05:29 rpi2 kernel: [671214.528193] LR is at __und_usr_fault_32+0x0/0x8
Feb 23 06:05:29 rpi2 kernel: [671214.534360] pc : [<80103780>]    lr : [<80100f80>]    psr: a0000013
Feb 23 06:05:29 rpi2 kernel: [671214.541957] sp : 8858ffb0  ip : 8858ff30  fp : 00000001
Feb 23 06:05:29 rpi2 kernel: [671214.548485] r10: 8858e1f0  r9 : 80100fe4  r8 : 8858e208
Feb 23 06:05:29 rpi2 kernel: [671214.554988] r7 : 00000001  r6 : 8858e22c  r5 : be59ec10  r4 : 00000000
Feb 23 06:05:29 rpi2 kernel: [671214.562796] r3 : 8114f08c  r2 : 76d6c230  r1 : 8242c000  r0 : ee189a10
Feb 23 06:05:29 rpi2 kernel: [671214.570589] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Feb 23 06:05:29 rpi2 kernel: [671214.578995] Control: 10c5387d  Table: 04cc806a  DAC: 00000055
Feb 23 06:05:29 rpi2 kernel: [671214.585984] Register r0 information: non-paged memory
Feb 23 06:05:29 rpi2 kernel: [671214.592260] Register r1 information: slab kmalloc-2k start 8242c000 pointer offset 0 size 2048
Feb 23 06:05:29 rpi2 kernel: [671214.603196] Register r2 information: non-paged memory
Feb 23 06:05:29 rpi2 kernel: [671214.609435] Register r3 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.616247] Register r4 information: NULL pointer
Feb 23 06:05:29 rpi2 kernel: [671214.622056] Register r5 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.628795] Register r6 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.635476] Register r7 information: non-paged memory
Feb 23 06:05:29 rpi2 kernel: [671214.641511] Register r8 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.648130] Register r9 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.654723] Register r10 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.661409] Register r11 information: non-paged memory
Feb 23 06:05:29 rpi2 kernel: [671214.667474] Register r12 information: non-slab/vmalloc memory
Feb 23 06:05:29 rpi2 kernel: [671214.674130] Process rtorrent main (pid: 9072, stack limit = 0xc37021d8)
Feb 23 06:05:29 rpi2 kernel: [671214.681672] Stack: (0x8858ffb0 to 0x88590000)
Feb 23 06:05:29 rpi2 kernel: [671214.686953] ffa0:                                     00004000 00000002 00000000 00000000
Feb 23 06:05:29 rpi2 kernel: [671214.696886] ffc0: 0091e488 01524000 01524000 00004000 00868bf8 007477a8 7ef31654 0091e490
Feb 23 06:05:29 rpi2 kernel: [671214.706843] ffe0: 76efd544 7ef31650 76efd980 76d6c230 60000010 ffffffff 00000000 00000000
Feb 23 06:05:29 rpi2 kernel: [671214.716871] Backtrace: invalid frame pointer 0x00000001
Feb 23 06:05:29 rpi2 kernel: [671214.723091] Code: 128aa080 e89a0162 e3110102 0a000003 (eee96a10) 
Feb 23 06:05:29 rpi2 kernel: [671214.730169] ---[ end trace 29c511d5697c583e ]---
Feb 24 08:47:19 rpi2 kernel: [    0.000000] Booting Linux on physical CPU 0xf00
Feb 24 08:47:19 rpi2 kernel: [    0.000000] Linux version 5.15.84-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1613 SMP Thu Jan 5 11:59:48 GMT 2023
```

i assume that the usb chip (usb+network) could have been under heavy load when this happend.

pelwell, 2023-02-25:
This particular problem has been seen before with rtorrent: https://github.com/raspberrypi/linux/issues/4000