raspberrypi/linux #5165: Error at startup - processing of config.txt file

GeoffDraper, 2022-09-06:
### Describe the bug

I'm new to Raspberry Pi and modern software development methods, so please forgive my inevitable breaches of protocol. 
Also, this may be a known bug - I felt unable to understand the 27 pages of rather cryptic bug reports - sorry. 

On Raspberry Pi 4B (4GB version) using Raspberry Pi OS version 10, I edited /boot/config.txt using the nano editor to append a single line 
dtoverlay=uart2
in an attempt to activate UART2. 
After rebooting, which seemed rather longer than usual, config.txt had reverted to its original form, and no new UART showed up in the 
ls -l /dev
report. 
There was a new file in /boot named config.txt.save which was my edited version of config.txt -- I assume this is a recovery step taken by the boot process if it deems config.txt to be invalid. However it looked valid to me, though the file format is not well documented. 
Eventually, after quite a lot of reading, I tried again, this time placing a comment line (#...) before the dtoverlay=uart2 line. This rebooted fine, and the new UART was visible as /dev/ttyAMA1 in the report by 
ls -l /dev
Below in section "Additional context" is a  copy/paste image of the config.txt.save file. 
I'd prefer to attach the actual config.txt.save file, so that non-printing characters can be examined, but I can't see how. 
Here are my ponderings re what went wrong:
1. I simply misunderstood the required file format. 
2. The nano editor put some unexpected non-printing characters in the config.txt file
3. The boot process has a bug which rejects some valid config.txt files. 
4. Maybe some part of the processing of the enable_uart=1 line? 

### Steps to reproduce the behaviour

As noted above. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@raspberrypi:~ $ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage5

Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
Revision	: c03114
Serial		: 100000006e1f07fb
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Nov 18 2021 16:16:49 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       12219440 8948396   2670264  78% /
devtmpfs         1800564       0   1800564   0% /dev
tmpfs            1965428   48668   1916760   3% /dev/shm
tmpfs            1965428    8780   1956648   1% /run
[... 702 lines cut ...]
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

# NOOBS Auto-generated Settings:
enable_uart=1

dtoverlay=uart2

pi@raspberrypi:~ $

pelwell, 2022-09-07:
Before we go any further, you need to know that the firmware _does not, under any circumstances, write to any filesystem_.

`,save` files are backup files created by `nano`. If the editor runs out of memory, crashes or is killed, the `.save` file remains, allowing the user to recover at least some of their changes.

GeoffDraper, 2022-09-07:
Thank you pelwell for your helpful comments. 
I'm relatively unfamiliar with Raspberry Pi, with Linux, with Raspberry Pi OS, and with the nano editor. 
Reviewing my experience, in light of your comments, I now see another interpretation -- that maybe I failed to complete my nano edit, and maybe I failed to check (with say $ cat) that my edit was successful. I no longer have a record of those steps. These events would have also led to what I saw happen on rebooting. 
Therefore I will close this bug report. 
Again, thank you for your help.