raspberrypi/linux #5166: h.264 decoder doesn't handle GDR streams?

ford-prefect, 2022-09-06:
### Describe the bug

I have an H.264 stream from a camera that seems to use GDR (gradual decoding refresh). The stream has no IDR frames, but the output frames seem to be constructed fine by ffmpeg.

### Steps to reproduce the behaviour

A sample capture from the camera is here: https://arunraghavan.net/temp/main-nosei.h264

The following pipeline (or just `ffplay`) works:

```sh
gst-launch-1.0 filesrc location=main-nosei.h264 ! h264parse ! avdec_h264 ! autovideosink
```

The following does not:

```sh
gst-launch-1.0 filesrc location=main-nosei.h264 ! h264parse ! v4l2h264dec ! autovideosink
```

The pipeline just feeds h.264 buffers to the decoder continuously, and no frames are output by the decoder.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```sh
$ cat /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage5

$ vcgencmd version
Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (star

$ uname -a
Linux streampi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
```

6by9, 2022-09-10:
I don't know for sure whether GDR is supported by the hardware. The software is definitely configured to wait for an IDR frame to sync to on the basis that it can't accurately decode anything until it has such.

I'm not in the office again until w/c 19/9, so will try and look then.

ford-prefect, 2022-09-10:
Thank you, appreciate it.

uzgit, 2023-09-23:
@6by9 Any answer on this?

6by9, 2023-09-25:
I can find no reference to GDR in any of the documentation or source that I have available, therefore I can only conclude that it isn't supported.

All that code was written by Broadcom and those involved originally are now employed elsewhere, so it's not the sort of thing that will be considered for implementation either. Sorry.

closed 2023-09-25 by 6by9

popcornmix, 2023-09-26:
I think at the mmal layer if you disable `MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT` then you will get frames out without requiring starting from an I frame.

They will be partially corrupt until enough frames have come out to have refreshed all macroblocks.

But this option isn't exposed through V4L2/gstreamer. If you were comfortable building the kernel then it would be possible to add it.

ndufresne, 2023-09-27:
@6by9 The HW decoder part is no different. The handling of intra-refresh (gdr) is usually done by the bitstream parser/state manager. On the coder side, you have a P frame composed of a mix of I and P prediction macroblocks. The result is a corrupted image with some good blockes. Over the intra refresh period, the image will be clean.  What I'm unsure is how this is signalled, but any compliant decoders will be able to handle this, as long as they don't hide corrupted frames from the users.

uzgit, 2023-09-27:
I am trying to test this right now, but never have recompiled the kernel. I have the code but I cannot figure out how to actually change MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT before recompiling. Any tips?

ylatuya, 2023-09-27:
For AVC, this is usually signaled through SEI Recovery Point messages. The file name is "nosei", which seems to indicate the SEI messages were removed.  The parser is unable to mark these frames as non-delta-unit.

popcornmix, 2023-09-27:
My (untested) guess would be something like:
```
diff --git a/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c b/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c
index e21112bbb90b..41a4e9cc1272 100644
--- a/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c
+++ b/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c
@@ -2666,6 +2666,15 @@ static int bcm2835_codec_create_component(struct bcm2835_codec_ctx *ctx)
                                              MMAL_PARAMETER_VIDEO_STOP_ON_PAR_COLOUR_CHANGE,
                                              &enable,
                                              sizeof(enable));
+               /*
+                * Disable firmware option to conceal frames with errors
+                */
+               enable = 0;
+               vchiq_mmal_port_parameter_set(dev->instance,
+                                             &ctx->component->control,
+                                             MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT,
+                                             &enable,
+                                             sizeof(enable));
        } else if (dev->role == DEINTERLACE) {
                /* Select the default deinterlace algorithm. */
                int half_framerate = 0;
```

6by9, 2023-09-27:
> as long as they don't hide corrupted frames from the users.

The firmware hides all corrupted frames unless told otherwise, and parts of it specifically look for getting a key frame signaled.
I can't see anything obvious in the V4L2 docs to say whether that is the expected behaviour or not.

ndufresne, 2023-09-27:
Can't help much on the firmware side here, but on V4L2 side, as long as it not tested, you can see corrupted buffer signalling as undefined. But the spec is clear though on how to signal corrupted frames:

```
V4L2_BUF_FLAG_ERROR
When this flag is set, the buffer has been dequeued successfully, although the data might have been **corrupted**. This is recoverable, streaming may continue as normal and the buffer may be reused normally. Drivers set this flag when the VIDIOC_DQBUF ioctl is called.
```

In practice, FLAG_ERROR + payload == 0 is interpreted like failed and unusable, were FLAG_ERROR + payload == sizeimage is a corrupted image.

uzgit, 2023-09-27:
@6by9 Judging by the information OP has offered, I assume he's using the DJI H20T, as this appears to be the exact same issue I'm currently having.

[Assuming that is the case, here are three h264 streams](https://drive.google.com/drive/folders/1RrGMCmDO9V7THJadu9ozU5kq6C4weCLy?usp=sharing) that demonstrate the problem (or at least a very similar problem). They all come from the same drone. The hardware decoder on the Raspberry Pi 4 B can demonstrably decode the FPV stream with no problems (it seems to be the most normal of the streams). But only the software decoders are able to decode the stream from the H20T and the P1. The hardware decoders return nothing. The H20T does give a resolution of 1920x1440 as has been pointed out, but the P1 gives a resolution of 1920x1080 which I guess should be just at the limit of what the Pi should be able to do.

uzgit, 2023-09-27:
> My (untested) guess would be something like:
>
> ```
> diff --git a/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c b/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c [...]

I tried this fix, but now even my FPV stream, which worked before using hardware decoding with h264_v4l2m2m, returns no frames. Everything still works with the software decoders.

ford-prefect, 2023-09-28:
I've not got access to that drone any more, but it should be it indeed, I believe @uzgit. Thanks for the samples!

uzgit, 2023-09-29:
@popcornmix: adding your fix to disable MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT works on the 64-bit OS only, in my tests. Do you have any tips for getting this to work on the 32-bit OS? Moving to 64-bit breaks a lot of things in my application.

popcornmix, 2023-09-29:
> @popcornmix: adding your fix to disable MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT works on the 64-bit OS only, in my tests. Do you have any tips for getting this to work on the 32-bit OS? Moving to 64-bit breaks a lot of things in my application.

I'd say it's very unlikely this patch only works on 64-bit.
It sounds more likely to me you have something not set up correctly in your 32-bit image. 
Does it play other video files (without GDR)?
Perhaps the kernel/modules/dtbs haven't been copied over correctly?

uzgit, 2023-09-29:
> I'd say it's very unlikely this patch only works on 64-bit. It sounds more likely to me you have something not set up correctly in your 32-bit image.

I have been working only with the stock OS images and making as little changes to them as possible, like copying my streams onto it and installing `ffmpeg` and then trying to play the streams. I am obviously open to the fact that it is something with the image though, and not necessarily based strictly on the difference between 32- and 64-bit. I just don't know enough about the decoder to really diagnose the issue.

> Does it play other video files (without GDR)?

Yes, it plays the h264 stream from the FPV camera and other `.mp4` files with the hardware decoder with no problems. I guess it's not super relevant but the software decoders also are still able to handle the GDR streams.

> Perhaps the kernel/modules/dtbs haven't been copied over correctly?

I am following the directions from [the Raspberry Pi Linux Kernel page](https://www.raspberrypi.com/documentation/computers/linux_kernel.html), and have tried several times with both images, getting consistent results. Will keep trying though.

Please let me know if you have any other tips or suggestions.

popcornmix, 2023-09-29:
Add a `printk("My modified kernel")` next to the code that sets `MMAL_PARAMETER_VIDEO_DECODE_ERROR_CONCEALMENT`  and run dmesg after playing a video. Do you see the printk on 64-bit and 32-bit?

uzgit, 2023-09-29:
I see it only on the 64-bit one.

popcornmix, 2023-09-29:
I suspect you are not running the kernel (or modules) you built. Does `uname -a` indicate today's date?

uzgit, 2023-09-29:
Your suspicions were correct. It seems to be building correctly but not running the newly built kernel. `uname -a` gives April 3. Not sure why the 64-bit version is loading correctly but not the 32-bit version. I have deleted/re-cloned the code, then remade the change and recompiled, and it is still not using the newly built stuff.

uzgit, 2023-09-29:
So I can also confirm that the kernel fix makes the [DJI Payload-SDK](https://github.com/uzgit/Payload-SDK) code work - i.e., using the h264_v4l2m2m codec in `dji_camera_stream_decoder.cpp` with the main cameras, not just the FPV camera. I haven't seen any actual answers to this on the DJI forums, only a ton of unanswered questions. So hopefully anybody who goes over this in the future will see this post.

Here is an example from the P1 (resized from 1920x1080 to 640x480).

https://github.com/raspberrypi/linux/assets/14451567/8e2c33cb-0ab0-4032-99b0-faea9e44be1f

Here is an example from the H20T (resized from 1920x1440 to 640x480). It shows the cut off bottom portion of the stream.

https://github.com/raspberrypi/linux/assets/14451567/8109bd08-0609-4b5b-8d9d-c2cc20414292

Both streams come at 30 FPS, and I only need to process them at around 10 FPS. The decoder is able to decode at about 90 FPS in my tests. This makes me wonder if I could possibly configure the decoder to like process each frame in two "passes" or something, so it could somehow generate an output stream at full resolution but at half the frame rate. The H20T is the camera that I would prefer to use because it has a wider-angle lens. Does anybody have any idea on how to obtain the full resolution of the H20T stream? Is it possible at all?