raspberrypi/linux #5167: Troubles with Custom DPI settings w/ vc4-kms-dpi-generic & vc4-kms-v3d-pi4

jmacato, 2022-09-07:
### Describe the bug

I have a RPi4 Model B + a compiled version of Linux 6.0 branch (rpi-6.0.y)... Let's just say.. i have a very weird "LCD" and i couldn't find any settings except setting my own custom timings. It almost worked except for the following troubles:

i've used the following setting with vc4-kms-dpi-generic 

```
dtoverlay=vc4-kms-dpi-generic
dtparam=hactive=512
dtparam=hsync_invert
dtparam=hfp=14
dtparam=hsync=178
dtparam=hbp=0
dtparam=vactive=342
dtparam=vsync_invert
dtparam=vfp=0
dtparam=vsync=4
dtparam=vbp=24
dtparam=clock-frequency=15667200,rgb888
```

and it's booting for a very very very long time for some reason. Also when i launch kmscube it just quits... The accelerated framebuffer also takes a long time to switch and at random times it hangs.. I'm seeing mostly errors like: 

```

[  117.675343] vc4-drm gpu: [drm:drm_atomic_commit [drm]] committing 00000000154dd6b2
[  127.710096] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  127.710119] vc4-drm gpu: [drm] *ERROR* [CRTC:75:crtc-1] commit wait timed out
[  137.950043] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  137.950099] vc4-drm gpu: [drm] *ERROR* [CONNECTOR:46:DPI-1] commit wait timed out
[  148.190044] vc4-drm gpu: [drm] *ERROR* flip_done timed out
[  148.190100] vc4-drm gpu: [drm] *ERROR* [PLANE:65:plane-1] commit wait timed out
[  158.430041] vc4-drm gpu: [drm] *ERROR* flip_done timed out
```

everytime something graphical related program launches.

### Steps to reproduce the behaviour

use a compiled version of the rpi-6.0.y branch and the DPI settings i used in the attached config.txt. Also need a way to log the dmesg without any display.

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS: Devuan Chimaera + Linux 6.0
Linux macintosh 6.0.0-rc4-v8+ #1 SMP PREEMPT Tue Sep 6 02:24:58 PST 2022 aarch64 GNU/Linux
vcgencmd seems to be missing on my custom build.

### Logs

here's my dmesg log for reference:
[dm1.txt](https://github.com/raspberrypi/linux/files/9507422/dm1.txt)

and drm-info and other logs:
[km2.txt](https://github.com/raspberrypi/linux/files/9507431/km2.txt)
[drm.txt](https://github.com/raspberrypi/linux/files/9507428/drm.txt)
[drmerror.txt](https://github.com/raspberrypi/linux/files/9507429/drmerror.txt)
[km1.txt](https://github.com/raspberrypi/linux/files/9507430/km1.txt)

my config.txt for reference: 
[config.txt](https://github.com/raspberrypi/linux/files/9507407/config.txt)

6by9, 2022-09-07:
```flip_done timed out``` is a 10 second timeout, and does stall things quite commonly for 3 timeouts as well.

Does your display really have no hbp or vfp? That would be very strange to go straight from the h sync pulse into the active region, or v active into sync.

I also couldn't say for absolute certain that the display pipeline will be happy with 0 values. I'd suggest increasing hbp to at least 2 and vfp to at least 1 and see if the stalls go away.

jmacato, 2022-09-08:
@6by9 holy crap... thanks so much! That fixed that stalling issue! :tada: 

Actually tbh, the "LCD" in question is not an LCD but an old CRT. Specifically it's a Mac SE's CRT that i modded to take input from the Pi.

After adding 1 to the vbp/vfp, the driver seemed happy about it and the stalling is gone but i've noticed some shimmering artifact horizontal-wise (like some kind of wavy interference). The references for this had the following timing diagram and i was wondering if the Pi can replicate it (or at least find a way to remove the shimmering to make it display perfectly):

[image]

And a quote from the Guide to Mac Hardware:

```

The pixel clockrate (the frequency at which pixels are displayed)is 15.6672MHz,or
about 0.064μs per pixel. For each scanline, 512 pixels are drawn on the screen,requiring
32.68μs. The horizontal blanking interval takes the time of an additional 192 pixels,or
12.25μs.Thus, each full scanline takes 44.93μs, making the horizontal scan rate 22.25kHz.

The visible portion of a full-screen display consists of 342 horizontal scanlines and
requires 15367.65μs,or about 15.37ms,to scan.During the vertical blanking interval,the
turned off beam invisibly traces out an additional 28 scan lines,taking 1258.17μs or 
about 1.26ms to do so.

This means the full frame is redisplayed every 370 scanlines, or once every 16625.8μs.
That's about 16.6ms per frame and makes the vertical scan rate equal to 60.15Hz
```

jmacato, 2022-09-08:
So i played around with the horizontal front and backporch, and i adjusted hfp from 14 to 12 and hbp from 0 to 2. The display is now perfect 👨‍🍳 💯 

Final Timings for the folks that would like to reproduce this to their old macs:

```
enable_dpi_lcd=1
display_default_lcd=1
dtoverlay=vc4-kms-dpi-generic
dtparam=hactive=512
dtparam=hfp=12,hsync=178,hbp=2
dtparam=vactive=342
dtparam=vfp=1,vsync=4,vbp=24
dtparam=clock-frequency=15628800,rgb888
```

Thanks so much @6by9 🙏

closed 2022-09-08 by jmacato

jmacato, 2022-09-22:
@6by9 BTW, with regards to this issue, is it possible to do 1-bit (black and white only) framebuffer with GPU acceleration still?

6by9, 2022-09-22:
> BTW, with regards to this issue, is it possible to do 1-bit (black and white only) framebuffer with GPU acceleration still?

Not trivially.
GL only composes to 32bpp RGB (either with alpha or padding).
In theory one could abuse the gamma tables to quanitse the data, but that's still in RGB space rather than single channel.
There are colour conversion matrices on the HDMI outputs, but I don't recall if there are on DPI. Likewise I think the HVS does, but they aren't exposed.

However a simplistic conversion from full colour to mono doesn't always look great. There was a presentation at the Embedded Linux Conference Europe just last week on how moving from fbdev to DRM has left a hole for lower bit depth displays. Slides at https://elinux.org/images/6/6d/Trading_Fbdev_for_DRM_no_returns_accepted_Handouts.pdf. Slide 30 covers mono and low bit depth.

6by9, 2022-09-22:
I was at the talk, and the presenter wasn't always that complementary about the way DRM does stuff.
fbdev was deprecated largely due to lack of maintainers and too many security issues. Someone has stepped forward recently as maintainer, but it is still largely a dead end.

You should be able to backport those patches fairly easily if you felt like building your own kernel. It still doesn't handle X actually rendering to lower bit depths, but does give you the conversion functions in the kernel.
I'm not sure that it helps you greatly if you're using DPI where the hardware has already composed the frame. It's of more useful for SPI type displays.

jmacato, 2022-09-22:
Ahh... yeah actually i built my own kernel for this build since i was trying to get the newest fixes from this repo. 
Right now im not using Xorg but only apps directly rendering via KMS (mpv, minivmac via SDL2, etc). Would those benefit with these patches?

6by9, 2022-09-22:
> Right now im not using Xorg but only apps directly rendering via KMS/DRM (mpv, minivmac via SDL2, etc). Would those benefit with these patches?

Pass, but probably not.
You'll gain the definitions for DRM_FORMAT_C[124], DRM_FORMAT_R[124], and DRM_FORMAT_D[1248], the HVS (composition pipeline on vc4) won't gain support for them based on these patches.
AIUI this is all for "input side" to create planes of suitable formats to be composed. With vc4 it will compose to 32bpp on the output side, so you've probably lost any benefit.

Looking at the specs for the HVS it does support 1, 2, 4, and 8 bpp palette formats. It should be possible to use those for for all the above formats (R[1248] and D[1248] become fixed palettes with defined greyscale values), but it's not terribly high on the priority list.

jmacato, 2022-09-22:
Ahh, that's too bad then. I hope it'll be considered to add those palette formats in the future vc4 drivers